openvehicles/Open-Vehicle-Monitoring-System-3 #342: OTA updates available (when not on wifi)

markwj, 2020-03-05:
The current logic for OTA updates is that if wifi is off (and unless that special OTA-over-3G flag is set), the system won’t check for OTA updates at all.

The change here is to always check for OTA updates (either 3G or WiFi), but to only _install_ them if on WiFi (or the special OTA-over-3G flag is set). If we were not able to install, we can still send out a PUSH notification that the update is _available_ to be installed. The check itself is low-bandwidth, and only adds a couple of KB of data each day to the data usage.

glynhudson, 2020-03-10:
Good idea. Some users without easy access to WiFi may opt to pay the extra data charge to install the update via GSM. Would it be possible to display the update size in Kb so the user can then work out the data cost and decide if they want to install or not?

markwj, 2020-03-11:
Resolved in #8a5aa85c. Also fixed a few bugs and improved the notification messages. I don't think that the ota auto.allow.modem setting worked before (it would probably have allowed the notification, but not the actual download) - that has now been fixed (although I have doubts as to whether the download could actually be successful over modem).

closed 2020-03-11 by markwj