openvehicles/Open-Vehicle-Monitoring-System-3 #344: Server V3 MQTT: post only selected metrics

glynhudson, 2020-03-10:
I've been using server V3 to post metrics to an MQTT server for logging using Emoncms, informing OpenEVSE chargepoint of vehicle SoC and logging GPS data to OwnTracks. It works great, I've been running this for flawlessly for the past year or so. 

Currently, when server V3 is enabled all metrics are periodically posted. This adds a *lot* of extra data to the GSM bandwidth. e.g Using hologam my monthly data cost when from $1.9/month (running only V2) to $17/month when enabling V3 server. 

I'm aware it's possible to reduce the rate the V3 data is posted, however this would have a detrimental effect on the metrics I'm logging. Ideally what I would like is just to be able to select which metrics are posted e.g just SoC, and GPS co-ordinates. This would significnaly reduce the amount of data required. 

Is this possible currently? 

I've been working with @trystanlea, the main developer behind https://emoncms.org to develop a MQTT integration with OVMS to enable users to easily log OVMS metrics to emoncms. We're keen to avoid a huge amount of metrics being unnecessarily posted to the server.

TrystanLea, 2020-03-10:
If providing per-metric selection is too complicated, I imagine it could be.. perhaps having a reduced metrics option that would contain a collection of metrics would be another approach e.g:

- metric/v/b/soc
- metric/v/b/soh
- metric/v/p/altitude
- metric/v/p/latitude
- metric/v/p/longitude
- metric/v/p/odometer
- metric/v/b/12v_voltage

leres, 2020-03-28:
My credit card statement tells me I should be interested in this (it's not outrageous but 3-4 times higher than it used to be).

The above list of metrics look good; two more I care about are m/net/mdm/sq (cellular signal strength) and v/p/satcount (number of gps satellites in view).

Can we do something simple such as a web console page with check boxes for all possible metrics and a default subset enabled by default?

markwj, 2020-04-02:
I think it should be simple to have an optional config to list the metrics that should be sent. If not defined, it will send all. Is that sufficient for this requirement? If so, I am happy to do it.

TrystanLea, 2020-04-08:
Thanks @markwj that would be perfect and really appreciated if possible.

glynhudson, 2020-08-26:
Hi @markwj, Is there any developments on this feature? It's currently a bit of roadblock with regard to encouraging use of V3 server, $200+/yr is quite a financial hit. I don't think many individuals would be happy to pay this. 

V3 server has so many useful integration possibilities if the data cost was more affordable. We're keen to launch OVMS SoC integration with OpenEVSE to enable automatic charge scheduling on dynamic electricity tariffs using [Demand Shaper](https://github.com/emoncms/demandshaper)

Steefph, 2020-11-20:
Perhaps I'm spotting it wrong but there is some parameter in the Server V3 " MyOvmsServerV3->m_sendall" which should do this? But there does not seem to be a way to set this to "false" as it will always be set to true on connected. There has been some discussion in the mail-list back in 2018 regarding a potential filter. 

I personally think that the nicest solution would be a filtered send with hysteresis. So if a value hasn't changed within the configured range just don't transmit it as it has not point to do so. 
An extra configure page for Server V3 would be awesome where you can tick a box if it has to be send in "connected" or "idle" state and the hysteresis for that metric.

dexterbg, 2020-11-20:
`m_sendall` is used to cause a full metrics update on initial connect to the server. Afterwards, metrics are now already only transmitted when their values change. That's the purpose of the `IsModifiedAndClear()` call.

elbow, 2021-01-16:
@glynhudson @TrystanLea ,

I've become interested in this as I've worked on the i3 vehicle support and I'm willing to do some work on the v3/mqtt code in OVMS.

But I have to ask - Hologram has very expensive data charges.  Here in South Africa a SIM with 500MB of data per month can be had for €2,10 per month on a month-to-month contract. Do you not have options like this in your market? (OK: roaming is not a concern for most users here and Hologram is helpful for that).

I'm with @dexterbg that the v3 server only sends all metrics on first connect - maybe your MQTT server is timing out connections and forcing OVMS to have to reconnect often?

Also - the v3 server in OVMS supports three update rates as it is now -

1) If you set "config vehicle stream 1" then the v3 code sends every metric immediately that it changes (well, max one update per second).  Most up to date, but most data if metrics change frequently.
2) Otherwise, if you connect a "client" then it will send changed metrics every "config set server.v3 updatetime.connected" seconds
3) Otherwise "config set server.v3 updatetime.idle" seconds.

See these values with "config list server.v3"

I set like so:

OVMS# config list server.v3
server.v3 (readable writeable)
  port: ...
  server: ...
  tls: ...
  topic.prefix: ovms/vehicleid/
  updatetime.connected: 10
  updatetime.idle: 300
  user: ...

To "connect" a client to the OVMS it needs to publish an MQTT message announcing itself.  The topic needs to be "<prefix>/client/<clientid>/active" with value of 1 to be active, 0 to be "disconnected".  <prefix> is set in the server.v3 config - "config set server.v3 topic.prefix ..." or in the GUI, and <clientid> is some unique id for your "client".

You must send the active=1 at least once per 120 seconds to stay active.

I just used a very simple NodeRed flow to push metrics to EmonCMS - it basically just renames the tags to be the format that EmonCMS expects.

I was intending to have that flow "connect" to OVMS when it sees the car come alive - driving or charging - so as to get more rapid updates when something is happening.

I must say it would be easy to have the v3 server on OVMS by itself decide to use a "updatetime.charging" and "updatetime.on" to automatically switch rates.

Regards
Steve (@elbow)

ericyeargan, 2023-04-10:
First thanks to the OVMS maintainers for an awesome product - I recently added an OVMS module to my IONIQ5 and it works great! 

The main sticking point has been hologram.io cost so I took a crack at implementing a basic v3 server metric white/black list. PR #867 adds `whitelist` and `blacklist` config instances to `server.v3`. I'll add some information on using them to the PR. If it looks good, I'd be happy to implement the web front-end as well.

PR #872 (2023-04-15): Implement Server v3 metrics filtering
Implement Server v3 metrics filtering with configurable include/exclude list.  See "OVMS Server v3" section of docs/source/userguide/components.rst for details.

A new id_filter component has been factored out of components/can/canlog for use by canlog and the new server v3 metrics filtering functionality.

(see PR #867 for an earlier attempt)

Resolves #344
commit 88a6ea7a (2023-04-15): Implement Server v3 metrics filtering with configurable include/exclude list.  See "OVMS Server v3" section of docs/source/userguide/components.rst for details.

Resolves #344
closed 2023-04-15 by dexterbg