openvehicles/Open-Vehicle-Monitoring-System-3 #353: Cell voltage 0.1V to high?

RoaldNakstad, 2020-04-15:
I have noticed that command 'bms status' gives voltage readout 0,1V higher on each cell than running 'xse bmsdiag' on my Smart 451 ED. 
I believe that 'bms status' is wrong, since it has reported over 4.2V during charge, and as I know the charging finishes before/at 4.2V

RoaldNakstad, 2020-04-16:
Here is output that shows the difference when commands are run at same time:

OVMS# xse bmsdiag
-------------------------------------------
---- ED Battery Management Diagnostics ----
----         OVMS Version 1.0          ----
-------------------------------------------
Battery VIN: WMEEJ9AAXEK733054
Time [hh:mm]: 09:52,   ODO : 24450km
-------------------------------------------
Battery Status   : DEGRADED: 00110101

Battery Production [Y/M/D]: 2013/7/17
Battery-FAT date   [Y/M/D]: 2013/7/17
Rev.[Y/WK/PL] HW:2012/38/0, SW:2014/18/0
-------------------------------------------
SOC : 100.0%, realSOC: 95.0%
HV  : 10.5 V, 0.00 A, 0.00 kW
LV  : 12.2 V
-------------------------------------------
CV mean :  4228 mV, dV = 7.25 mV
CV min  :  4205 mV
CV max  :  4253 mV
OCVtimer: 20181 s
-------------------------------------------
Last measurement      : 44 day(s)
Measurement estimation: 0.177
Actual estimation     : 0.333
CAP mean: 19062 As/10, 53.0 Ah
CAP min : 17898 As/10, 49.7 Ah
CAP max : 19037 As/10, 52.9 Ah
-------------------------------------------
HV contactor state OFF, for: 30579 s
Cycles left   : 289830
of max. cycles: 300000
DC isolation  : 10545 kOhm, OK
-------------------------------------------
Temperatures Battery-Unit /degC: 
module 1: 6.9, 8.4, 7.8
module 2: 7.1, 11.7, 8.3
module 3: 6.8, 10.0, 7.7
   mean : 8.3, min : 6.8, max : 11.7
coolant : 5.4
-------------------------------------------
 # ;mV   ;As/10
  1; 4128; 18202
  2; 4128; 18303
  3; 4128; 18738
  4; 4145; 17110
  5; 4127; 19040
  6; 4129; 18974
  7; 4124; 19194
  8; 4128; 19040
  9; 4124; 19194
 10; 4125; 19150
 11; 4127; 19084
 12; 4127; 19062
 13; 4128; 19106
 14; 4126; 19128
 15; 4128; 19062
 16; 4127; 19040
 17; 4129; 19040
 18; 4130; 19062
 19; 4130; 19062
 20; 4129; 19128
 21; 4128; 18996
 22; 4127; 19106
 23; 4126; 19128
 24; 4128; 19040
 25; 4128; 19106
[... 250 lines cut ...]
    +--------+--------+
 82 | 4.230V |   0.0C |
    +--------+--------+
 83 | 4.227V |   0.0C |
    +--------+--------+
 84 | 4.230V |  -0.0C |
    +--------+--------+
 85 | 4.227V |   1.0C |
    +--------+--------+
 86 | 4.228V |  -0.0C |
    +--------+--------+
 87 | 4.228V |   2.0C |
    +--------+--------+
 88 | 4.228V |   0.0C |
    +--------+--------+
 89 | 4.229V |   1.0C |
    +--------+--------+
 90 | 4.230V |   1.0C |
    +--------+--------+
 91 | 4.228V |   0.0C |
    +--------+--------+
 92 | 4.230V | 71859503208318044733440.0C |
    +--------+--------+
 93 | 4.216V | 71426891664741958651414514237440.0C |
    +--------+--------+

KommyKT, 2020-04-16:
The differenc is from the offset in source code (OvmsMetricInt*
mt_myBMS_ADCvoltsOffset;       //!< calculated offset between RAW cell
voltages and ADCref, about 90mV)

The bms status show the raw values of cells.
The xse bmsdiag show raw values - mt_myBMS_ADCvoltsOffset; which is 90mV

RoaldNakstad, 2020-04-16:
Ok, so this difference is not a bug. If I understand this correct, bms status show the same as if cells were messaured directly with a volt-meter. Any idea why bmsdiag calculates ADCvoltsOffset?

dexterbg, 2021-12-30:
Has there been any progress on this? The BMS cell values shall be assigned the real world voltages & temperatures, not some internal raw values, so if @KommyKT is right, this is a bug.