openvehicles/Open-Vehicle-Monitoring-System-3 #354: Flush UART buffer when power cycling modem

markwj, 2020-04-16:
Thomas Heder has had some success with using uart_flush to flush the UART ring buffer when power cycling the modem. This might explain why in some cases we cannot seem to recover modem functionality by simply power cycling the modem, but resetting the OVMS module seems to recover the modem. Flushing the buffer would not seem to affect this, but perhaps something is corrupt in the ring buffer itself, so using uart_flush clears that problem.

Anyway, it does not seem to do any harm, so not a problem to add it and see if it improves the situation.

leres, 2020-04-16:
That would be awesome if it fixed it.

markwj, 2020-04-16:
Change applied in 9de2ecd9.

closed 2020-04-16 by markwj