openvehicles/Open-Vehicle-Monitoring-System-3 #373: OVMS loses last SOC after crash

justingist, 2020-05-20:
I've been having a really frustrating issue since first installing OVMS on my Mitsubishi I-MiEV... the unit keeps crashing and after a reboot it always shows 0% battery, even if the car is on or charging. The only way I've found to fix this so far is to unplug from OBDII slot and then plug it back in. Restarting the unit has so far done nothing... I'm trying to figure out actually if it is even getting any CANBUS data after crash, tbh.

Here's an example of my latest crash... I only just turned off car like 10 mins ago and the unit is once again showing blank SOC.

Last boot was 318 second(s) ago
Time at boot: 2020-05-20 10:44:54 BST
This is reset #303597462 since last power cycle
Detected boot reason: Crash (12/12)
Reset reason: Interrupt watchdog (5)
Crash counters: 1 total, 0 early

Last crash: Alloca exception on core 0
Registers:
PC : 0x40098853 PS : 0x00060934 A0 : 0x8009829c A1 : 0x3ffafa40
A2 : 0x3f800000 A3 : 0x00000018 A4 : 0x00000000 A5 : 0x3fa07ed0
A6 : 0x00000000 A7 : 0x3f800010 A8 : 0x80098851 A9 : 0x3ffafa20
A10 : 0x00000001 A11 : 0x000001af A12 : 0x3fa07ed0 A13 : 0x00000001
A14 : 0x000000fe A15 : 0x3ffafb50 SAR : 0x00000000 EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000 LBEG : 0x4008f228 LEND : 0x4008f256 LCOUNT : 0xffffffff
Backtrace:
0x40098853 0x40098299 0x4008426d 0x4008430e 0x40086de5 0x4020a171 0x400e6378
Version: 3.2.012/ota_1/main (build idf v3.3-beta3-775-gdc1ca69 Apr 22 2020 09:37:08)

Example basic stats:

Not charging
SOC: -
Ideal range: 0M
ODO: 22850.4M
AC charge: 0.000kWh kWh
DC charge: 0.000kWh kWh
Currently at 0.000000,0.000000 (without GPS lock)
There are 2 locations defined
No active locations

KommyKT, 2020-05-31:
I don't understand your problem. If the key not turned to ON/ACC, or not charging no can data access because most ecu is sleeping. But after crash you start your car or charge it, wake up ecu and get soc values.

justingist, 2020-05-31:
So of course, the can data is basically silent when the car is not on. Agreed.

Here's the deal: when this crashes and goes to 0% SOC, it never recovers when the car is turned on. It may also crash while the car is on and then not receive any more SOC values from that time on.

The only thing that seems to rest this behaviour is to unplug OBDII and plug back in.

KommyKT, 2020-06-01:
Please try this firmware to see what it does. ( kommykt-nas.ddns.net:22289/dev/ovms3.bin ) this is a edge fw for the Trio, with the latest updates from the main repo. 
If you get a crash please send it.
The fw download speed is not to fastest but working.

KommyKT, 2020-07-27:
Anyone with this problem ?

dexterbg, 2021-12-30:
Seems to be fixed or no longer reproducable, closing.

closed 2021-12-30 by dexterbg