openvehicles/Open-Vehicle-Monitoring-System-3 #375: Verbose/Debug logging over ssh/web

markwj, 2020-06-02:
I don't think we can verbose/debug log over ssh/web. In particular, any verbose/debug logging in the functions of the ssh/web code itself will cause problems. Injecting a new log message for each log message sent to the remote. Endless loop.

dexterbg, 2020-06-02:
Possibly not as a general level, but it works if applied specifically to components.

This reminds me of my suggestion to generally remove or guard low level log outputs:
http://lists.openvehicles.com/pipermail/ovmsdev/2019-November/006460.html

I think most low level debug logs are no longer needed and can be commented out.

dexterbg, 2020-06-13:
Additionally, I've added a fix to the logging in timer context to our esp-idf fork:
https://github.com/openvehicles/esp-idf/commit/9024cd38ecbc78a46ed16b79d63f57812e72b123

That may lose some log messages in the worst case but it won't affect timing integrity anymore.

closed 2020-06-13 by dexterbg