openvehicles/Open-Vehicle-Monitoring-System-3 #392: Metrics error messages on boot

markwj, 2020-07-16:
On first boot (after flash), I get:

```
rst:0x1 (POWERON_RESET),boot:0x3b (SPI_FAST_FLASH_BOOT)
...
E (95) metrics: pmetrics_check: bad magic
E (100) metrics: pmetrics_check: bad version
E (104) metrics: pmetrics_check: bad size
E (109) metrics: pmetrics_check: out of range used
I (115) metrics: Persistent metrics serial 0 using 420 bytes
...
I (1102) housekeeping: reset_reason: cpu0=1, cpu1=14
```

dexterbg, 2020-07-16:
That's intended behaviour, the persistent metrics storage reports the type of errors found that lead to reinit (loss of persistence). Flashing via USB generally trashes the RTC RAM. I've also had OTA flashes trash the RTC RAM sometimes, but very infrequent.

See https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/380

leres, 2020-07-16:
If I'm reading rom/rtc.h correctly, this was a cold boot (power newly applied). If that was the case I agree with Michael this is expected.

Meanwhile I was thinking the other day that I could bypass the (newly added) persistent metrics check on a cold boot. Does that sound right?

dexterbg, 2020-07-16:
Yes, that's an option to avoid unnecessary error messages.

leres, 2020-07-17:
The fix was simple (and how I should have done it in the first place): #393

leres, 2020-07-17:
BTW, while testing I see that serial flashing has the side effect of wiping all memory including rtc slow ram. This is disappointing but I suppose expected; it provides a consistent starting point.

closed 2020-07-17 by dexterbg