openvehicles/Open-Vehicle-Monitoring-System-3 #398: SSH login failing with "invalid algorithm id" for RSA key

larroy, 2020-07-25:
I was trying to setup ssh with an RSA key, I always get:

E (998688) ssh: Error -20 in reception: invalid algorithm id

Key is generated in ubuntu 18.04. with 

```
ssh-keygen -t rsa -f .ssh/id_rsa_ovms
```

Looking at https://github.com/wolfSSL/wolfssh/blob/master/src/internal.c seems there's newer code for wolfssh. I tried to figure out if I could generate the key with a different key exchange but didn't see an obvious way to do this. Maybe updating wolfssh would help?

dexterbg, 2020-07-25:
Did you follow the instructions?
→ https://docs.openvehicles.com/en/latest/userguide/console.html#ssh-console

elbow, 2021-01-16:
```
ssh -c aes128-cbc admin@196.16.4.1
```

closed 2021-01-16 by dexterbg