openvehicles/Open-Vehicle-Monitoring-System-3 #409: Nissan LEAF P312A DTC & dashboard warning light

glynhudson, 2020-08-19:
I've been running OVMS in my 2014 Nissan LEAF for the past couple of years without any issues. 

Recently (possibly after upgrading to 3.2.014) I've started getting an EV warning light on the dashboard caused by DTC P312A. The error seems to happen when starting up the car for the 2nd time on a particular day without any charging taking place. The error does not seem to effect the operation of the car. 

I tried switching off CAN writes, but this did not have any effect. Error still occurred. 

After unplugging OVMS the error does not occur. I will try and testing rolling back to a previous OVMS release this week. Just wondering if anyone else has experienced the same issue?  

There have been a number of changes to the NL code in this release : https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/400 https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/399/ https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/397/ https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/394

What model of Nissan Leaf are you using to test these changes @drc38?

P312A is a CAN bus communication error:

[image]

Here is the DTC from Leafspy:

[image]

drc38, 2020-08-19:
Hi Glyn,
I'm using a 30kWh Leaf, the recent changes that affected can bus transmits were to the polling regime. Previously, all polls were being sent to can bus 2 (CAR can). This meant battery/temp/qc polls would fail when charging as the CAR can is asleep.  Initially, I switched polling to can bus 1 (EV can bus) as battery polls were more important than obtaining the vin etc. That said, polls are not frequent and I've never had an issue with a failed poll on an alternate bus causing an issue with the ovms module or car.

To allow the polls to use the correct can bus per poll type required an extra function to switch without resetting/interfering with the poll timing. This is in the EAP version as it was after the 3.2.014 release. 

Suggest trying the EAP version, but I cannot see that the ovms module or code would cause the DTC. Let's see what others report back.
Cheers Derek

drc38, 2020-08-19:
Looking at the NL code, setting the canwrite parameter is not currently used/implemented which would explain why changing the setting is not having any effect. I will submit a pull request so that it does actually turn off polling and commands.

Let me know how you get on with the either rolling back or using the EAP version. Also what do your logs show at the time of the DTC flagging?

glynhudson, 2020-08-19:
Thanks for your reply. It's fantastic you're working on improving the NL implementation. 

> This meant battery/temp/qc polls would fail when charging as the CAR can is asleep

This is not what I've experienced, I've been logging the SoC and battery temp during charging and the metrics have been updating correctly e.g here's a number of rapid charges increasing battery temp, the battery temp metric visibly increases during charging which indicates that the OVMS polling is working 

[image]

I'm already one EAP, 3.2.014 is the current EAP release (as far as I'm aware), I'll try and switch to EDGE. I will setup logging errors to an SD card. Capturing the error may be tricky as the DTC error happens sporadically. 

I will report back.

drc38, 2020-08-20:
Hi Glyn,
I think you may be looking at metrics that are not coming from the Poll PIDs in your chart. If you check v.b.p.voltage.avg as that's derived from individual cell voltages received from the Poller.

Yes, try the `edge` version as it now has the ability to turn off can writes too 🙂

drc38, 2020-08-25:
Hi Glyn,
I've been looking at the Polling code and there are possibly still issues because of the two separate can buses on the Leaf. Suggest leaving canwrites disabled for now. I am in the process of updating the Leaf code with code from @dexterbg's Twizy <IncomingPoller> and <ObdRequest>. With the ability to send an Obd Request to either canbus. @dexterbg appreciate if you can review the pull request particularly around the mutex and semaphore handling aspects when I submit it. Note it will remove using the mutex from <vehicle.cpp> as well.

That said, I suspect there may still be another issue when the Leaf is turned on, possibly the HV battery contactor upsets either the ovms module supply voltage or can buses flood / get corrupted. I plan to add a delay before the Poller starts to see if that provides a suitable workaround for now.

glynhudson, 2020-08-25:
Thanks for the info. I will leave CAN writes switched off for now. The error does not seem to have happened since upgrading to edge.

closed 2020-08-26 by glynhudson

glynhudson, 2020-08-27:
Another possible report of this issue: https://www.openvehicles.com/node/2578#comment-6882

drc38, 2020-09-03:
@glynhudson appreciate if you can retest with latest `edge` release and can writes enabled. Quite a few fixes required but it should now be far more stable. Just want to make sure on a gen1 though :)

glynhudson, 2020-09-03:
I installed latest edge build last night. Went for a drive today. All seems
good 👍

- sent from my mobile device