openvehicles/Open-Vehicle-Monitoring-System-3 #418: WolfSSL RSAKeyGen watchdog timeout / blocking events task

dexterbg, 2020-08-29:
Just saw this happen after a factory reset, will also happen on a new module. The module needed multiple attempts (reboots), so the time needed for the RSA key generation seems to be on the edge to an endless reboot loop after a factory reset.

I'll push the solution (lowering RSAKeyGen task priority to 0) in a few minutes.

But besides the primary issue, this is also unexpected behaviour in terms of the FreeRTOS task prioritization: the RSAKeyGen task is running at priority 5, so the events task running at priority 8 should get all the time it needs to process events. So why do we see an event queue overflow here?

```
Generating SSH Server key, wait before attempting access.
E (20802) events: SignalEvent: queue overflow, event 'ticker.10' dropped
E (21802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (22802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (23802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (24802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (25802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (26802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (27802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (28802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (29802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (30802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (30802) events: SignalEvent: queue overflow, event 'ticker.10' dropped
E (31802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (32802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (33802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (34802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (35802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (36802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (37802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (38802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (39802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (40802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (40802) events: SignalEvent: queue overflow, event 'ticker.10' dropped
E (41802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (42802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (43802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (44802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (45802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (46802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (47802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (48802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (50802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (50802) events: SignalEvent: queue overflow, event 'ticker.10' dropped
E (51802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (52802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (53802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (54802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (55802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (56802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (57802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (58802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (59802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (60802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (60802) events: SignalEvent: queue overflow, event 'ticker.10' dropped
E (60802) events: SignalEvent: queue overflow, event 'ticker.60' dropped
E (61802) events: SignalEvent: queue overflow, event 'ticker.1' dropped
E (62462) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (62462) task_wdt:  - IDLE1 (CPU 1)
E (62462) task_wdt: Tasks currently running:
E (62462) task_wdt: CPU 0: IDLE0
E (62462) task_wdt: CPU 1: RSAKeyGen
E (62462) task_wdt: Aborting.
abort() was called at PC 0x400e61c4 on core 0
```

With the RSAKeyGen task running at priority 0, everything works as expected: CPU time is shared equally between the RSA and the idle task, and events get processed normally:

```
Generating SSH Server key, wait before attempting access.
OVMS# mo ta
Number of Tasks = 17        Stack:  Now   Max Total    Heap 32-bit SPIRAM C# PRI CPU% BPR/MH
3FFAFB88    1 Rdy esp_timer         436   644  4096   40288    644  31256  0  22   0%  22/ 0
3FFC0E90    2 Blk eventTask         476   860  4608     104      0      0  0  20   0%  20/ 0
3FFC3314    3 Blk OVMS Events       480  2640  8192   52016      0  23276  1   8   4%   8/ 0
3FFC6764    4 Blk OVMS DukTape      496 10864 12288     188      0  92360  1   3   7%   3/ 0
3FFC93B8    5 Blk OVMS CanRx        468   468  4096    3152      0  35668  0  23   0%  23/ 0
3FFCA148    6 Blk ipc0              420   420  1024    7804      0      0  0  24   7%  24/ 0
3FFCA74C    7 Blk ipc1              424   520  1024      32      0      0  1  24  28%  24/ 0
3FFCC5BC   10 Rdy IDLE0             408   504  1024       0      0      0  0   0  59%   0/ 0
3FFCCB54   11 Rdy IDLE1             416   512  1024       0      0      0  1   0  39%   0/ 0
3FFCD8EC   12 Blk Tmr Svc           376   888  3072      88      0      0  0  20   0%  20/ 0
3FFCBA24   17 Blk tiT               544   960  3072     704      0    520  1  18   0%  18/ 0
3FFD1EEC   20 Blk OVMS SIMCOM       632  1880  4096    5836      0      0  0  20   0%  20/ 0
3FFD4F34   21 Blk wifi              480  2256  3584   38112      0   5092  0  22  13%  22/ 1
3FFDFDB0   22 Rdy OVMS Console      844  2444  6144       0   3608  24020  1   5   0%   5/ 1
3FFE2490   23 Blk OVMS NetMan       904  1384  8192     188      0     88  1   5   0%   5/ 0
3FFE3B38   24 Blk mdns              500  1396  4096     384      0      4  0   1   0%   1/ 0
3FFE7710   25 Rdy RSAKeyGen        1660  2076  4096   10736      0      0  1   0  22%   0/ 0
I (7202) sdcard: SD CARD has been inserted
I (7272) sdcard: mount done
I (12202) housekeeping: System considered stable (RAM: 8b=80252-80516 32b=5888)
I (17202) simcom: State timeout, transition to 13
I (17202) simcom: State: Enter PoweredOff state
I (17202) gsm-mux: Stop MUX
OVMS# mo ta
Number of Tasks = 17        Stack:  Now   Max Total    Heap 32-bit SPIRAM C# PRI CPU% BPR/MH
3FFAFB88    1 Blk esp_timer         436   676  4096   40288    644  31256  0  22   0%  22/ 0
3FFC0E90    2 Blk eventTask         476   860  4608     104      0      0  0  20   0%  20/ 0
3FFC3314    3 Blk OVMS Events       480  2640  8192   59532      0  23276  1   8   0%   8/ 0
3FFC6764    4 Blk OVMS DukTape      496 10864 12288     188      0  99168  1   3   1%   3/ 0
3FFC93B8    5 Blk OVMS CanRx        468   468  4096    3152      0  35668  0  23   0%  23/ 0
3FFCA148    6 Blk ipc0              420   420  1024    7804      0      0  0  24   0%  24/ 0
3FFCA74C    7 Blk ipc1              424   520  1024      32      0      0  1  24   0%  24/ 0
3FFCC5BC   10 Rdy IDLE0             408   504  1024       0      0      0  0   0  99%   0/ 0
3FFCCB54   11 Rdy IDLE1             416   512  1024       0      0      0  1   0  49%   0/ 0
3FFCD8EC   12 Blk Tmr Svc           376   888  3072      88      0      0  0  20   0%  20/ 0
3FFCBA24   17 Blk tiT               544   960  3072     704      0    520  1  18   0%  18/ 0
3FFD1EEC   20 Blk OVMS SIMCOM       632  1880  4096    5836      0      0  0  20   0%  20/ 0
3FFD4F34   21 Blk wifi              480  2256  3584   38112      0   5092  0  22   0%  22/ 1
3FFDFDB0   22 Rdy OVMS Console      844  2492  6144     360   3608  24428  1   5   1%   5/ 1
3FFE2490   23 Blk OVMS NetMan       904  1384  8192     216      0     88  1   5   0%   5/ 0
3FFE3B38   24 Blk mdns              500  1396  4096     104      0      4  0   1   0%   1/ 0
3FFE7710   25 Rdy RSAKeyGen        1660  2156  4096   10736      0      0  1   0  48%   0/ 0
SSH server key installed with fingerprint bNmgXkcbvoe6njcIkKv2jHhSxdZAKn6hZrYcPuje8v4
```

So… why does the RSAKeyGen task running at prio 5 stop the events task running at 8 from working? Does the events task need to wait for another prio 5 task? Solving this could as well solve the spurious event overflow crashes we still see.

markwj, 2020-08-29:
Should we increase queue size, as a workaround for long running tasks?

dexterbg, 2020-08-29:
No, I think I found the reason: the events get dispatched to the Duktape task synchronously. The Duktape task runs on prio 3, so doesn't get CPU share if a task at prio 5 is running continuously. That blocks the events task waiting for Duktape as well.

Instead of raising the Duktape priority to 8 (which wouldn't cure higher priority tasks hogging the CPU, and wouldn't feel right as scripts then can block the console) I currently think about running event scripts asynchronously.

slcasner, 2020-08-29:
On Sat, 29 Aug 2020, Michael Balzer wrote:

> No, I think I found the reason: the events get dispatched to the
> Duktape task synchronously. The Duktape task runs on prio 3, so
> doesn't get CPU share if a task at prio 5 is running
> continuously. That blocks the events task waiting for Duktape as
> well.

Aha.  When I read your first post about this I wanted to ask what had
changed the host key generation was implemented; now I see that at
least part of the answer is that Duktape was introduced.

                                                        -- Steve

dexterbg, 2021-12-30:
I think I forgot to close this, the async event scripts fixed this.

closed 2021-12-30 by dexterbg