openvehicles/Open-Vehicle-Monitoring-System-3 #422: v3 server metric: GPS location JSON

ethan4, 2020-09-01:
Hi guys,

I’m trying to integrate my OVMS with Home Assistant to provide location tracking for the car to trigger automations when home etc. but having trouble with the current format of the GPS location. 

Is it possible to supply the GPS location (and location name as a separate topic) in a JSON string similar to what HA is expecting in this example?

> {"longitude": 1.0,"gps_accuracy": 60,"latitude": 2.0,"battery_level": 99.9}

See this article for the example source and usage description https://www.home-assistant.io/integrations/mqtt_json/

ethan4, 2020-09-01:
Hi guys,

Is it possible to supply the GPS location as a separate metric in a format similar to this example? 

`{"longitude":` 1.0,"gps_accuracy": 60,"latitude": 2.0,"battery_level": 99.9}`

This is for use with Home Assistant automation but may be useful for other people as well. See here for more detail on the example: https://www.home-assistant.io/integrations/mqtt_json/

dexterbg, 2020-09-04:
I think you could be better off implementing an HA plugin tailored to the OVMS metrics topics, and/or using the OVMS builtin location events to trigger your actions.

If you absolutely need to use that plugin, you could post your JSON string to a "data" or "info" type notification. Notifications are published in topic `<prefix>/notify/<type>/<subtype>`. Data notifications add their IDs and timestamps to the topic, so may be tricky to use with that plugin. Info notifications are forwarded as push notifications by default, but the channels can be configured per subtype, so you can easily restrict your location notification to be sent to the v3 server only.

See:
- https://docs.openvehicles.com/en/latest/userguide/scripting.html#ovmsmetrics
- https://docs.openvehicles.com/en/latest/userguide/scripting.html#ovmsnotify
- https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/commit/d2c2b1edd696f9c94446a332ef4efff4e29b8359

Regards,
Michael

ethan4, 2020-10-15:
Hi Michael,

Any chance of making the geofenced location available as a standard metric instead?

dexterbg, 2020-10-18:
Good idea, I've implemented that in commit 9ab6b6b852b05f09f3b9e94b66778ded0d9276b6.

Regards,
Michael

pascalsaul, 2020-10-18:
Did you try this in HA automations_old.yaml?
```
- alias: ovms_tracker
  trigger:
  - platform: state
    entity_id: sensor.ovms_metric_v_p_latitude,sensor.ovms_metric_v_p_longitude
  action:
  - service: device_tracker.see
    data_template:
      dev_id: ovms
      gps:
        - "{{ states('sensor.ovms_metric_v_p_latitude') }}"
        - "{{ states('sensor.ovms_metric_v_p_longitude') }}"
```

ethan4, 2020-10-18:
> Did you try this in HA automations_old.yaml?
> 
> ```
> - alias: ovms_tracker
>   trigger:
>   - platform: state
>     entity_id: sensor.ovms_metric_v_p_latitude,sensor.ovms_metric_v_p_longitude
>   action:
>   - service: device_tracker.see
>     data_template:
>       dev_id: ovms
>       gps:
>         - "{{ states('sensor.ovms_metric_v_p_latitude') }}"
>         - "{{ states('sensor.ovms_metric_v_p_longitude') }}"
> ```
> 
> HA preview:
> [image]

Thanks for the suggestion, this is helpful!

closed 2020-11-08 by ethan4