openvehicles/Open-Vehicle-Monitoring-System-3 #472: NL: Trigger 12V Battery Charging manually

mjkapkan, 2020-11-23:
As you may know Leaf has a function to top-up the 12V battery when charging. However it also has a feature to charge the battery if it goes below 10V. 

The problem is that anything below 11.5V is not healthy for the battery. Also some older batteries no longer have enough juice in them to trigger the high voltage relay at such a low voltage resulting in vehicle being undrivable before 12V battery is externally charged.

I noticed there is specific CAN frame (TBD) being sent when the HV relay is triggered for starting the 12V battery charging. This means that we can have a separate command in OVMS like 12vbatterycharge start/stop and can integrate this in to existing 12V battery monitoring GUI.

dalathegreat, 2020-11-28:
I think you're onto something here! The 2011-2013 especially has some very flawed top-up logic when it comes to maintaining the 12V battery, having this feature available would be a gamechanger!

glynhudson, 2020-11-29:
Agree, this would be a fantastic feature. Even better than remotely trigger a 12V topup would be to instruct OVMS to automatically trigger a topup when 12V is low. OVMS already has a feature to send a 12V low alert message at a predefined voltage level. 

>As you may know Leaf has a function to top-up the 12V battery when charging. However it also has a feature to charge the battery if it goes below 10V.

I don't think this is true, at least not in experience. I've been logging the 12V battery voltage for three different leafs ranging from 2012-2019, they all seem to top up the 12V in the same way. Every 24hrs exactly (if the car has not been used) the leaf triggers a 12V topup. This topup lasts exactly 5min and is often not sufficient to stop the 12V battery slowly draining. In my experience the Leaf doesn't have a feature to do a longer topup when the 12V battery reaches critical. Over a period of weeks the 12V battery dies. 
  
[image]

It's also useful to note that if a Leaf is left plugged into a charge point / granny cable / wall box it does not do any topup at all! It just lets the 12V die, which is crazy really when it has unlimited access if required from the EVSE. 

[image]

I've been using the remote climate function of OVMS as a way to remotely trigger a 12V topup, since when the remove HVAC is switched on the DC-DC converter also chargers the 12V, this is obviously a very energy wasteful way to charge a 12V! 

As a side note a Leaf will successfully switch on with 8V. I manage a number of community share Nissan Leafs, the cars poor management of 12V battery causes me a lot of pain :-/

mjkapkan, 2020-12-04:
@glynhudson I have tried this method again now several times and the car started charging the battery just when it dropped down to 10V. Have observed the CAN traffic at that point and narrowed down it to these messages:
`
can2/5a9 | 7b 1d b0 11 c4 40 00
-- | --
can1/5a9 | 7b 1d b0 11 c4 40 00
can1/5b9 | 60 3c 40 46 28 3c 80
can1/5fc | 0e 72 0c 60 27 00 ff
can2/5f9 | ff 1f e0 80 20 0e 00 00
can2/5fb | ff 1f e1 00 03 00
can2/5fc | 0e 72 0c 60 27 00 ff
can2/5e3 | 83 11 c4 40 80
can1/59e | 00 00 09 54 60 00 00 00
can1/5c0 | 40 5a 5a 00 75 04 0c 03
can2/5c0 | 40 5a 5a 00 75 04 0c 03
can2/509 | ff 09 71 81 00 55 40 ff
can1/54f | 2e 40 00 00 04 00 d0 27
can2/54b | 10 49 88 09 34 00 00 00
can1/54c | 89 65 40 00 00 a8 50 00
can2/54a | 12 80 70 06 2f 00 00 28
can1/54a | 12 80 70 06 2f 00 00 28
can2/50d | 5e 00 00 00 00 00 00 00
can2/510 | 04 00 10 00 1e 00 00 50
can2/5e4 | 04 00 00
can1/54b | 10 49 88 09 34 00 00 00
can2/50a | 05 b2 a0 0c 00 21 04 00
can2/551 | 00 00 00 80 ff 00 40 00
can2/5b3 | 50 82 ff f3 c0 ae 93 8a
can2/351 | 00 00 00 00 00 02 42 02
can2/02a | 00 00 00 00 00 0c
can2/559 | 48 00 10
can2/35d | 80 03 00 00 00 00 00 00
can1/50a | 05 f2 a0 0c 00 21 04 00
can1/50b | 00 00 04 c0 00 00 00
can1/55a | 00 2c 2b 00 5f 00 5b 28
can1/55b | e5 80 55 00 e5 c0 12 75
can1/5bc | 2b 80 93 50 83 00 1f ff
can2/354 | 00 00 00 00 00 08 00 00
can2/260 | c8 c8 7d 00
can2/292 | 80 08 64 80 20 00 00 01
can2/285 | 00 00 00 00 00 00 a1 28
can2/245 | 7f e8 02 18 a1 00 7f e1
can2/284 | 00 00 00 00 00 00 a1 27
can1/284 | 00 00 00 00 00 00 a1 27
can2/1d5 | 00 00 00 00 d6
can2/2de | 00 00 00 00 00 00 03 b6
can2/1cb | 00 03 00 02 10 01 2c
can2/1cc | 00 00 05 50
can2/176 | 00 00 00 00 00 00 04
can1/11a | 01 40 04 aa c0 00 00 d5
can1/1d4 | 6e 6e 00 04 07 46 41 5d
can1/1f2 | 10 64 00 a0 00 78 00 8e
can1/1da | c3 00 18 00 00 01 02 a5
can2/174 | 00 00 00 aa 04 00 00 00
can2/180 | 00 00 00 00 00 00 24 00
can1/1db | ff 40 c3 eb 5e 00 03 ff
can1/1dc | 6e 01 3f fd 08 c0 c7 a2
can2/002 | fd ff 00 07 9c
`

glynhudson, 2020-12-04:
@mjkapkan how can I activate this feature? It would be very useful if the Leaf did charge when the battery got down to 10V. As you can see from my graphs all the Leaf's I've been involved in managing just let the 12V slowly die over a period of weeks without any effort to topup when the voltage is 10V. Which Leaf model have you observed this on? I agree, it should really topup more before it gets to this critical stage.

mjkapkan, 2020-12-04:
@glynhudson, interesting. My Leaf is a 2013 model manufactured in USA. Might be a firmware version or something? 
Anyway I think we can figure this out, we just need to find the right message sequence.

dalathegreat, 2020-12-04:
If the 12V battery goes below ~11.5V, the relays inside the battery will have issues triggering. Meaning that if you still send the right messages, it will fail to charge the battery.

It is only the 2011-2013 ZE0 which has the worst charging logic, and way more likely to end up with a depleted 12V battery.

If we want this to work, we need to be more pro-active and charge when it drops ti 11.5V

mjkapkan, 2020-12-04:
@dalathegreat Exactly, but this is just a formality once we know gow to do it ;-)

mjkapkan, 2020-12-04:
@glynhudson I think I found a catch. All the tests I have done so far were done with the vehicle in accessory mode. I It was an easiest way to drain down the battery to 10V by turning on all the seat heaters and fan to max.
However, when I tried to do the same with the car completely off, it did not work. Anyway, it might still be possible to do this by identifying the exact messages that trigger this.

drc38, 2020-12-10:
@mjkapkan if you can trigger it successfully in accessory mode by sending a can write (I'd start with trying ones like 5e4 or 559) then to go from there to when the car is off will require sending a bus wakeup command first eg 68c.

drc38, 2020-12-13:
This can message might be useful, on my Leaf 0x1F2 d[7]/10 appears to be the 12V charger output voltage. When charging it goes from 13.x up to 14.2V. If not charging it reads circa 1V

glynhudson, 2021-03-25:
>If the 12V battery goes below ~11.5V, the relays inside the battery will have issues triggering. Meaning that if you still send the right messages, it will fail to charge the battery.

I've experienced a Leaf being able to start up with about 8.5V, probably not reliably but anything above 11V should be more than enough.

mjkapkan, 2021-03-29:
It also depends on the battery health. My old and weak 12V battery can not trigger the relays below 11V anymore. But I know that previously it used to work even with 8V. Anyway, this feature would really help. So I will come back to this once I have a chance.

jabelone, 2021-04-11:
I had the 12v die a couple of times after I installed a dash cam as the low voltage cut off did not work correctly (since replaced the cut off device and it's working now). It was quite amusing having to jump start my electric car from my ICE motorbike. 😅 I also find that after a week or so of not driving, the 12v battery tends to get quite low as the dash cam already drains it pretty low before it cuts off so I would find this feature very useful.

Do you have any pointers on how to get started with reading CAN messages from the bus and working out which one(s) possibly trigger a 12v battery topup? I've got some experience reverse engineering protocols (mainly http/web based) so am happy to poke around if someone can point me in the right direction.

Cheers!

mjkapkan, 2021-04-17:
@jabelone I personally prefer using ODB2 dongle and sniffing the packets using Wireshark. But would suggest to try the re-tools plugin from OVMS official page. It should help you to get started.

matchett808, 2021-08-07:
Did anyone find the correct command?

dalathegreat, 2021-12-26:
By sending 0x679 with 1length 0 content, the DC-DC converter kicks in on AZE0 LEAFs, caveat, only when plugged in and fully charged. Will continue to experiment.

mjkapkan, 2021-12-28:
@dalathegreat Exactly. This could be easily triggered by adding a command to existing 12v battery monitoring feature that sends an alert when voltage goes below specified limit. We can add a checkbox to feature page to enable/disable this.

guidinturkka, 2022-10-17:
Any progress on this? How would I try this out with the OVMS command line?

Turee, 2022-10-17:
Any progress on this? How would I try this out with the OVMS command line?

dhi20, 2023-12-20:
Did anyone manage this? Wondering about how useful it would be for running an external inverter off the 12 dc/dc converter without having to put the car in ready mode. It is possible to trigger by starting then immediately stopping a charge session but the car then turns off after 5 minutes.

bohaman1, 2025-01-05:
Hello, may I ask if anyone have any luck with 2015-2016 leaf to get this feature working ? Thank you for your time.

mjkapkan, 2025-01-19:
I no longer have access to Nissan Leaf. So can't test anymore. :( But would gladly help any other way I can.

rileyg98, 2025-03-13:
Quite interested in this as well, for similar reasons as to @dhi20 - running an inverter off the 12v DC/DC. #671 implemented the wakeup for EVSE with a CAN write 0x679 with 1 byte of data for a ZE0 at least - seems this is sent by the OBC when a J1772 is inserted. It seems to activate the DC-DC on the climate control command, and it disengages relatively quickly after climate control is disabled. 

It looks like the VCM handles DC-DC over connector E62 wired to F11 on the junction box. It may be entirely controlled by the VCM.