openvehicles/Open-Vehicle-Monitoring-System-3 #473: incorrectly shows as "Not Charging" when it is charging (i-MiEV)

jjlink, 2020-11-23:
I verified the SOC is going up so the vehicle is charging; however both the web page and the Android app indicate its not charging most of the time. Using the web interface it did show charging earlier today, so its seems to be sporadic. I monitored it using MQTT explorer and it showed charging at that same time when the web interface showed charging.

jjlink, 2020-11-23:
These MQTT records are from earlier today when it was working:
ovms/jjlink/OVMS-PLUGNEV2/notify/info/charge.started
 - Charging
0.0V/0.0A
SOC: 88.5%
Ideal range: 49km
Est. range: 38km
ODO: 101856.0km
CAC: 27.5Ah
SOH: 57%
AC charge: 0.001kWh kWh
DC charge: 0.001kWh kWh
 - Charging
0.0V/0.0A
SOC: 88.5%
Ideal range: 49km
Est. range: 38km
ODO: 101856.0km
CAC: 27.5Ah
SOH: 57%
AC charge: 0.001kWh kWh
DC charge: 0.001kWh kWh

ovms/jjlink/OVMS-PLUGNEV2/notify/alert/charge.stopped
Not charging
SOC: 90.0%
Ideal range: 50km
Est. range: 39km
ODO: 101856.0km
CAC: 27.5Ah
SOH: 57%
AC charge: 0.194kWh kWh
DC charge: 0.111kWh kWh

KommyKT, 2020-11-26:
Hi jjlink!
Please,  try this version of fw: kommykt-nas.ddns.net:22289/dev/test/ovms3.bin 
I modified a bit the "charging detection"
This is a public charger or a home charger?

jjlink, 2020-11-26:
This is a home charger (OpenEVSE). That seems to have got it working again. Thanks!

Type1 - Charging
253.0V/5.9A
SOC: 78.5%
Ideal range: 44km
Est. range: 34km
ODO: 101861.0km
CAC: 27.5Ah
SOH: 57%
AC charge: 0.001kWh kWh
DC charge: 0.000kWh kWh

jjlink, 2020-11-26:
The BMS cell monitor page is blank. 

Also the am getting a bunch of these alerts:
Voltage: SD=7mV OK
Temperature: SD=1.4C !47:+3.1C

jjlink, 2020-11-26:
I just tried charging again and now it not working, so its only works sometimes.

Not charging
SOC: 82.0%
Ideal range: 46km
Est. range: 35km
ODO: 101861.0km
CAC: 27.5Ah
SOH: 57%
AC charge: 0.427kWh kWh
DC charge: 0.282kWh kWh

KommyKT, 2020-11-27:
I have a 2012 Peugeot iON and OpenEVSE. But I have no problem with charge detection, only problem the previous version if crash the module while charging not showing the type of charging, now this is fixed.
Next charge please, send the following output result, if show the charging and if not: 
me li v.c
me li xmi.c

jjlink, 2020-11-30:
This time it shows charging..

I will try it again later and see if it works a second time (last time it didn't).

[image]

The web interface:
[OVMS Status.pdf](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/5618504/OVMS.Status.pdf)

[image]

OpenEVSE:
[image]

jjlink, 2020-11-30:
Ok, a little later after I stopped the previous session for like a half hour or so and then started it again. Now this time it says not charging:
[image]
[image]
[image]
[image]

KommyKT, 2020-12-01:
- The BMS cell monitor page is blank.
     - Please go to TRIO / Settings and  hit Save button, then start the car, I think this fix this.
- Also the am getting a bunch of these alerts:
   Voltage: SD=7mV OK
   Temperature: SD=1.4C !47:+3.1C
     -  You can edit notifications in BMS Cell page in Alert Config

For charge detection OVMS use the 286 pid, please set up [retools plugin ](https://docs.openvehicles.com/en/latest/plugin/retools/README.html)
and start it and in search type 286, and send a screenshot if charge not detected

KommyKT, 2020-12-21:
Nothing?
Really hard to help, if I can not get data, because my car charge sensing is works 100%. Now I fix some small bug.
Now I changed the test version to my last version: kommykt-nas.ddns.net:22289/dev/test/ovms3.bin

jjlink, 2020-12-21:
Yes, I am looking at setting up the re tool plugin right now. Thanks

jjlink, 2020-12-21:
I have the retool running. I did a capture while doing a brief charge. It did say charging as usual the on the first try.
How do I see the detail records for a single PID (like 286)? I don't see the instructions in the developer docs.
[image]

KommyKT, 2020-12-30:
Please create a log file with charge detected, and one without detect, if you can.
I think the simplest method create a folder on SD card with Tools/Editor named "logs"
then Tools/Shell and type:
can log start vfs crtd /sd/logs/charge.csv 
This create a log file to sd card logs folder with charge.csv name run about 1-2 minutes then type the following command:
can log stop
Now open another tab and open: ip_of_ovms/logs
and download the charge.csv file, and send for my

my email is my name and gmail.com

jjlink, 2021-01-03:
I just tried doing a charging log and OVMS crashed in the middle of it. So I got no log file. I will try again...

jjlink, 2021-04-04:
Seems to be working now.

closed 2021-04-04 by jjlink