openvehicles/Open-Vehicle-Monitoring-System-3 #475: OVMS only connects to the first configured wifi AP

speculatrix, 2020-11-28:
**Failing Scenario**
When vehicle is at home, it connects to my home wifi. When I leave home, OVMS should connect to my phone;s hotspot which will come on automatically.

**Reproducing**
Define two or more APs for client mode.
Reboot the OVMS unit and observe that it connects to the first AP.
Turn off the first AP or move out of its range.
Observe that the OVMS unit doesn't connect to the 2nd AP.
Reboot the OVMS unit when only the 2nd AP and notice that it still doesn't connect.

**Further Checks**
Swap the wireless client settings, and check that it will connect to what is now the first AP.
Turn off the newly first AP and observe it will not connect to the newly second AP.
Reboot the OVMS unit just to see if it's a startup fault, it still doesn't connect.

**Expectation**
The OVMS unit will connect to any of the configured APs.

dexterbg, 2020-11-30:
Paul, did you set your unit to scanning client mode?
- Config → Autostart → … client mode SSID: Any known SSID (scan mode)

speculatrix, 2020-12-02:
I hadn't got that turned on, so I turned it on. Didn't help. Tried rebooting a few times and OVMS unit didn't connect to my phone's wifi. I'll do a little more testing but I think it doesn't work.

My module info:

Firmware: 3.2.015-157-g489b2570/ota_1/main (build idf v3.3.4-845-gd59ed8bba Nov 18 2020 14:23:19)
Running partition: ota_1
Boot partition: ota_1
Factory image: 3.2.008
OTA_O image: 3.2.015
OTA_1 image: 3.2.015-157-g489b2570

speculatrix, 2020-12-02:
[image]

dexterbg, 2020-12-05:
You're the only one having this issue, so it's most probably in your configuration. My module perfectly switches between the defined Wifi client networks as necessary, also to & from my phone's hotspot (Android).

Set your logging to level verbose for tags `esp32wifi` and `events` to follow the network transitions and scans in your log. Example of a network transition:

a) Connection to home station lost:

```
2020-12-05 12:25:50 CET D (951418344) events: Signal(network.wifi.sta.bad)
2020-12-05 12:25:50 CET I (951418344) netmanager: WIFI client has bad signal quality (-89.1 dBm); disconnect
2020-12-05 12:25:50 CET I (951418354) netmanager: Interface priority is pp3 (10.149.26.113/255.255.255.255 gateway 10.64.64.64)
2020-12-05 12:25:50 CET I (951418354) netmanager: Set DNS#2 0.0.0.0
2020-12-05 12:25:50 CET I (951418354) netmanager: WIFI client down (with MODEM up): reconfigured for MODEM priority
2020-12-05 12:25:50 CET D (951418354) events: Signal(network.wifi.down)
2020-12-05 12:25:50 CET D (951418364) events: Signal(network.reconfigured)
2020-12-05 12:25:50 CET I (951418364) time: Network was reconfigured: restarting SNTP client
2020-12-05 12:25:50 CET D (951418374) events: Signal(network.interface.change)
2020-12-05 12:25:50 CET D (951418374) events: Signal(server.v2.connectwait)
2020-12-05 12:25:50 CET D (951418584) events: Signal(server.v2.waitreconnect)
2020-12-05 12:25:56 CET I (951424114) wifi:2020-12-05 12:25:56 CET bcn_timout,ap_probe_send_start2020-12-05 12:25:56 CET 
2020-12-05 12:25:58 CET I (951426624) wifi:2020-12-05 12:25:58 CET ap_probe_send over, resett wifi status to disassoc2020-12-05 12:25:58 CET 
2020-12-05 12:25:58 CET I (951426624) wifi:2020-12-05 12:25:58 CET state: run -> init (c800)2020-12-05 12:25:58 CET 
2020-12-05 12:25:58 CET I (951426624) wifi:2020-12-05 12:25:58 CET pm stop, total sleep time: 1093046374 us / 2030636014 us
2020-12-05 12:25:58 CET 
2020-12-05 12:25:58 CET I (951426634) wifi:2020-12-05 12:25:58 CET new:<6,0>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:12020-12-05 12:25:59 CET 
2020-12-05 12:25:59 CET D (951426644) events: Signal(system.event)
2020-12-05 12:25:59 CET D (951426654) events: Signal(system.wifi.sta.disconnected)
2020-12-05 12:25:59 CET I (951426654) netmanager: WIFI client stop
2020-12-05 12:25:59 CET I (951426654) netmanager: Set DNS#2 0.0.0.0
2020-12-05 12:25:59 CET I (951426654) esp32wifi: STA disconnected with reason 200 = BEACON_TIMEOUT
2020-12-05 12:25:59 CET D (951426684) events: Signal(server.web.socket.closed)
```

b) Scan for available networks:

```
2020-12-05 12:25:59 CET V (951427344) esp32wifi: StartConnect: scan started
2020-12-05 12:26:00 CET D (951428344) events: Signal(clock.1226)
2020-12-05 12:26:02 CET D (951429794) events: Signal(system.event)
2020-12-05 12:26:02 CET D (951429794) events: Signal(system.wifi.scan.done)
2020-12-05 12:26:02 CET V (951429804) esp32wifi: ScanDone: #01 ssid='DextersPhone' bssid='a4:71:74:83:65:7f' chan=6 rssi=-63
2020-12-05 12:26:02 CET V (951429804) esp32wifi: ScanDone: #02 ssid='Telekom_FON' bssid='78:dd:12:09:dc:be' chan=1 rssi=-87
2020-12-05 12:26:02 CET V (951429804) esp32wifi: ScanDone: #03 ssid='WLAN-184248' bssid='78:dd:12:09:dc:bc' chan=1 rssi=-88
2020-12-05 12:26:02 CET I (951429804) esp32wifi: ScanDone: connect to ssid='DextersPhone' bssid='a4:71:74:83:65:7f' chan=6 rssi=-63
```

c) Connect to found station (my phone):

```
2020-12-05 12:26:02 CET I (951429924) wifi:2020-12-05 12:26:02 CET new:<6,2>, old:<6,0>, ap:<6,2>, sta:<6,2>, prof:12020-12-05 12:26:02 CET 
2020-12-05 12:26:02 CET I (951429924) wifi:2020-12-05 12:26:02 CET state: init -> auth (b0)2020-12-05 12:26:02 CET 
2020-12-05 12:26:02 CET I (951429934) wifi:2020-12-05 12:26:02 CET state: auth -> assoc (0)2020-12-05 12:26:02 CET 
2020-12-05 12:26:02 CET I (951429944) wifi:2020-12-05 12:26:02 CET state: assoc -> run (10)2020-12-05 12:26:02 CET 
2020-12-05 12:26:02 CET I (951429974) wifi:2020-12-05 12:26:02 CET connected with DextersPhone, aid = 1, channel 6, 40D, bssid = a4:71:74:83:65:7f2020-12-05 12:26:02 CET 
2020-12-05 12:26:02 CET I (951429974) wifi:2020-12-05 12:26:02 CET security type: 3, phy: bgn, rssi: -572020-12-05 12:26:02 CET 
2020-12-05 12:26:02 CET I (951429974) wifi:2020-12-05 12:26:02 CET pm start, type: 1
2020-12-05 12:26:02 CET 
2020-12-05 12:26:02 CET D (951429984) events: Signal(system.event)
2020-12-05 12:26:02 CET D (951429984) events: Signal(system.wifi.sta.connected)
2020-12-05 12:26:02 CET I (951429984) netmanager: Set DNS#2 0.0.0.0
2020-12-05 12:26:02 CET I (951429984) esp32wifi: STA connected with SSID: DextersPhone, BSSID: a4:71:74:83:65:7f, Channel: 6, Auth: WPA2
2020-12-05 12:26:02 CET I (951430024) event: sta ip: 192.168.43.182, mask: 255.255.255.0, gw: 192.168.43.1
2020-12-05 12:26:02 CET D (951430024) events: Signal(system.event)
2020-12-05 12:26:02 CET D (951430034) events: Signal(network.interface.up)
2020-12-05 12:26:02 CET D (951430034) events: Signal(system.wifi.sta.gotip)
2020-12-05 12:26:02 CET I (951430034) netmanager: WIFI client got IP
2020-12-05 12:26:02 CET I (951430034) netmanager: Set DNS#2 0.0.0.0
2020-12-05 12:26:02 CET I (951430034) esp32wifi: STA got IP with SSID 'DextersPhone' AP a4:71:74:83:65:7f: MAC: 30:ae:a4:5f:e7:ec, IP: 192.168.43.182, mask: 255.255.255.0, gw: 192.168.43.1
2020-12-05 12:26:02 CET D (951430044) events: Signal(network.wifi.sta.bad)
[... cut ...]
2020-12-05 12:26:03 CET D (951431364) events: Signal(network.interface.change)
```

d) Reconnect to the server:

```
2020-12-05 12:26:03 CET D (951431374) events: Signal(server.v2.connectwait)
2020-12-05 12:26:13 CET D (951441464) events: Signal(server.v2.connecting)
2020-12-05 12:26:13 CET D (951441554) events: Signal(server.v2.authenticating)
2020-12-05 12:26:13 CET D (951441614) events: Signal(server.v2.connected)
2020-12-05 12:26:13 CET D (951441624) events: Signal(app.connected)
```

Regards,
Michael

speculatrix, 2020-12-14:
I'm not sure whether it was me making a mistake or something else, but my OVMS unit now connects find to my phone's hotspot. thanks for the help!

Am still trying to persuade Tasker to automatically switch on the hotspot when the phone connects to the MG's bluetooth, but that's another issue, I suspect power saving is making Tasker miss the event.

closed 2020-12-14 by speculatrix

speculatrix, 2020-12-16:
@dexterbg my phone seems to be happy to keep wifi running *and* be a hotspot, so I made the wifi client profile on my phone not automatically connect to the OVMS device.