openvehicles/Open-Vehicle-Monitoring-System-3 #476: NL: Shows vehicle plugged in even when not

dalathegreat, 2020-11-30:
I have two LEAFs, one 2012 and one 2015

The 2012 gets reported as being plugged in all the time, and the option to swipe to start charging in the app is always visible. On the 2015 this works as intended, and the display to charge is only visible when the vehicle is actually plugged in.

Has this worked before with the 2012? Old or new issue?

glynhudson, 2020-11-30:
Yes, this is an issue, My 2012 leaf always shows on OVMS that is plugged in when it's not. The plugged in state works correctly on 2014+ models

dalathegreat, 2020-11-30:
By quickly looking at the code, in the main switch case for CAN messages there is one for AZE0 (case 0x390:), but ZE0 (0x380) is missing!

drc38, 2020-12-08:
For the app to work properly on a gen1 you'd need to include StandardMetrics.ms_v_charge_pilot->SetValue(false) either based on the existing charger state logic or a can message that corresponds to pilot/voltage presence

jabelone, 2021-04-11:
I can confirm this has recently started happening for me. I'm fairly certain that it used to work so I presume that a recent OTA update is causing the OVMS to incorrectly report the connector as plugged in. This has given me a heart attack a few times as I thought I left the car plugged in to our apartment's communal charger when I wasn't meant to be using it. 😅

@drc38 can you please explain a little bit more about what you mean? Happy to look into this and/or test as I have a little bit of C++/Arduino experience. I'll see if I can get a more specific (and reproducible) test case other than "sometimes it reports as plugged in when it's not".

drc38, 2021-04-12:
@jabelone on the older Leafs 0x380 is the CAN message with the charger state details encoded within it. If you log that CAN message while plugging in connector, charging, disconnecting you should be able to decode it and identify various charge state "cases", similar to those used in the 0x390 logic for AZE0 models, then set the correct pilot state in OVMS in the 0x380 CAN message code.

drc38, 2021-04-12:
Fyi, looks like the existing code is based on the 0x5bf CAN message rather than 0x380 for Gen1 Leafs

jabelone, 2021-04-12:
cheers, thanks for the heads up.

dalathegreat, 2021-12-22:
Pull request is up for fixing this: https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/661

dexterbg, 2021-12-24:
PR #661 merged, so closing this.

closed 2021-12-24 by dexterbg