openvehicles/Open-Vehicle-Monitoring-System-3 #483: NL: Strange push notification, voltage diff?

dalathegreat, 2020-12-09:
[image]

So, I consider myself a pretty tech-savvy user, but I can't figure out what this event message is supposed to relay. Is there an issue? 13mV diff seems fine to me?

dexterbg, 2020-12-09:
That's the BMS cell alert: Standard voltage deviation is 13 mV, but cell #93 has a deviation of -31 mV. That's an indication something's wrong/bad with that cell.

If you think that deviation is still OK, you can raise your alert thresholds using the battery monitoring UI or by setting the config parameters.

The message text could be a bit more descriptive. It's origin was the V2 SMS notification, and it needed to cover possibly a range of cell alerts so was optimized for small text lengths. We could at least add a "BMS alert" prefix… and should explain the message in the user documentation.

dalathegreat, 2020-12-09:
Ok! For sure the message should be improved, displaying "!93:-31mV" is not so user friendly :D

I suggest a general "Cell voltage imbalance warning" to start with? And then split it up so that it displays "Cell #93 -31mV"

dexterbg, 2020-12-09:
New example alert:

```
Battery cell deviation:
Voltage: StdDev 32mV
 ! #14: -112mV
Temperature: StdDev 1.5C
 ! #03: +3.4C
Details: bms status
```

closed 2020-12-09 by dexterbg