openvehicles/Open-Vehicle-Monitoring-System-3 #484: NL: Enabling CAN writes on 2012 model set off alarm

glynhudson, 2020-12-09:
When CAN writes is enabled on my 2012 Nissan LEAF the alarm will randomly trigger, often in the middle of the night! Disabling CAN writes fixed this. However, on my 2014/2015 LEAF enabling CAN writes works fine and does not trigger the alarm. I checked the the model year is correctly set in OVMS. 

As anyone else experiance this issue?

dalathegreat, 2021-01-09:
Does not happen on my 2012. But I haven't done the remote heater (Wire RC3 to TCU pin 11) yet, have you?

glynhudson, 2021-03-25:
No mods, just plugged into ODB2 port. Turning on remote climate works when the vehicle is aware e.g charging. But I've had to turn off CAN writes, my neighbours didn't appreciate the alarm going off randomly in the night! Another issue I've noticed with the 2012 leaf is that OVMS always reports that the vehicle is plugged in when it's not. Do you experience the same issue with your 2012?

glynhudson, 2021-03-25:
Ah yes, I see you do https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/476

closed 2022-11-04 by glynhudson