openvehicles/Open-Vehicle-Monitoring-System-3 #493: Kia Soul 30kWh problems

mnmn2, 2020-12-15:
Hello,
I realized, that the SOH reading is getting false info many times...
At the afternoon it shows 110% (it's a valid value, because the newest BMS fw "unleasehed" the safety margins of the battery pack), then after an automatic fw update, it shows 60% in the morning.

[image]

Additionally, the battery pack of the 30kWh version contains 100 cells. But the BMS cell monitor page shows only 96.
[image]

dalathegreat, 2021-01-14:
To fix the BMS showing all 100 cells, this file needs to be changed
\Open-Vehicle-Monitoring-System-3-master\vehicle\OVMS.V3\components\vehicle_kiasoulev\src\vehicle_kiasoulev.cpp:
  300:   `BmsSetCellArrangementVoltage(96, 1);`
  
However, how should we determine which 30kWh generation that is present.

MY 2016-2017 | MY 2018-2019
30,5 kWh (96s2p) | 31,8 kWh (100s2p) 

Based on model year?

mnmn2, 2021-01-14:
Hello, 
thank you for your help (in the future also :) )

Yes, the MY18 cars have the 30kWH pack, and I think we must accept the "If user then sets this to bigger value, we can assume that the MY2018 is there" solution.

Unfortunately the VIN number (and the production year) is not enough to decide, because there are many MY2018 cars, which were manufactured in 2017 (mine was on 2017-03-21), but I know many MY17 cars (with 27kWh battery), which also were manufactured in 2017.
There are online VIN decoder sites, which can give back to you a very lengthy list of the equipments and options of the car, and there is the option "3713A1 BATTERY CAPACITY HV TYPE - BATTERY HV EX", which clearly means, that this car has a 30kWh battery. But this is quite hard to use services like these, to make the decision automatic.

I think your idea is much better. If a user set the battery to 30000 then it's a MY18, with 30kWh battery.

dalathegreat, 2021-01-15:
I made this PR: https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/532 lets continue the discussion there for getting 100S fixed

dalathegreat, 2021-01-18:
Here is how SOH currently is calculated: 
StdMetrics.ms_v_bat_soh->SetValue( 110 - ( m_b_cell_det_max->AsFloat(0) + m_b_cell_det_min->AsFloat(0)) / 2 );

Could you look into the values of 
m_b_cell_det_max and m_b_cell_det_min?