openvehicles/Open-Vehicle-Monitoring-System-3 #494: Incorrect 'modem connection' message when using WiFi

glynhudson, 2020-12-16:
While updating the FW on OVMS via WiFi network (see local IP address in address bar) I get a message stating that OVMS is using cellular data. I think this message is incorrect since the unit is connected to my WiFi 

[image]

markwj, 2020-12-17:
Can you show the output of 'network status' at the time that this is shown. Also what is 'config show ota' showing?

glynhudson, 2020-12-17:
I'm now testing on a different vehicle with a different module, the result is the same however the local IP address for this module is 192.168.86.195

[image]

```
OVMS# network status
Interface#7: st7 (ifup=1 linkup=1)
  IPv4: 192.168.86.195/255.255.255.0 gateway 192.168.86.1

Interface#3: pp3 (ifup=1 linkup=1)
  IPv4: 10.52.133.152/255.255.255.255 gateway 10.64.64.64

Interface#2: ap2 (ifup=1 linkup=1)
  IPv4: 192.168.4.1/255.255.255.0 gateway 192.168.4.1

DNS: 212.9.0.135 212.9.0.136

Default Interface: pp3 (10.52.133.152/255.255.255.255 gateway 10.64.64.64)
```

```
OVMS# config list ota
ota (readable writeable)
  auto.allow.modem: no
  auto.hour: 2
  http.mru: api.openvehicles.com/firmware/ota/v3.1/eap/ovms3.bin
  server: api.openvehicles.com/firmware/ota
  tag: eap
```

```
OVMS# simcom
Network Registration: RegisteredRoaming
Provider: 3 UK Hologram
Signal: -91 dBm

State: NetMode

PPP: Connected on channel: #2

GPS: Connected on channel: #1
```

dexterbg, 2020-12-17:
Glyn, check your Wifi signal quality when getting this, it's probably due to the Wifi signal not being good enough for the module to use Wifi as the _default_ network. You may tune the thresholds, see https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/commit/057f14442ba96c2711d1887a5da2faf85b0f2517

glynhudson, 2021-01-03:
Good point, the WiFi signal is low (-88dBm). However, it's still a solid connection which is still faster (and cheaper!) than GSM. Should OVMS not use WiFi for the update if WiFi is connected?

dexterbg, 2021-01-03:
In my experience, -88 is exactly in the unstable / tipping region, i.e. can become unusable any moment. That's why I introduced the whole signal quality assessment and set the default thresholds at -87 & -89 dBm.

But of course, your mileage may vary, as Wifi stability depends on a number of environmental influences. That's why you can configure the thresholds to suit your needs.

dexterbg, 2021-01-03:
To be precise: in your case, your signal must already have been below -89 dBm before to let the netmanager switch to modem.

IOW: if you know your network still is reliable at, say, -92 dBm, try something like…
```
config set network wifi.sq.good -91
config set network wifi.sq.bad -93
```

closed 2021-01-03 by glynhudson