openvehicles/Open-Vehicle-Monitoring-System-3 #524: Fetching files using scp can crash logging it seems

elbow, 2021-01-10:
Hi,

I've been finding my sd card logging stopping.  For instance now my SD card "log" file has nothing written since 50 minutes.

It looks to me like a task quit.  Here's what I see running now:

```
Number of Tasks = 18        Stack:  Now   Max Total    Heap 32-bit SPIRAM C# PRI CPU% BPR/MH
3FFAFB88    1 Blk esp_timer         436   708  4096   41176    644  33480  0  22   0%  22/ 0
3FFBEE60    2 Blk eventTask         476  2060  4608     104      0      0  0  20   0%  20/ 0
3FFC1424    3 Blk OVMS Events       496  3456  8192   71576      0  74032  1   8   0%   8/ 0
3FFC4A54    4 Blk OVMS DukTape      496 10864 12288     580      0 112632  1   3   1%   3/ 0
3FFC7714    5 Blk OVMS CanRx        480  1904  4096    3240      0  37816  0  23   0%  23/ 0
3FFC8510    6 Blk ipc0              428   508  1024    7804      0      0  0  24   0%  24/ 0
3FFC8B14    7 Blk ipc1              432   528  1024     120      0      0  1  24   0%  24/ 0
3FFCA984   10 Rdy IDLE0             416   512  1024       0      0      0  0   0  94%   0/ 0
3FFCAF1C   11 Rdy IDLE1             408   584  1024       0      0      0  1   0  92%   0/ 0
3FFCBCB4   12 Blk Tmr Svc           384  1408  3072     124      0      0  0  20   0%  20/ 0
3FFC8EF4   17 Blk tiT               536  2344  3072     928      0  10224  1  18   2%  18/ 0
3FFD0CFC   20 Blk OVMS SIMCOM       632  2440  4096    6100      0   3732  0  20   0%  20/ 0
3FFDE794   22 Blk OVMS Vehicle      488  2392  8192     648      0     76  1  10   0%  10/ 0
3FFE2C90   23 Blk OVMS Console      584  1704  6144       0      0     20  1   5   0%   5/ 0
3FFE36C8   25 Blk mdns              512  1408  4096     108      0      4  0   1   0%   1/ 0
3FFD53AC   35 Blk wifi              468  2708  3584   36124      0   5800  0  22   5%  22/ 1
3FFD2CE4   37 Blk OVMS NetMan       912  4816  8192    8300      0   8088  1   5   3%   5/ 0
3FFDC4EC   94 Rdy module tasks      812  2060  6144     228   3608  24548  1   4   0%   4/ 1
```

It does look like the FileLog task is missing.  Here's how that looks after a reboot:

```
3FFEC148   26 Rdy OVMS FileLog      564  1572  3072      36      0      0  1   2   0%   2/ 0
```

Around the time that the logging quit I was busy scp-ing some log files off the OVMS via wifi.  

There's nothing in the sd card log showing what happened (presumably because the logging stopped...).

dexterbg, 2021-01-10:
I assume your scp load was a bit too much for the file system. If the file logger runs into a file error, it can only terminate. It will output a final error message, but you need to be monitoring the volatile log to see that, i.e. have some shell open with logging enabled.

The file system has some issues. We've been hoping to see these get fixed since a few esp-idf releases…

dexterbg, 2021-01-10:
If you can reproduce that: it would be an option to add a watchdog on a ticker, check if the log task should be active but isn't, restart it if necessary.