openvehicles/Open-Vehicle-Monitoring-System-3 #531: Repeating crash - abort() called looks like freertos

elbow, 2021-01-15:
Hi,

In a period of 30 minutes in the car I had 4 OVMS restarts:

```
+---------------------+------------------+--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------+
| h_timestamp         | h_recordtype     | h_data                                                                                                                                                                                                                               |
+---------------------+------------------+--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------+
| 2021-01-15 09:24:51 | *-OVM-DebugCrash | 3.2.015-437-g56ef187d-dirty/ota_0/main (build idf v3.3.4-845-gd59ed8bba-dirty Jan 12 2021 17:13:39),5,Crash,12,12,2,0,abort(),0,,0x4008e57e 0x4008e819 0x400e7558 0x4008405a ,6,Task watchdog,,,0,IDLE1                              |
| 2021-01-15 09:38:50 | *-OVM-DebugCrash | 3.2.015-437-g56ef187d-dirty/ota_0/main (build idf v3.3.4-845-gd59ed8bba-dirty Jan 12 2021 17:13:39),6,Crash,12,12,3,0,abort(),0,,0x4008e57e 0x4008e819 0x400e7558 0x4008405a ,6,Task watchdog,,,0,OVMS DukTape                       |
| 2021-01-15 09:44:25 | *-OVM-DebugCrash | 3.2.015-437-g56ef187d-dirty/ota_0/main (build idf v3.3.4-845-gd59ed8bba-dirty Jan 12 2021 17:13:39),7,Crash,12,12,4,0,abort(),0,,0x4008e57e 0x4008e819 0x400e7558 0x4008405a ,6,Task watchdog,ticker.1,ovms-server-v3,0,OVMS DukTape |
| 2021-01-15 10:01:32 | *-OVM-DebugCrash | 3.2.015-437-g56ef187d-dirty/ota_0/main (build idf v3.3.4-845-gd59ed8bba-dirty Jan 12 2021 17:13:39),8,Crash,12,12,5,0,abort(),0,,0x4008e57e 0x4008e819 0x400e7558 0x4008405a ,6,Task watchdog,,,0,IDLE1                              |
+---------------------+------------------+--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------+
```

As you can see they are all the same - the backtrace comes out like so:

```
steve@maximus support % ./a2l 0x4008e57e 0x4008e819 0x400e7558 0x4008405a
Using elf file: /Users/steve/OVMS/OVMS-elbow/vehicle/OVMS.V3/build/ovms3.elf
0x4008e57e is in invoke_abort (/Users/steve/OVMS/esp/esp-idf/components/esp32/panic.c:156).
151	#endif
152	    while (1) {
153	        if (esp_cpu_in_ocd_debug_mode()) {
154	            __asm__ ("break 0,0");
155	        }
156	        *((int *) 0) = 0;
157	    }
158	}
159
160	void abort()
0x4008e819 is in abort (/Users/steve/OVMS/esp/esp-idf/components/esp32/panic.c:171).
166	     * don't overwrite that.
167	     */
168	    if (esp_reset_reason_get_hint() == ESP_RST_UNKNOWN) {
169	        esp_reset_reason_set_hint(ESP_RST_PANIC);
170	    }
171	    invoke_abort();
172	}
173
174
175	static const char *edesc[] = {
0x400e7558 is in task_wdt_isr (/Users/steve/OVMS/esp/esp-idf/components/esp32/task_wdt.c:174).
169	    esp_task_wdt_isr_user_handler();
170	    if (twdt_config->panic){     //Trigger Panic if configured to do so
171	        ESP_EARLY_LOGE(TAG, "Aborting.");
172	        portEXIT_CRITICAL_ISR(&twdt_spinlock);
173	        esp_reset_reason_set_hint(ESP_RST_TASK_WDT);
174	        abort();
175	    }
176
177	    portEXIT_CRITICAL_ISR(&twdt_spinlock);
178	}
0x4008405a is at /Users/steve/OVMS/esp/esp-idf/components/freertos/xtensa_vectors.S:1154.
1149	    wsr     a0, PS
1150	    rsync
1151
1152	    /* OK to call C code at this point, dispatch user ISRs */
1153
1154	    dispatch_c_isr 1 XCHAL_INTLEVEL1_MASK
1155
1156	    /* Done handling interrupts, transfer control to OS */
1157	    call0   XT_RTOS_INT_EXIT                /* does not return directly here */
1158
```

task_wdt.c code is "task_wdt_isr".  It called abort() - apparently because "Watchdog got triggered because at least one task did not reset in time.".

I'm not sure where to find the log messages written from task_wdt_isr which will presumably tell me which was the offending task.

Steve

elbow, 2021-01-15:
I found issue #418 which is also a task watchdog reset.

I'm trying to run the ABRP script with Duktape and my guess is that that is what is destabilising my setup.

Steve

dexterbg, 2021-01-15:
Steve, we've had some discussion on the watchdog issue on the developer list during the last months, check out the list archive.

The watchdog is triggered if a task (or combination of tasks) hogs the CPU, so the idle task starves. I haven't had this on my module for months, but you can still see these in the crash report on my server (again, see list archive).

If you can nail this down, that would be very helpful.

dexterbg, 2021-12-30:
As the crash report shows, this isn't solved yet:
- https://dexters-web.de/f/firmware/developer/crashreport.csv

It seems to affect all vehicles and doesn't seem to have changed significantly over the releases. I wonder if we will see a difference with ESP32 revision 3… oh, we need a hardware info in the crash report now, I'll take care of that.

dexterbg, 2021-12-30:
Firmware & crash debug data now include the hardware info so we'll be able to do some statistics here as soon as the OVMS v3.3 has been rolled out.

dteirney, 2022-04-21:
Looks like a later version of the ABRP script is now located at https://github.com/iternio/ovms-link/blob/dev/abrp.js for which v1.4 resolves an issue with crashing, but which isn't referenced in the official docs (that still point to the source for the old 1.3 version).

pascalsaul, 2022-10-10:
I wonder if there is any update since this is still occurring?

dexterbg, 2022-10-10:
The crash reports show no clear circumstances. Only thing we can say now is, it's not related to the ESP32 version, so is probably a software bug. Unclear if in our application code or in the esp-idf libs.