openvehicles/Open-Vehicle-Monitoring-System-3 #544: NL: Metrics Not Updated on wakeup

mjkapkan, 2021-01-28:
I noticed that a lot of metrics on NL MY 2013 don't regular update and can remain unchanged for hours. While sending `wakeup` command does not make a difference, if I send `start charge` command (when vehicle is unplugged) it seems to update most of those metrics.

I would suggest to add the CAN frame that is sent by start charge to actively poll metrics, or otherwise at least at it to the wakeup command.

Any thoughts?

dalathegreat, 2021-12-26:
Fix for this is pending in PR #662

dalathegreat, 2021-12-27:
It turned out this is not yet fixed completely. In my PR, I changed so that 0x679 is sent instead of 0x68c. This only worked when vehicle was plugged in and idle. So I'll need to do more testing with which commands are suitable for when not plugged in.

Sending 0x5c0 works, it is a bit strange, maybe the VCM listens for this message since it contains battery heating request? That might be why the vehicle wakes up when receiving it?

Bonus, the 0x679 command when plugged in and idle turns on the DC-DC converter and tops up the 12V battery, nice addition.

Thoughts @mjkapkan ?

mjkapkan, 2021-12-28:
I think we can just add `0x5c0` to the wakeup command. Does not seem to do any harm ;)

dalathegreat, 2022-01-03:
Fix for this is pending in PR https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/671

mjkapkan, 2022-01-05:
This is not yet fixed. At least for the MY 2013 when unplugged. Wakeup does not update v.e.cabintemp or v.e.temp. One trick that worked is to send 'charge start' command. So we might need to pick a couple of commands from that sequence.