openvehicles/Open-Vehicle-Monitoring-System-3 #569: Kia eSoul is showing wrong locking indicator

Sternwichtel, 2021-02-13:
I"m driving a Kia eSoul with the eNiro / Kona Konfiguration.
In the vehicle overview I noticed the locking indicator to be wrong.

If the vehicle is locked the indicator is showing an open padlock in red color.
If the vehicle is unlocked the indicator is showing a closed padlock in green color.
So while the correct color is shown the padlock symbol is inversed.
[image]

slcasner, 2021-02-13:
On Sat, 13 Feb 2021, Sternwichtel wrote:

> I"m driving a Kia eSoul with the eNiro / Kona Konfiguration.
> In the vehicle overview I noticed the locking indicator to be wrong.
>
> If the vehicle is locked the indicator is showing an open padlock [...]

I think the colors are as intended.  The open padlock is red as a
warning that the car is not locked.  The closed padlock is green
indicating safe.  That's how I see them for the Tesla Roadster, but of
course with the state of the padlock symbol matching the state of the
car.

                                                        -- Steve

glynhudson, 2021-10-03:
Is your Kia Soul a 2020 model?