openvehicles/Open-Vehicle-Monitoring-System-3 #581: MQTT -  lower interval for certain metrics

Wireheadbe, 2021-02-27:
Hi - this is more of a feature request, but would there be a way to lower the MQTT interval for the most important metrics: speed,SoC,lat/lon,power draw?

I'm using the MQTT implementation to integrate into Node-Red, and from there on, I send the data to ABetterRoutePlanner and Home-Assistant. 

I realize sending all metrics every minute might become a bandwidth hog, but some critical measurements could probably be sent every minute.

happyjaxx, 2021-02-27:
Hi,
nodered too ? ;-) (got some influxdb/grafana as well) I worked around an ugly hack with an openwrt device subscribing to the websocket to stream realtime data over an unlimited data 4G hotspot by reinjecting the metrics to a local mqtt broker bridged to the usual one.

But I'd really like a setting to change the interval. Global, and/or per metric.

I have a pretty cheap simcard for data in the ovms and don't mind sending more.
But the setting could be changed on a wifi.connected event to change the rate if the 4G hotspotis in range

happyjaxx, 2021-03-05:
Looks like the latest release partially answers the request, that is, if you can cope with less essential metrics being sent more often with it:

3.2.016:

    b) New config settings:
         "updatetime.on" which sets updatetime when the car is on (driving)
         "updatetime.charging" which sets updatetime when the car is charging
         "updatetime.awake" which sets updatetime when the car is awake
       If unset the server uses the "idle" or "connected" interval as before
    c) New config setting "updatetime.sendall". If you set this we will publish
       _all_ valid metrics at this interval. Helpful to keep a steady flow of
       values for charting.

I will definitly be setting this dynamically on wifi up/down events

vfs mkdir /store/events
vfs mkdir /store/events/network.wifi.down
vfs mkdir /store/events/network.wifi.up
vfs append 'config set server.v3 updatetime.on 30' /store/events/network.wifi.down/updateOnInterval
vfs append 'config set server.v3 updatetime.on 10' /store/events/network.wifi.up/updateOnInterval

(untested, might not be the good syntax)
JB./.

Wireheadbe, 2021-03-05:
That's an interesting approach! Very helpful!
So I guess putting updatetime.on to e.g. 60 would report the values every 60 seconds

Also -> On first connection to MQTT broker we used to send all metrics. Now we only send metrics for which we have a value; so would I be right in assuming that a lower value of updatetime.on would only send changed metrics a bit faster? That would be great!

Wireheadbe, 2021-03-05:
Trying to flash that version over wifi caused the wifi stack to crash, over modem went fine.

config set server.v3 updatetime.on 60 seems to do a great job!

markwj, 2021-03-07:
Please remember that the configuration store is in on-chip flash memory and that has a very limited number of write cycles (significantly less than SD card or other such storage). I would be wary of writing to that every time wifi goes up/down. Once the write cycle limit is hit, the flash starts to fail, and the whole module is useless.

Wireheadbe, 2021-03-07:
> Please remember that the configuration store is in on-chip flash memory and that has a very limited number of write cycles (significantly less than SD card or other such storage). I would be wary of [...]

Good catch indeed. I only set "config set server.v3 updatetime.on 60" - not via script - and it seems to behave great - disabled the V2 server as I don't need it, to save some data.

Wireheadbe, 2021-03-07:
So I guess, with 3.2.016 all is fine - closing!

closed 2021-03-07 by Wireheadbe

happyjaxx, 2021-03-07:
> Please remember that the configuration store is in on-chip flash memory and that has a very limited number of write cycles (significantly less than SD card or other such storage). I would be wary of [...]

Bummer... totally forgot about that... I'll drop the change then... Because of the 12V drain I unplug it most of the time, and on long drives, my messy setup is on with wifi, so a low value wouldn't bother in any case