openvehicles/Open-Vehicle-Monitoring-System-3 #584: OVMS misdetects the charge stop

jheredianet, 2021-03-09:
'm using the OVMS on a Hyundai Kona EV and noticed that OVMS module  misdetects the charge stop. I begins to charge up then a few seconds it is stopped, then a few seconds later continues charging, and of course I can see that in the car the charge is not stopping, it happens every time that it begins to charge, and also in a long charge session it randomly stops,   as you can see in the following log (full log attached as well):

2021-02-19 06:25:41 CET I (30189868) v-kianiroev: Charging starting
2021-02-19 06:25:41 CET D (30189878) websocket: UpdateTicker: can't lock client list, ticker run skipped
2021-02-19 06:25:41 CET D (30189878) events: Signal(vehicle.charge.prepare)
2021-02-19 06:25:41 CET I (30189878) ovms-server-v3: Tx event vehicle.charge.prepare
2021-02-19 06:25:41 CET D (30189888) events: Signal(vehicle.charge.state)
2021-02-19 06:25:41 CET D (30189978) events: Signal(vehicle.charge.start)
2021-02-19 06:25:41 CET I (30189978) ovms-server-v3: Tx event vehicle.charge.start
2021-02-19 06:25:41 CET D (30189978) events: Signal(vehicle.charge.12v.start)
2021-02-19 06:25:41 CET I (30189988) ovms-server-v3: Tx event vehicle.charge.12v.start
2021-02-19 06:25:41 CET D (30189988) events: Signal(vehicle.charge.pilot.on)
2021-02-19 06:25:41 CET I (30189988) ovms-server-v3: Tx event vehicle.charge.pilot.on
2021-02-19 06:25:42 CET I (30190868) powermgmt: Charging 12V battery..
2021-02-19 06:25:43 CET D (30191878) notify: Created entry type 'info' subtype 'charge.started' size 171 has 4 readers pending
2021-02-19 06:25:43 CET D (30191898) events: Signal(notify.info.charge.started)
2021-02-19 06:25:43 CET I (30191898) ovms-server-v3: Tx event notify.info.charge.started

2021-02-19 06:26:14 CET I (30222868) v-kianiroev: Charging done...
2021-02-19 06:26:14 CET D (30223058) events: Signal(vehicle.charge.finish)
2021-02-19 06:26:14 CET I (30223058) ovms-server-v3: Tx event vehicle.charge.finish
2021-02-19 06:26:14 CET D (30223138) events: Signal(vehicle.charge.stop)
2021-02-19 06:26:14 CET I (30223138) ovms-server-v3: Tx event vehicle.charge.stop
2021-02-19 06:26:14 CET D (30223148) events: Signal(vehicle.charge.12v.stop)
2021-02-19 06:26:14 CET I (30223148) ovms-server-v3: Tx event vehicle.charge.12v.stop
2021-02-19 06:26:14 CET D (30223158) events: Signal(vehicle.charge.pilot.off)
2021-02-19 06:26:14 CET I (30223158) ovms-server-v3: Tx event vehicle.charge.pilot.off
2021-02-19 06:26:15 CET I (30223868) powermgmt: No longer charging 12V battery..

2021-02-19 06:26:16 CET I (30224868) v-kianiroev: Charging starting
2021-02-19 06:26:16 CET D (30225048) events: Signal(vehicle.charge.start)
2021-02-19 06:26:16 CET I (30225048) ovms-server-v3: Tx event vehicle.charge.start
2021-02-19 06:26:16 CET D (30225048) events: Signal(vehicle.charge.12v.start)
2021-02-19 06:26:16 CET I (30225058) ovms-server-v3: Tx event vehicle.charge.12v.start
2021-02-19 06:26:16 CET D (30225058) events: Signal(vehicle.charge.pilot.on)
2021-02-19 06:26:16 CET I (30225058) ovms-server-v3: Tx event vehicle.charge.pilot.on
2021-02-19 06:26:17 CET I (30225868) powermgmt: Charging 12V battery..

This is causing that the notifications on the smartphone triggers several times this events
[ovmslogs.txt](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/6105126/ovmslogs.txt)

Zaphod-dev, 2021-12-27:
Same here, with a Twizy, notifications every minute that charge has stopped (when it hasn't). I've rebooted the OVMS module but to no avail. Nothing like this ever happened in the previous two years of OVMS daily use.

```
2021-12-27 17:32:29 | 2556 | A | Charge StoppedCHG: 4 (~5) WhFull: 503 min.Range: 24 - 36kmSOC: 54.3% (54.1..54.3%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:32:31 | 2557 | I | ChargingCHG: 4 (~5) WhFull: 503 min.Range: 24 - 36kmSOC: 54.3% (54.1..54.3%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:32:50 | 2558 | A | Charge StoppedCHG: 6 (~7) WhFull: 503 min.Range: 24 - 36kmSOC: 54.4% (54.1..54.4%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:32:51 | 2559 | I | ChargingCHG: 6 (~7) WhFull: 503 min.Range: 24 - 36kmSOC: 54.4% (54.1..54.4%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:33:10 | 2560 | A | Charge StoppedCHG: 8 (~9) WhFull: 503 min.Range: 24 - 37kmSOC: 54.4% (54.1..54.4%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:33:10 | 2561 | I | ChargingCHG: 8 (~9) WhFull: 503 min.Range: 24 - 37kmSOC: 54.4% (54.1..54.4%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:33:29 | 2562 | A | Charge StoppedCHG: 9 (~11) WhFull: 503 min.Range: 24 - 37kmSOC: 54.5% (54.1..54.5%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:33:31 | 2563 | I | ChargingCHG: 9 (~11) WhFull: 503 min.Range: 24 - 37kmSOC: 54.5% (54.1..54.5%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:33:50 | 2564 | A | Charge StoppedCHG: 11 (~12) WhFull: 503 min.Range: 24 - 37kmSOC: 54.5% (54.1..54.5%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:33:51 | 2565 | I | ChargingCHG: 11 (~12) WhFull: 503 min.Range: 24 - 37kmSOC: 54.5% (54.1..54.5%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:34:10 | 2566 | A | Charge StoppedCHG: 12 (~14) WhFull: 503 min.Range: 24 - 37kmSOC: 54.6% (54.1..54.6%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:34:10 | 2567 | I | ChargingCHG: 12 (~14) WhFull: 503 min.Range: 24 - 37kmSOC: 54.6% (54.1..54.6%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:34:29 | 2568 | A | Charge StoppedCHG: 14 (~16) WhFull: 497 min.Range: 24 - 37kmSOC: 54.6% (54.1..54.6%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:34:31 | 2569 | I | ChargingCHG: 14 (~16) WhFull: 497 min.Range: 24 - 37kmSOC: 54.6% (54.1..54.6%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:34:50 | 2570 | A | Charge StoppedCHG: 16 (~18) WhFull: 497 min.Range: 24 - 37kmSOC: 54.7% (54.1..54.7%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:34:51 | 2571 | I | ChargingCHG: 16 (~18) WhFull: 497 min.Range: 24 - 37kmSOC: 54.7% (54.1..54.7%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:35:10 | 2576 | A | Charge StoppedCHG: 17 (~20) WhFull: 497 min.Range: 24 - 37kmSOC: 54.7% (54.1..54.7%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:35:10 | 2577 | I | ChargingCHG: 17 (~20) WhFull: 497 min.Range: 24 - 37kmSOC: 54.7% (54.1..54.7%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:35:29 | 2578 | A | Charge StoppedCHG: 19 (~22) WhFull: 497 min.Range: 24 - 37kmSOC: 54.8% (54.1..54.8%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
2021-12-27 17:35:31 | 2579 | I | ChargingCHG: 19 (~22) WhFull: 497 min.Range: 24 - 37kmSOC: 54.8% (54.1..54.8%)ODO: 24941.8kmCAP: 85.1% 91.9AhSOH: 94%
```

dexterbg, 2021-12-27:
Please provide:
- a debug level module log excerpt
- …including a CAN log in crtd format
…covering 3-5 minutes of the effect.

Regards,
Michael

Zaphod-dev, 2021-12-27:
Hi Michael and thanks for the follow-up. I've done a cold restart and the notification surge has ceased. Next time it happens I will capture a debug level log output.