openvehicles/Open-Vehicle-Monitoring-System-3 #604: ipv6 Support

schlecht-network, 2021-04-06:
The whole Project is currently running on ipv4 only.
In view of the ipv4 exhaustion dual stack or even ipv6 only support would be nice.
(the only thing that really needs ipv6 is the client <=> server connection)

dexterbg, 2021-04-06:
Go ahead. AFAICT the IDF 3.3 LWIP version already includes IPv6 support.

leres, 2021-04-06:
I would like to see this happen and am willing and able to test changes.

stumbaumr, 2025-12-11:
I just got fiber at home and run Home Assistant with the Mosquitto AddOn to collect OVMS statistics for my smart EQ and enable evcc with the Home Assistant configuration option to wake the car up. The provider only gives out IPv6 and CGNAT IPv4 - and charges a fee for real IPv4.

Is that a valid use case to ask for the addition of IPv6?