openvehicles/Open-Vehicle-Monitoring-System-3 #607: The RC3 pin and 2012 Nissan LEAF Remote A/C

jabelone, 2021-04-11:
Hey everyone,

I am happy to update the documentation once I find an answer to this but am struggling to find answers. My 2012 Nissan LEAF remote climate control only works when charging (as expected). [The documentation](https://docs.openvehicles.com/en/latest/components/vehicle_nissanleaf/docs/index.html#models-ze0) suggests that you need to connect the "EV system activation request signal" (pin 11 on TCU connector) to "RC3" on the OVMS hardware. However, I'm struggling to find where "RC3" is on the OVMS hardware.

I dug through the code and found this section: https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/33bfce440fb0e419a23251c7ddf457c5d62cdd6f/vehicle/OVMS.V3/components/vehicle_nissanleaf/src/vehicle_nissanleaf.cpp#L1826

It appears the logic that triggers the RC3 pin is commented out, so it may not even work? Some clarification about whether or not this feature still works, and what pin to connect the TCU pin 11 to would be fantastic.

Thank-you for an awesome project btw!

dexterbg, 2021-04-11:
Jaimyn, there is no RC3 on the OVMS V3.

Read https://docs.openvehicles.com/en/latest/userguide/egpio.html on how the EGPIO works and can be used.

See https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/master/vehicle/OVMS.V3/components/vehicle_renaulttwizy/src/rt_ticker.cpp#L476 for an example on how to address the MAX7317 directly from the vehicle component.

Regards,
Michael

jabelone, 2021-04-11:
Hi @dexterbg,

Thank-you, that's very helpful. Do you know if there's any particular reason that the RC3 GPIO toggle was turned off or uncommented (lack of v3 hardware output)? It might be useful to know why, otherwise I may end up down a rabbit hole that has already been "solved".

Cheers.

slcasner, 2021-04-11:
On Sun, 11 Apr 2021, Jaimyn Mayer wrote:

> Thank-you, that's very helpful. Do you know if there's any particular reason that the RC3 GPIO toggle was turned off or uncommented (lack of v3 hardware output)? It might be useful to know why, otherwise I may end up down a rabbit hole that has already been "solved".

Completely different hardware.  OVMS V1 and V2 were based on a PIC
chip whereas OVMS V3 is based on the ESP32 plus several peripheral
chips.

                                                        -- Steve

jabelone, 2021-04-11:
Ok great, that makes sense. I'll look into adding back support if I can find some time in the next couple of weeks.

jabelone, 2021-04-11:
I've updated the title of this issue to better capture the problem. I've also opened a PR (#608) to update the documentation to clarify this issue until it can (hopefully) be resolved within the next couple of months.

jabelone, 2021-04-12:
Just found this older issue (#156) which appears to be the same thing. Referenced here for documentation purposes, not sure which issue should be closed.

mjkapkan, 2021-04-16:
Hi, would suggest to close the previous issue as that one is more of a question regarding pin-out. And thanks for bringing this up @jabelone . I somehow assumed all the NL code was ported to V3 a while back, but it seems like no. Anyway, seems like a quick fix.

ICULikeMac, 2021-11-19:
Hi, I would love to activate remote climate control, does anyone know of a guide on how to pull TCU pin 11 high as the manual states? Thanks

dalathegreat, 2021-12-26:
Should we agree on to use Pin 1 (SW_CTL) on the EGPIO? I can write the code for it, but I am unsure if the board works this way...

mjkapkan, 2021-12-28:
I use EGPIO controls in one of my plugins so that part is quite clear. However do I understand it correctly that NL 2011-2012 users have to connect a separate wire from TCU to OVMS EGPIO output? Can someone explain why is it needed? Is it simply there to activate TCU before sending the climate control commands or do we want to receive something from TCU?

dalathegreat, 2021-12-28:
The ZE0 LEAF only listens to remote heat commands if a pin is set to +12V on the wiring harness. So that is why we need the extra wire. They removed this strange requirement on the 2013+ models.

mjkapkan, 2021-12-28:
Ah okay, that explains it. In that case it's very simple. We just need to agree if we should use:

a) EXT_12V output (pin 9 on DA26 connector), which might conflict with other equipment already connected to the EXT_12V output.

b) One of the low voltage EGPIO ports which will require additional relay, and wiring from SW_12V output (pin 18 on DA26 connector) , but less chance to have a conflict due to leas frequent usage.

@dalathegreat @dexterbg @glynhudson what are your thoughts?

mjkapkan, 2021-12-28:
Actually, the best option would be to add a dropdown option called climate control activation pin (would appear only if MY is set to <2013 in the features page). So would suggest to go with this option. 😅

We should also update the documentation with a scheme of how everything should be connected for both low voltage and direct 12V pins.

mjkapkan, 2021-12-28:
Oh and lastly we should find out the timing when 12V should be active on that TCU pin.

a) Turn ON X seconds before sending CC activation CAN messages, and turn OFF right afterwards following Y seconds delay.
b) Turn ON X seconds before sending CC activation CAN messages, and turn OFF Y seconds after CC shuts down.

dalathegreat, 2021-12-29:
@mjkapkan I started on the code in my branch https://github.com/dalathegreat/Open-Vehicle-Monitoring-System-3/commit/09c58a2feda13c32f79c6599fdcb6a7512cf1a48

Let's use the old timing method on turning on/off the pins since it was already there.

I also think we can go for a configurable pin now during development, but maybe it would confuse the users even more? It is a quite hard concept to wrap your head around to get heating going.

Now I need to figure out how to make changes to the web configuration...

mjkapkan, 2021-12-29:
Yeah, was thinking the same @dalathegreat, we could leave the old timing in place assuming it worked 😅. But regarding the pins, I think leaving them configurable will not add more confusion. We can leave the EXT_12V pin as default one and just add remaining ones to drop-down. So as long as we add proper documentation this solution will be more easily understandable than it was until now. (I did not even get it myself 😂). BTW, I can help you modify the web config this weekend.

dalathegreat, 2021-12-30:
I had some spare time, **so this is now ready for testing!**
Firmware available here: dalasevrepair.fi/wp-content/OVMS/ovms3.bin

I made the initial draft for web config:
[image]

So now it should default to 1, which is pin 18  on the DB26 connector (Ext12V), and the user can make changes if wanted. I am not really that satisfied with the look of this window, so I will maybe tweak it a bit still. But this all should work now for testing! I will update my 2012 LEAF later this weekend, I don't have it at my house at the moment.

mjkapkan, 2021-12-31:
Tested with my 2013 car. I set the MY to 2012, and entered 1 in the config, but the 12V did not activate. (It is an easy check for me as it is connected to aftermarket PTC heater on my car :D). Also if the MY is > 2012, it should not show the entry for TCU activation pin or at least not required to enter something there. I think I can help to fix that during weekend if you get fed up with this ;).

dalathegreat, 2021-12-31:
@mjkapkan So to test this, you issued an AC request?

The "1" I took from peripherals.h, it had a definiton for MAX7317_SW_CTL. Is this the same you have in your PTC heater code?

Does the code to trigger the output look OK? Or do we have an issue with the logic that wraps around it and counts? I will think a bit about this...

mjkapkan, 2021-12-31:
@dalathegreat I think we should add logging for that for now to see when the logic is triggered. Like we have for this message:
`I (6539995) v-nissanleaf: Auto Disable Climate Control`
Regarding my heatassist plugin, I trigger the EXT_12V from JavaScript by executing a simple command (same as from console): 
to turn on:
`egpio output 1 1`
to turn off:
`egpio output 1 0`

Also I see this when I trigger the ext 12V from my plugin or with a command `power ext12v on` ,but not when activating CC for Leaf MY12 with your firmware.
`I (6709055) ext12v: Powering on external 12V devices`

I wonder if there is a sample code somewhere that turns the switch on from C... :)

dalathegreat, 2022-01-01:
Good idea, I added logging in latest push https://github.com/dalathegreat/Open-Vehicle-Monitoring-System-3/commit/89122eab0b4cabf95932f7c192e91f3414f58bdc

Firmware available here: dalasevrepair.fi/wp-content/OVMS/ovms3.bin

Our code does the same thing as the built in ext12v.cpp
`void ext12v::SetPowerMode(PowerMode powermode)
  {
  m_powermode = powermode;
  switch (powermode)
    {
    case On:
#ifdef CONFIG_OVMS_COMP_MAX7317
      ESP_LOGI(TAG, "Powering on external 12V devices");
      MyPeripherals->m_max7317->Output(MAX7317_SW_CTL, 1);
#endif // #ifdef CONFIG_OVMS_COMP_MAX7317
      break;
    case Sleep:
    case DeepSleep:
    case Off:
#ifdef CONFIG_OVMS_COMP_MAX7317
      ESP_LOGI(TAG, "Powering off external 12V devices");
      MyPeripherals->m_max7317->Output(MAX7317_SW_CTL, 0);
#endif // #ifdef CONFIG_OVMS_COMP_MAX7317`

So it should just work?

dalathegreat, 2022-01-01:
Hmm, I ran some tests on my 2015. By running the 'wakeup' command, the 12V is triggered. However, by looking in the log it seems like it also resets the modem somehow. Do you get this also @mjkapkan ?

[image]

Note for myself, running wakeup will now trigger the EV activation logic and leaf it high, so we need to handle also resetting this if user wakes up 2011-2012 vehicle, so that it doesn't deplete the 12V battery.

mjkapkan, 2022-01-02:
As discussed the problem here is that the value is not passed correctly from the web config page. If you follow the similar logic that is used for setting the `maxgids` it should work fine. Also no need to use `MAX7317_SW_CTL` as it is defined as `1` anyway.

dalathegreat, 2022-01-02:
I fixed the config writing and pushed a fix for it. My repo and server now has the latest version.

Now we only have to
- Fix the webUI so that it is a dropdown menu for the pins (and make it not possible to select the pin powering the modem!)
- Fix the wakeup command so that it doesn't leave on the +12V pin on 2011-2012. Maybe move the activating of the pin down to the nl_remote_command_ticker?

dexterbg, 2022-01-02:
There's a web UI template you can copy, see `OvmsWebServer::HandleCfgBrakelight()`:
- https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/master/vehicle/OVMS.V3/components/ovms_webserver/src/web_displays.cpp#L1373

dalathegreat, 2022-01-03:
I pushed the last set of fixes to my branch. Now we just need to test the solution. Thanks to @mjkapkan for confirming that the signal is ON for 1s when sending remote commands. I will try on my 2012 LEAF tonight.

I added a picture guide here on how to connect the TCU to OVMS: https://www.mynissanleaf.com/viewtopic.php?f=37&p=616096#p616096

I linked to this guide in the documentation. If all goes well we'll open a PR soon!

mjkapkan, 2022-01-05:
@dalathegreat  One more cosmetic fix needed here:
[image]

mjkapkan, 2022-01-05:
Fixed.
[image]

mjkapkan, 2022-01-14:
@jabelone @dexterbg This can be closed now. @dalathegreat verified that ot works.

closed 2022-01-14 by dexterbg