openvehicles/Open-Vehicle-Monitoring-System-3 #620: NL: kWh charged not showing in push notification

dalathegreat, 2021-04-30:
See attached screenshot. When charge is started, the notification starts at 0.0kWh. However, when the charging stops, the amount charged is not stored in the push notification, making the whole metric kinda pointless. 

[image]

Would be superb to have this stat included!

dexterbg, 2021-05-01:
Daniel,

the messages shown indicate the Leaf adaption has no proper support for the `ms_v_door_chargeport` metric yet.

The charge port needs to be "open" for the vehicle framework to send a charge status notification instead of the standard status notification "Not charging" you show. If a vehicle adaptor cannot read the actual charge port status, the standard workaround is to set the flag to true on charge start and to false only at the beginning of the next drive cycle.

Another option is to override `OvmsVehicle::CommandStat()` to replace the standard notification by your own implementation.

You can find examples for both of these options in the other vehicle modules.

Of course, generally speaken, the `ms_v_charge_kwh` metric also needs to be supported, but I see the Leaf does that.

Regards,
Michael

drc38, 2021-05-11:
@dalathegreat  I was just about to submit a pull request with some extra metrics based on your AZE0 dbc file. So have modified the code to leave `ms_v_door_chargeport` open until the car is turned on as per @dexterbg suggestion. If you want to test it is on my repo under Leaf-updates.

In relation to the updated dbc file what does `LB_MAX_POWER_FOR_CHARGER` in 0x1DC actually mean, I have 31kW reported on my 30kWh Leaf?  and I get 110kW reported for max battery discharge limit and 11.25kW for max battery charge limit when AC charging. For the charging limit I would of expected it to be 3.3 or 3.6kW which is the maximum for the onboard AC/DC charger?

drc38, 2021-05-12:
@dalathegreat, I'm still somewhat confused by the terminology. What is the difference between the max power for charger and the battery charge limit? On a DC fast charge my Leaf charges at around 45kW which is higher than the 31kW max power reported.

dalathegreat, 2021-05-12:
Me too! Take a look at the official document (can't share it all)
[image]

It goes from -10 to 90kW, it really doesn't make much sense. The discharge/charge power limits above are used for when you drive the vehicle. If you modify these with a CAN-bridge, you affect the amount of power and regen that is available. If you modify the LB_MAX_POWER_FOR_CHARGER during charging you affect the current going into the battery. So basically from my testing the battery charge limit is not used during charging, but REGEN.

drc38, 2021-05-12:
Thanks, that makes more sense. I've renamed the metrics to make their use clearer.

drc38, 2021-05-13:
Daniel (@dalathegreat) do you have the official table for 0x390? I think I have figured out the bulk of the message but there are still a few bit fields that would be useful to understand. See the updated code here: [https://github.com/drc38/Open-Vehicle-Monitoring-System-3](url) 
If you are able to compile (or I can send you the bin file) and test with a DC charger (as I don't typically use one) that would be great.

dalathegreat, 2021-05-13:
@drc38 The official documentation doesn't say anything more about 0x390 other than I've already put in the .dbc file at https://github.com/dalathegreat/leaf_can_bus_messages . I suspect there is more in this message, but the gen1 ZE0 LEAF has tons of more content in the 0x380 message (like AC voltage)

I don't have possibility to test at the moment unfortunately!

drc38, 2021-05-14:
@dalathegreat ok no worries I have used some old fast charge can logs to verify it against. Suggest adding the charge power to the  dbc file which is contained here `(d[0] & 0x01) << 8 | d[1] )` in 0x390

drc38, 2021-05-23:
I've created PR #632 to use the charge pilot stale status to close charge port door, it does not address if the charge cable is left plugged (and switched off at the wall/charger though) but should be an improvement over simply waiting for the next drive cycle.

glynhudson, 2021-06-03:
[image]

This is working for me, I'm running edge build

closed 2021-12-26 by dalathegreat