openvehicles/Open-Vehicle-Monitoring-System-3 #624: dexters-web.de HTTP API?

glynhudson, 2021-05-04:
The docs only document an API for `api.openvehicles.com`, is there an equivalent API for dexters-web.de? I've tried `api.dexters-web.de` but this doesn't seem to work? 

https://docs.openvehicles.com/en/latest/protocol_httpapi/index.html

dexterbg, 2021-05-04:
That API is implemented by any V2 server, as it's part of the perl server implementation.
So the server address to use is identical to the V2 server address.

Quoting from my previous info on this on the developer list:
http://lists.openvehicles.com/pipermail/ovmsdev/2018-July/012696.html

> my server offers the standard OVMS server APIs…
> 
>   * TCP: https://github.com/openvehicles/Open-Vehicle-Monitoring-System/raw/master/docs/OVMS-Protocol.pdf
>   * REST/JSON: http://lists.openvehicles.com/pipermail/ovmsdev/2013-October/001667.html
>       o Test: https://ovms.dexters-web.de/f/simpleclient.htm
> 
> …plus an additional stateless REST API on http/https:
> 
>   * Text command:
> 
>     https://dexters-web.de/api/call?fn.name=ovms/cmd&fn.vehicleid=…&fn.carpass=…&fn.cmd=…
> 
>   * CSV data export:
> 
>     https://dexters-web.de/api/call?fn.name=ovms/export&fn.vehicleid=…&fn.carpass=…
> 
>     Optional parameters:
>     - fn.format: "csv" (=inline) / "zip" (default)
>     - fn.types: string, data types to download; comma separated list or SQL LIKE pattern
>     - fn.since: ISO format timestamp, filter newer records
>     - fn.last: integer, output last n records
> 
>     Example:
>     https://dexters-web.de/api/call?fn.name=ovms/export&fn.vehicleid=…&fn.carpass=…&fn.format=csv&fn.types=D,S&fn.last=3

Regards,
Michael

closed 2021-05-04 by dexterbg

glynhudson, 2021-05-05:
Hi Michael,

Thanks for your reply, your mailing list post seems to document a csv export API which is not documented on the official docs page. 

However, the link to your simple client demo is very useful https://ovms.dexters-web.de/f/simpleclient.htm and clearly shows where I was going wrong, the URL is `ovms.dexters-web.de` 

e.g 
Login:
`https://ovms.dexters-web.de:6869/api/cookie?username=USERNAME&password=PASSOWORD`
List vehicles:
`https://ovms.dexters-web.de:6869/api/vehicles`
Status: 
`https://ovms.dexters-web.de:6869/api/status/VEHICLEID`

dolam, 2021-07-05:
hello glynhudson, 
sorry if I disturb you, you could post how I have to configure HA for OVMS Twizy, I am on dexters.de server but I am unable to use the computer, but I can copy the configuration made by others, you want to help me to insert my Twizy on home assistant?
I would be very happy if I could too, thank you if you can / want to help me realize this little dream.
Hello

dexterbg, 2022-10-21:
To use cookie based authentication at ovms.dexters-web.de with Home Assistant, use this sensor template:

```
  - platform: command_line
    scan_interval: 60
    name: car_charging
    command: |-
      username='…'; password='…'; vehicleid='…'; ip='https://ovms.dexters-web.de:6869';\
      c=$(curl -X GET -sc - "$ip/api/cookie?username=$username&password=$password" -o /dev/null);\
      echo "${c}" | curl -X GET -sb - "$ip/api/status/$vehicleid"
    value_template: "{{ value_json.chargestate }}"
```

Fill in username, password & vehicleid. This is just an example, the `/api/status` response contains a lot more detail, you can extend the `value_template` accordingly. You can also access the other API calls using the same approach.

jamessimpson007, 2023-05-12:
Dear clever people, 

Thanks for the details of how to get this sensor up and running Michael. Please could I ask for one bit of help? I'm struggling to work out how to extend the value_template as you mentioned. 

I'd like to get soc, chargestate and charge_etr_full. It seems a bit clunky to set up totally different sensors and API calls. How could I configure the above code you posted to give me those three values, ideally feeding from one API call into three sensors please? 

Thanks in advance!
James

dexterbg, 2023-05-12:
James, have a look at the documentation and the simpleclient test bed regarding the "status" response and other responses:
- https://docs.openvehicles.com/en/latest/protocol_httpapi/requests.html#get-api-status-vehicleid
- https://ovms.dexters-web.de/f/simpleclient.htm

You can read all those fields from the same status API call result at once. I can't help you with Home Assistant on the syntax for reading multiple values at once, but that sure is a standard operation.

From a quick duck, I'd assume this page explains how to do that:
- https://www.home-assistant.io/integrations/sensor.rest/#fetch-multiple-json-attributes-and-present-them-as-values

Regards,
Michael

jamessimpson007, 2023-05-12:
Thanks very much. Very kind of you.

BW, James.

dexterbg, 2023-09-18:
James,

it seems you've just posted your password, so I suggest changing that ASAP.

I don't know if you can chain multiple JSON responses like that with HomeAssistant -- each curl call in your command list returns a separate data object, maybe you need to merge these before processing?

Also, maybe HA is tolerant with the command syntax, but I'd normally expect you need to add line continuations after each command (`\;`) except the last.

The odometer value is in the "/status" response, but be aware it's scaled by factor 10 (V2 legacy). The availability of the TPMS pressure values depends on your vehicle, for example an Up/Mii/Go will only deliver health & alert states.

Regards,
Michael

jamessimpson007, 2023-09-19:
Dear Michael,

Thanks for your reply. I feel like a complete idiot for sending my password through! Obviously I have changed it, thank you for pointing it out!

With your help, I have figured it out. I had to create a new sensor for each section. So one sensor for status, and another sensor for all of the TPMS data.

Thank you for all your hard work with the server, it is hugely appreciated.

Cheers,
James

James Simpson
21B Park Rd
Hatherleigh
EX20 3JS
07980 832286