openvehicles/Open-Vehicle-Monitoring-System-3 #634: NL: range display incorrect whie rapid charging

glynhudson, 2021-06-03:
[image]

When rapid charging is in progress, the displayed range is crazy high! When rapid charging finishes, the range display reverts to normal

drc38, 2021-06-04:
Try changing this parameter to `true` `MyConfig.GetParamValueBool("xnl", "soh.newcar", false)` which will then use the `maxGids` parameter along with `whPerGid` to do the calculation.

drc38, 2021-06-04:
If the `soh.newcar` param is set to `false` the code is using `m_battery_energy_capacity` which is set by can message 0x59e for a 24kWh battery. Based on your experience with fast charging this can message may not be appropriately handled or may mean something slightly different, hence the weird values.

glynhudson, 2021-06-07:
Thanks, I've just tried to apply the change you suggested, but It get the following error:

[image]

drc38, 2021-06-07:
That's the code snippet, from memory you can set it in the menu or use the shell command `config set` with the parameter and value

glynhudson, 2021-06-08:
Ok got it, sorry I should have realised that. I managed to set it with 

`config set xnl soh.newcar true`

To view the Leaf config:

`config list xnl`

```
xnl (readable writeable)
  acvoltagemultiplier: 1.25
  cabintempoffset: .0
  canwrite: yes
  maxGids: 246
  modelyear: 2012
  newCarAh: 66
  soc.newcar: no
  soh.newcar: yes
  suffrange: 0
  suffsoc: 
```

I'll let you know how I get on next time I rapid charge. Cheers

glynhudson, 2021-07-23:
Issue is now resolved 

[image]

closed 2021-07-23 by glynhudson