openvehicles/Open-Vehicle-Monitoring-System-3 #642: Make docs.openvehicles.com default to stable instead of latest.

johnduffield, 2021-07-14:
The documentation available at https://docs.openvehicles.com defaults to /en/latest, which can be very confusing for new users when the bleeding edge documentation doesn't match the current stable release.

Make the default /en/stable so that the default documentation is consistent with the firmware that most users will be running.

danielklejnstrup, 2026-09-13:
I agree, this is confusing.
Also, https://docs.openvehicles.com/en/stable/ returns an 404 error page.

dexterbg, 2026-09-13:
You're right, the ReadTheDocs side seems to be broken again, I can only switch between "latest" and "3.3.004", and there is no longer a "stable" option.

@markwj Mark, I think you need to check the RTD configuration once again. "latest" seems to be working, it would be best if each release branch automatically registers as a version, and "stable" should be the most recent release branch. If that cannot be automated, maybe we can use a tag "stable"?

markwj, 2026-09-14:
> [@markwj](https://github.com/markwj) Mark, I think you need to check the RTD configuration once again. "latest" seems to be working, it would be best if each release branch automatically registers [...]

Seems we have to manually tag versions in readthedocs for release. I have done that now for 3.3.004, 3.3.005, and 3.3.006. We get 'latest' as standard.

I have also tried to create some automation rules to do this automatically for whenever we tag a new version. We'll have to wait to see if that works (and it may also conflict with our new approach of releasing updates to tagged versions).

Regarding 'stable', I can't see how to get that working on our project. It should be automatic, but doesn't seem to recognize our tags as semver standard (which they should be). I will keep trying, but it is not trivial.

markwj, 2026-09-14:
I think I've found the problem (the 'stable' version was hidden, and a PITA to get it unhidden).

Please check now.

dexterbg, 2026-09-14:
Doesn't seem to work yet: I picked my docs addition into the `release-3.3.006` branch, but both `stable` and `3.3.006` RTD versions remain unchanged.

It seems RTD only picks up the tags, but these are our fixed reference points for the commit count distance, so don't reflect later additions to the release branches.