openvehicles/Open-Vehicle-Monitoring-System-3 #644: NL: HVAC metrics incorrect for MY2013

mjkapkan, 2021-07-30:
It looks like the HVAC gets set to yes every time the car is charging. This was already fixed in #281, but it looks like something broke it afterwards.

glynhudson, 2021-07-30:
I''ve also noticed this issue

drc38, 2021-08-24:
The only functional difference I can spot is the code for the fan speed. 
```
int fanspeed_int = ((d[4] & 0x38) >> 3); //use bits 5:3 to read fan speed
m_climate_fan_speed->SetValue(fanspeed_int / 7.0 * 100);
```
versus
```
int fanspeed_int = d[4] / 8;
// todo: actually we need to use individual bits (7:3) here as remaining ones are for something else
if ((fanspeed_int < 1) || (fanspeed_int > 7))
      {
        fanspeed_int = 0;
      }

m_climate_fan_speed->SetValue(fanspeed_int);
```
is the fan speed also showing as >0 when charging?

drc38, 2021-08-24:
Also in https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/376/files on line 752 should it be a bitwise & or logical &&?
`hvac_calculated = (climate_on & (m_climate_fan_speed->AsInt() != 0));`

mjkapkan, 2021-08-25:
Yeah, I now remember that CC flag used to be on during charging so I added the fan speed as an additional check (you can never have hvac on if the fan is not running). So this one needs to be logical. I will test how it looks tonight.

mjkapkan, 2021-08-29:
Checked. So it seems whenever the car is charging we get 57% of fanspeed. We could use this rule+xnl.cc.fanonly metric to mitigate this until we find another solution.

dalathegreat, 2021-12-28:
I checked this yesterday on my 2015 LEAF running the latest build, fanspeed was at 0%, so no issue here?

mjkapkan, 2021-12-28:
@dalathegreat Checked just now. Issue still persists. If vehicle is charging, then fanspeed is set to 14%, and hence CC flag is enabled. If you dont't have this issue, then it could be limited to <2015 MY cars.

mjkapkan, 2021-12-28:
Basically this should be an easy fix. We just need yo find one more metric to validate against.

mjkapkan, 2022-01-05:
Implemented a hacky fix, le't see if it works for most and then we can close it. Link to beta firmware in PR: #676 .

mjkapkan, 2022-01-07:
Permanent fix implemented. PR #676 should close this issue.

drc38, 2022-01-07:
The changes look useful. The substate transitions may need some tweaking however. Plugging in to my charger which does not charge until a set time of day no longer transitions out of a powerwait substate. 
[image]

closed 2022-01-13 by mjkapkan