openvehicles/Open-Vehicle-Monitoring-System-3 #645: NL: Unable to Start Charge from Android App

mjkapkan, 2021-07-30:
Sliding the charger plug in the app does nothing in Android app, but works in iOS. Also the charge start command is working as well. So this must be somewhere in the Android app.

glynhudson, 2021-07-30:
Starting a charge works fine for me with a Nissan Leaf using the Android app. On the Leaf starting a charge only works when the vehicle is plugged in and the charging timer is enabled.

mjkapkan, 2021-08-02:
Interesting. seems to work now.Might have been some connection issues...

closed 2021-08-02 by mjkapkan

mjkapkan, 2022-01-10:
Found out what was the problem. The charge port was set to false after 2 minutes even pilot signal is set to true. This caused the slidder to not appear. Now fixed in #676 .