openvehicles/Open-Vehicle-Monitoring-System-3 #647: Adding text to vehicle alerts

einhalv, 2021-08-21:
I frequently receive vehicle alerts from the OVMS in my car (a Tesla Roadster) and I appreciate that functionality very much. However, I struggle with remembering the meaning of the vehicle-alert codes and would therefore like to see descriptive text messages appended to the alerts. I am willing to code that myself, but I have some questions to help me on the path (or dissuade me):

1) I this of interest to others?
2) If yes, is someone else working on this?
3) Are we strongly memory constrained? I have made a stand-alone program with a command line interface on a Linux computer as a test. The overhead of the data structure and the code to access it appears to be ~50kB compared to a reference program without these parts. It is yet to be determined how much it would actually add to the firmware. 
4) Is it OK to add it to the vehicle specific code? I much prefer that because I have identified where to put it and with my infinitesimal C++-skills knowing that is a good start. I have installed the development framework and am ready to experiment.
5) Is there any documentation on the Notify-methods? It seems to me that adding a notification just after the line
310                         MyNotify.NotifyErrorCode(k1, k2, true); // Raise error notification
may do the job. I see there are some string notifcation methods that could potentially be used, even one with a printf-like interface. But I can't figure out the significance of the two first arguments.

slcasner, 2021-08-21:
Sorry, you can't fix the problem in the place you are looking.  That notification code is about notifications distributed among the tasks in the OVMS firmware.  If you are talking about notifications displayed in the iOS or Android app on your phone, those are delivered as numeric codes over the server-v2 protocol and are translated into strings by the server.  I worked with Mark to fill in some of the missing strings, but it seems there is more to do.

dexterbg, 2021-08-22:
First of all, if you're on my server (dexters-web), the missing error code translations should be fixed now. I somehow somewhen (probably long ago) missed keeping the .vece files in the server directory, which are now back in place.

If that doesn't fix the issue for you: the place to add missing code translations for the Tesla Roadster with the current ``error`` notification function is here:
https://github.com/openvehicles/Open-Vehicle-Server/blob/master/v3/server/vece/tr.vece

Generally speaking, relying on the V2/MP server to translate error codes doesn't fit the V3 framework's modular notification transport and notification channels not being routed through a custom server.

Up to now, the Roadster is the only vehicle using the VECE translation, other vehicles use text ``alert`` type notifications. That's because the current error code & data types are too restricted to be usable by other vehicles. The basic concept of error occurrence & presence tracking by the framework is still good, but the storage structure needs to be more flexible to be generally applicable. But that's another issue.

To answer question 3: the TR VECE file has 11 KB, that should be about the amount of firmware space added by moving these into the module. That's currently no problem, we've got ~810 KB of the 4 MB available, which still leaves plenty of space for new vehicles.

On question 5: I've explained the basic notification concepts including  the ``type`` & ``subtype`` arguments in the user manual:
https://docs.openvehicles.com/en/latest/userguide/notifications.html

The MyNotify API has no developer documentation other than the source and the discussion on the developer list some years ago. But it's straight forward:
- ``NotifyString[f]`` sends a notification of any type/subtype from a string
- ``NotifyCommand[f]`` sends a notification of any type/subtype from a command result ([f] variant for constructing the command call)
- ``NotifyErrorCode`` sends a notification of type ``error`` subtype ``code``, if the error code is newly present or cleared

Grep the source for examples.

But don't simply replace or extend the error handling on the Tesla Roadster by text alerts. If you'd like to work on generalizing the error notifications, I can give you some other vehicle examples that would need to be supported.

Regards,
Michael

einhalv, 2021-08-22:
Thanks!

I just provoked some errors and now the messages appear as I would like
them to. My immediate concern is thus resolved. However, I will look at the
code following your advice and see if there is something that I can
contribute.

Einar

closed 2021-08-22 by einhalv

markwj, 2021-08-23:
The reasons to do the VECE system (converting error codes to textual messages) were as follows:

1. v2 modules had very little firmware space
2. v2 modules were hard to update (with no OTA)
3. The error codes were under active development at the time, with not many known

Those are less of an issue for v3, so if someone wants to migrate the codes there it would be fine by me. But that would need to be done in an optimised manner.