openvehicles/Open-Vehicle-Monitoring-System-3 #650: Generate TOKEN with API

gianfrdp, 2021-09-13:
Hello,
following issue #624, I'm trying to generate an API token based on API documentation https://docs.openvehicles.com/en/latest/protocol_httpapi/authentication.html#api-token-maintenance.

Trying https://ovms.dexters-web.de:6869/api/cookie?username=USERNAME&password=PASSOWORD
to login works.
Then I try to generate a token with:
curl -v -X POST -F 'application=homeassistant' -F 'purpose=read' -F 'permit=token.admin' 'https://ovms.dexters-web.de:6869/api/token'
but it says: Unrecognised API call

What I'm doing wrong?

Thanks

dexterbg, 2021-09-13:
I'm still running server version 2, which doesn't support token authentication. Use password authentication instead.

closed 2021-09-13 by dexterbg

dexterbg, 2022-10-21:
To use cookie based authentication at ovms.dexters-web.de with Home Assistant, use this sensor template:

```
  - platform: command_line
    scan_interval: 60
    name: car_charging
    command: |-
      username='…'; password='…'; vehicleid='…'; ip='https://ovms.dexters-web.de:6869';\
      c=$(curl -X GET -sc - "$ip/api/cookie?username=$username&password=$password" -o /dev/null);\
      echo "${c}" | curl -X GET -sb - "$ip/api/status/$vehicleid"
    value_template: "{{ value_json.chargestate }}"
```

Fill in username, password & vehicleid. This is just an example, the `/api/status` response contains a lot more detail, you can extend the `value_template` accordingly. You can also access the other API calls using the same approach.

dexterbg, 2022-12-07:
To also [connect to the module](https://docs.openvehicles.com/en/latest/protocol_httpapi/requests.html#get-api-vehicle-vehicleid) (log in), so it applies the higher update frequency, simply add another curl for `/api/vehicle` like this:

```
  - platform: command_line
    scan_interval: 60
    name: car_charging
    command: |-
      username='…'; password='…'; vehicleid='…'; ip='https://ovms.dexters-web.de:6869';\
      c=$(curl -X GET -sc - "$ip/api/cookie?username=$username&password=$password" -o /dev/null);\
      echo "${c}" | curl -X GET -sb - "$ip/api/vehicle/$vehicleid" -o /dev/null;\
      echo "${c}" | curl -X GET -sb - "$ip/api/status/$vehicleid"
    value_template: "{{ value_json.chargestate }}"
```

As processing is asynchronous between module and server, the first status call sent won't receive an up to date status result. But as the sensor scan interval is 60 seconds, the next will. Be aware this will keep a client connection active basically forever, so the module will not return to the "idle" update interval.

To avoid that, you may possibly put the connect & disconnect calls onto some Home Assistant command or schedule. Check the HA documentation on how to do this.