openvehicles/Open-Vehicle-Monitoring-System-3 #652: Total number of charge

bunyevacz, 2021-09-30:
Hi,

Maybe this is a real noob question, but could not find in the documentation:

_Which metric corresponds to the total number of charges since the vehicle's manufacturing date?_

Just like the odometer, but for the number of charging events.

Motivation:
I want to buy a Nissan Leaf 30kWh model (2016,2017), and by talking with some of the dealers locally, 
it is may or may not be the case that the vehicle's battery got reset in the dealership.
One of the possible motivations behind the battery reset, that the vehicles display near 100% SOH for a couple houndred kilometers.

I want to simply check the odometer value against the total number of charges.
 Eg. the vehicle got 38000km, and only 3 charge events, then it's fishy. On the other way around, if the odometer has 38000km, and the number of charge events is 3500, then the odometer displays incorrect number.

I specifically bought the OVMS (ordered in july, got it 3 days ago in EU) for vehicle checking before purchase.

glynhudson, 2021-10-01:
This information is easily obtain using Leafspy.

dalathegreat, 2021-12-26:
Number of charges is only fetchable with dev commands (like Leafspy does). Here's how it is done:

Name: L1 L2 Charge
Type: Integer
Description: Number of L1/L2 connections and charges, even not fully. In the example L1 L2 Charge is 115.
Query: 0x797 03 22 12 05 00 00 00 00
Answer: 0x79A 05 62 12 05 **00 73** 00 00
Formula: L1 L2 Count = (data[4] << 8) | data[5]

Since dev commands puts the various control modules into tight timing mode, I suggest we do not add this to OVMS at all. Use Leafspy!

dexterbg, 2021-12-26:
That's a standard OBD PID read, why would there be an issue with tight timing?

You should be able to query this info any time using the shell command: `obd can1 req dev 797 79a 221205` -- does that have any timing issues?

Regards,
Michael

dalathegreat, 2021-12-26:
When the LEAF detects 0x7## debug request messages, it puts all the control modules into tight timing mode and makes them more alert to CAN timing errors. The official Nissan Consult3+ documentation even says the car should not be operated or charged during a CONSULT session, which is the only moment 0x7## debug messages are normally requested. This is also one of the reasons there are extremely rare instances of the car getting into a fault state when driving with Leaf Spy. There has been reports of brake failures on LEAFs, and it was theorized it was due to OBD2 dongle and Leafspy going at the same time. For live or everyday logging/messaging I highly recommend just using the regular CAN bus messages and not trying to send debugging and engineering messages.

dalathegreat, 2021-12-30:
@dexterbg this issue can be resolved, since OVMS actually polls for this info and lists it as a metric.

OP, it is available as L1L2_COUNT_PID and QC_COUNT_PID. But again, easiest to use Leafspy pro before purchasing a vehicle :)

dexterbg, 2021-12-30:
@bunyevacz to query these metrics as a user, issue command
```
metrics list v.c.count
```

To query all Nissan Leaf specific metrics, issue…
```
metrics list xnl
```

You can do this for example in the web shell or from the App's messages page.

@dalathegreat I suggest adding some info on the xnl metrics to the user manual at https://docs.openvehicles.com/en/latest/components/vehicle_nissanleaf/docs/index.html

Regards,
Michael

closed 2021-12-30 by dexterbg