openvehicles/Open-Vehicle-Monitoring-System-3 #668: Bolt EV interest?

kallisti5, 2022-01-01:
Any interest in patches for the Bolt EV?   I've been struggling with how unreliable Chevy's Onstar is for a while, and have been toying with the idea of a ODB2 -> ESP32 -> LTE vehicle monitoring system.

Contributing to OVMS seems like a lot easier of a plan though :-)

Here's what to expect from Onstar.  Making API calls every 30 minutes over a little under 2 days:

(23 working Onstar API calls)
(14 errored OnStar API calls)
(13 working Onstar API calls)
(33 errored Onstar API calls)

dexterbg, 2022-01-02:
Alex,

we're always interested in new vehicle support, and glad to help any new developer joining the team -- welcome! :-)

I was already wondering why no Bolt drivers turned up after the battery issues lately. Using the OVMS to monitor the cells and sending alerts on strange behaviour could certainly help lowering the risk and evaluating if the recently rolled out fix is effective. And it doesn't seem the Onstar API is useful on these topics.

So, of course, go ahead. Please check our developer guidelines & ressources:
- https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3#development-and-contributions

If the Bolt has many similarities to the Volt, consider adding a model selection to the Volt adaption. If not, introduce a new vehicle.
- https://docs.google.com/document/d/1q5M9Lb5jzQhJzPMnkMKwy4Es5YK12ACQejX_NWEixr0/edit#heading=h.t6s09orr514x

A good template for a new vehicle module is the Hyundai Ioniq vFL:
- https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/tree/master/vehicle/OVMS.V3/components/vehicle_hyundai_ioniqvfl

If you need help on this, ask on the developer list.

Regards,
Michael

dexterbg, 2022-01-02:
Btw, there was some activity on the Bolt here. @Edwintenhaaf supplied two pull requests to add Bolt support, but withdrew both before review:
- https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/625
- https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/626

Maybe you could join forces on this?

Edwintenhaaf, 2022-01-02:
Yes I did make a start with the advice I got but couldn’t get things working.( https://www.openvehicles.com/node/2345)
Ended up with an edited volt template that’s giving some basic data.
Would like to help but I’m not a developer (yet)
I am driving the rebranded Bolt, the Opel Ampera-e.

  *   #625<https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/625>
  *   #626<https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/626>

kallisti5, 2022-01-25:
oof.   I just got my OVMS v3, and was able to get some data going just based on the Volt template as well.

[image]

I'll take a look at those PR's and see if I can help them.

kallisti5, 2022-01-25:
I looked at the PID's for the Bolt EV... everything is actually really close to functional.
https://github.com/kallisti5/Open-Vehicle-Monitoring-System-3/tree/boltev

There are some variations in PID's for model year 2017 - 2018  vs 2019+.  How are these normally handled?

dexterbg, 2022-01-25:
If car generations or variants only differ in some detail and you cannot determine the model variant from CAN data (e.g. VIN): add a model type / year user configuration. Store the model variant in a member variable and query it wherever necessary. Do likewise for additional car specific attributes (e.g. battery type) as needed.

Search for "model year" in the manual & code for examples.

kallisti5, 2022-01-25:
Hm. might be able to tell from the VIN since it's GM.  The 10th digit is the year.

My early 2019 Bolt EV is K

Random sampling from the internet:
2017 is H
2017 is H
2017 is H
2018 is J
2018 is J
2019 is K
2020 is L

So, if H or J, use the old decoder, otherwise use the new one for 2019+

kallisti5, 2022-01-25:
ok.  Things working now.

```
OVMS# stat
Not charging
SOC: 78.0%
Ideal range: 201M
ODO: 26923.5M
```

[image]
[image]

Is there a way to dump the current state of things collected via the cli?  (cell voltages, etc?)

slcasner, 2022-01-25:
> Is there a way to dump the current state of things collected via the
> cli?  (cell voltages, etc?)

You mean like this?

OVMS# metrics list
m.egpio.input                            0,1,2,3,4,5,6,7,8,9
m.egpio.monitor
m.egpio.output                           1,3,4,5,6,7,8,9
m.freeram                                3553012
m.hardware                               OVMS WIFI BLE BT cores=2 rev=ESP32/1
m.monotonic                              10263631Sec
...

kallisti5, 2022-01-25:
ah, yup.

```
OVMS# metric list v.b
v.b.12v.current
v.b.12v.voltage                          12.7V
v.b.12v.voltage.alert
v.b.12v.voltage.ref                      12.66V
v.b.c.temp
v.b.c.temp.alert
v.b.c.temp.dev.max
v.b.c.temp.max
v.b.c.temp.min
v.b.c.voltage
v.b.c.voltage.alert
v.b.c.voltage.dev.max
v.b.c.voltage.max
v.b.c.voltage.min
v.b.cac
v.b.consumption                          0Wh/km
v.b.coulomb.recd
v.b.coulomb.recd.total
v.b.coulomb.used
v.b.coulomb.used.total
v.b.current
v.b.energy.recd
v.b.energy.recd.total
v.b.energy.used
v.b.energy.used.total
v.b.health
v.b.p.level.avg
v.b.p.level.max
v.b.p.level.min
v.b.p.level.stddev
v.b.p.temp.avg
v.b.p.temp.max
v.b.p.temp.min
v.b.p.temp.stddev
v.b.p.temp.stddev.max                    0°C
v.b.p.voltage.avg
v.b.p.voltage.grad
v.b.p.voltage.max
v.b.p.voltage.min
v.b.p.voltage.stddev
v.b.p.voltage.stddev.max                 0V
v.b.power
v.b.range.est
v.b.range.full
v.b.range.ideal                          324km
v.b.range.speed
v.b.soc                                  78%
v.b.soh
v.b.temp                                 13°C
v.b.voltage
xva.v.b.charging_limits                  0
```

No vin, no battery info.. hm

dexterbg, 2022-01-25:
The vin is metric name `v.vin` so wouldn't be visible when filtering for `v.b`.

kallisti5, 2022-01-25:
@dexterbg no, I know :-)  Forgot to include it above..

```
OVMS# metric list v.vin
v.vin
```

kallisti5, 2022-01-25:
nevermind :-)    Didn't realize data wasn't available with the vehicle off.    Powered vehicle on and getting data.

kssmll, 2022-02-01:
> nevermind :-) Didn't realize data wasn't available with the vehicle off. Powered vehicle on and getting data.

Hi. Maybe you already do this, but just for notice, I think Bolt also as a Volt needs for a swcan adapter for correct wake up. And of course swcan bus must be connected properly.

kallisti5, 2022-03-04:
> Hi. Maybe you already do this, but just for notice, I think Bolt also as a Volt needs for a swcan adapter for correct wake up. And of course swcan bus must be connected properly.

Yeah.. this seems needed. The vehicle goes to sleep after some time and OVMS can no longer pull the data.

I'm going to split this one into a few Bolt EV related issues.  However, with this work merged the "bare minimum" is there resolving this one :-)

closed 2022-03-04 by kallisti5