openvehicles/Open-Vehicle-Monitoring-System-3 #674: NL: 2011-2012 LEAFs show wrong charging info with latest master

dalathegreat, 2022-01-04:
Take a look at the attached picture:
[image]

Looks like something broke with the 2011-2012 LEAF after my introduction of separate handling of the two. I will try and fix this, since it makes the phone notifications spam quite a lot:
[image]

Zuikkis, 2024-10-30:
This bug seems to be still there in latest firmware.

Interestingly the voltage normally shows battery volts (like 367V in above picture) but if A/C is on the voltage occasionally jumps to >700V (perhaps double battery volts?). It maybe happens when A/C is using more than the charger provides.

Also your screenshot shows range 0km, this is fixed by setting "soh.newcar = yes".. This is odd as SOH itself is showing fine without the setting.

developerfromjokela, 2024-12-14:
Hi. Any news on fixing this?