openvehicles/Open-Vehicle-Monitoring-System-3 #680: OTA Firmware update failing from

robot-army, 2022-01-12:
Hi, 

I'm trying to set up a new OVMS kit, and I'm at my cabin in the woods with no SD card to hand. 

I've got to the 'firmware update' step of the configuration and it doesn't seem to want to do the update via the modem or the WiFi client connection. 

On either the WiFi or modem I see this when attempting to update to `3.2.018` from `3.2.015/factory/main (build idf v3.3.2-881-g22d636b Sep 2 2020 08:21:02)`

```
Processing… (do not interrupt, may take some minutes)
Current running partition is: factory
Target partition is: ota_0
Download firmware from api.openvehicles.com/firmware/ota/v3.1/main/ovms3.bin to ota_0
Expected file size is 3318144
Preparing flash partition...
Request error, please retry
```

It seems that the WiFi is connected and the modem as well, as both have IP addresses and 'good looking' connections in the status display. I've tried the WiFi on a hotspot from my phone as well as my router.

Also, it seems there is a 3.2.018 'version' in one of the partitions, but it refuses to boot from it. 

Is there a workaround to get the bin onto the box somehow other than using the WiFi client connection, modem, or SD card? Something involving SCP?

I also came across this thread - http://lists.openvehicles.com/pipermail/ovmsdev/2019-July/013597.html - which talks about 
```there's a new idf 3.3 config against the starvation, did you set that?

Your sdkconfig should have this:

CONFIG_SPI_FLASH_YIELD_DURING_ERASE=y
CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS=20
CONFIG_SPI_FLASH_ERASE_YIELD_TICKS=1```

Do I need to do something there?

Thanks! A beautiful bit of hardware and so far other than this, very impressive software as well. Looking forward to getting it running!

markwj, 2022-01-12:
It looks like the HTTP request failed during the download. I suggest turning off the modem (power it off) and use wifi. Then re-try it. If the fault is intermittent, it should recover.

Firmware can be downloaded either via HTTP (OTA), SD CARD, or USB (esp flasher) from a laptop.

dexterbg, 2022-01-12:
Maybe your internet connection "in the woods" has some issue the module cannot cope with. As you've got your smartphone, you might try downloading the firmware to that and running a simple web server on it (e.g. Tiny Web Server for Android), so the unit can download the file from there.

You can also try downloading the firmware from the EU server (`ovms.dexters-web.de/firmware/ota/v3.1/main/ovms3.bin`).

Regards,
Michael

dexterbg, 2022-01-12:
> Something involving SCP?

That would be possible but also need an SD card in the module to copy the firmware to.

robot-army, 2022-01-12:
> Maybe your internet connection "in the woods" has some issue the module cannot cope with.

I suspect this is the case, that's why I mentioned it. Is there some timeout that's a bit too aggressive or something? For what it's worth, I can watch youtube videos?

I'll try the workaround with the HTTP server on my phone, but it seems like there's something that could be fixed up in OVMS itself eventually maybe?

robot-army, 2022-01-12:
If there's anything I can do to help diagnose (get logs from terminal? enable some kind of debugging output?) I'm very happy to do that.

dexterbg, 2022-01-12:
Just checked the actual OTA process error responses, I think your issue isn't the download, rather the flash preparation takes too long and causes a module watchdog reboot. I've seen that before, and made some changes to help avoiding this, but the flash preparation is done by the esp-idf system, we don't have much control over it.

I recommend finishing the setup without doing the update first. Then wait until the ssh key generation has finished -- that's a CPU intense background process that can take up to five minutes, it will log a message when done and you can monitor the process by command `module tasks`, the process name is `RSAKeyGen`.

When `module tasks` shows both idle processes at > 95%, try updating.

It may also help to change the boot partition (just that, no reboot necessary), that seemed to help speeding up the flash preparation process.

robot-army, 2022-01-12:
> When `module tasks` shows both idle processes at > 95%, try updating.
> 
> It may also help to change the boot partition (just that, no reboot necessary), that seemed to help speeding up the flash preparation process.

OK, thanks for all that, I'll give those things a shot as soon as I can and report back.

robot-army, 2022-01-12:
I was not able to get the idle processes both down to 95%. One would stay in the high 90s and the other would hit like 92. 

I ran the flash again, and it's thrown that same error. 

Interestingly, I can see a line in the tasks 
```
3FFE81B8   90 Blk ota flash http   2256  2816  6144     316   3608  24132  1  22   7%  22/ 1
```
even after closing the flash window after seeing that `Request error, please retry` bit. 

Now, in the console, I'm seeing 
```
I (49396494) boot_comm: chip revision: 1, min. application chip revision: 0
I (49396504) esp_image: segment 0: paddr=0x00410020 vaddr=0x3f400020 size=0x1011fc (1053180) map
I (49396704) esp_image: segment 1: paddr=0x00511224 vaddr=0x3ffb0000 size=0x04588 ( 17800) 
I (49396724) esp_image: segment 2: paddr=0x005157b4 vaddr=0x40080000 size=0x00400 (  1024) 
I (49396734) esp_image: segment 3: paddr=0x00515bbc vaddr=0x40080400 size=0x0a454 ( 42068) 
I (49396744) esp_image: segment 4: paddr=0x00520018 vaddr=0x400d0018 size=0x2073cc (2126796) map
I (49397154) esp_image: segment 5: paddr=0x007273ec vaddr=0x4008a854 size=0x12d30 ( 77104) 
I (49397184) esp_image: segment 6: paddr=0x0073a124 vaddr=0x400c0000 size=0x00038 (    56) 
I (49397214) boot_comm: chip revision: 1, min. application chip revision: 0
I (49397214) esp_image: segment 0: paddr=0x00410020 vaddr=0x3f400020 size=0x1011fc (1053180) map
I (49397414) esp_image: segment 1: paddr=0x00511224 vaddr=0x3ffb0000 size=0x04588 ( 17800) 
I (49397434) esp_image: segment 2: paddr=0x005157b4 vaddr=0x40080000 size=0x00400 (  1024) 
I (49397444) esp_image: segment 3: paddr=0x00515bbc vaddr=0x40080400 size=0x0a454 ( 42068) 
I (49397464) esp_image: segment 4: paddr=0x00520018 vaddr=0x400d0018 size=0x2073cc (2126796) map
I (49397864) esp_image: segment 5: paddr=0x007273ec vaddr=0x4008a854 size=0x12d30 ( 77104) 
I (49397894) esp_image: segment 6: paddr=0x0073a124 vaddr=0x400c0000 size=0x00038 (    56) 
```

Then executing `module tasks` again, the `ota flash http` task was gone. 

I've double-checked the boot partition was set, and I'm rebooting now.

robot-army, 2022-01-12:
I think that `request error` was a red herring. 

I'm on 3.2.018 now. It all kept working in the background and now it's booted fine.

dexterbg, 2022-01-13:
Glad it finally worked out, good idea to check the tasks again for the flash process.

The request error came from the web frontend, but that has a timeout of 300 seconds for OTA commands, that's why I thought your module would perform a watchdog reboot. Please report if the issue persists with the new firmware version (i.e. when doing the next update or switching to "edge").