openvehicles/Open-Vehicle-Monitoring-System-3 #683: Server V2: optimize data volume for regular updates

dexterbg, 2022-01-18:
Especially with clients being logged in all the time (like home automation scripts or an App running in background service mode), the data volume needed for the regular updates is higher than necessary.

Messages 'F', 'S' and 'D' are retransmitted every 60 seconds by default in that setup, also while parking and with only minor or no visible changes in the metrics.

Some clients currently probably depend on these regular updates.

'L' has frequent unnecessary updates mostly from the GPS jitter and satellite count. 'S' and 'D' could also be reduced by allowing some value delta without retransmission. That would break clients depending on regular updates, so needs a new configuration option to enable.

I've prepared a change to reduce the 'F' retransmissions, as that most probably has no clients depending on and is currently the largest regular message. 'F' would normally need seldom retransmissions (carrying mostly static info like firmware & VIN), but it contains the network status. My change checks for substantial changes in the signal level. A better option would be to move the network info to another message, which will need updating the Apps.

slcasner, 2022-01-18:
Optimizing these log messages is a good idea.  But I would guess that
many of the situations where applications are depending on regular
updates would only occur when wifi connectivity is available.  Perhaps
in addition to the configuration option you propose there could be a
check for wifi connectivity?

dexterbg, 2022-01-18:
Good idea, probably can be combined into one mode selection: "Optimize data volume: No / On cellular / Always"

markwj, 2022-01-20:
Don't the apps depend on the 'F' message to determine connection status to the car? I can't really remember, but seem to recall something like that.

dexterbg, 2022-01-20:
Standard hold time of the last 'F' message on the server is 24 hours, and on connection of a new peer, a full retransmission of all messages is done. An App won't normally stay connected for longer than 24 hours and even if it does, it would keep the last 'F' message received, which then also still is valid (as the signal level then hasn't changed substantially).

So that should be no issue (it isn't on Android), unless the iOS App implements an expiration/invalidation scheme for messages not being repeated within some period -- does it?