openvehicles/Open-Vehicle-Monitoring-System-3 #690: Vehicle: VW e-Golf

dexterbg, 2022-01-29:
The VW e-Golf seems to have some similarities to the VW e-Up.

Chris van der Meijden (@devmarxx) began working on an adaptation, but lost access to a test car so cannot continue on this. He sent me the code tree for someone else to continue his work (attached).

He wrote you can already see some values he could access via the OBD port (diag CAN). The comfort CAN hasn't been decoded yet, it's supposed to run at 500 kbit (not 100 kbit as in the e-Up).

Volunteers welcome.

[vehicle_vwegolf.tar.gz](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/7964058/vehicle_vwegolf.tar.gz)

brewser123, 2022-01-29:
i have a 2015 egolf SEL Prem I can offer. if I need to order an OVMS device please advise and i can start on getting it ready

Jagganaut, 2022-02-06:
i have just received the new units for testing inside my ModelS and the VW eGolf.
Than we could closely work on that topic.

Beside that we intend to deliver an un-interrupted storage option for the OVMS data ...

qtwre, 2022-02-14:
Hi, I have an e-Golf and just received my OMVS unit. My plan was to try the e-Up profile but I'd be happy to help provide diagnostics or other help where possible.

brewser123, 2022-02-23:
I also have OVMS, is there someone who needs to access logs?

brewser123, 2022-02-23:
> 

did you connect with E-up as vehicle?

qtwre, 2022-02-27:
> > 
> 
> did you connect with E-up as vehicle?
> 
> > Hi, I have an e-Golf and just received my OMVS unit. My plan was to try the e-Up profile but I'd be happy to help provide diagnostics or other help where possible.

I did, though it did not read any values other than 12v battery voltage.

Edzo1988, 2022-03-01:
I have a 2019 SEL Premium in US I can support any type of testing. I will need to procure the hardware interfaces. Thanks for this project!

qtwre, 2022-05-02:
In the spirit of attempting progress of any sort, I compiled a firmware with the vehicle profile started in the original post. Maybe I'm doing something wrong but plugging the OVMS into my car didn't show any values other than 12v battery. I thought it would have shown something as some work had been done in the code for the e-Golf.

If anyone wants to load it on their device and tinker in the menus, [here is the bin](https://drive.google.com/file/d/1-J4ff63qhf5DDeZCSZepOoscTyaYquzK/view).

paultoben, 2022-07-05:
@qtwre did you plug into OBD port? I'm curious if the e-Golf has a Comfort CAN bus port under the seat like the e-Up.  My OVMS is on order and I plan to do more testing with my 2019 SEL.

brewser123, 2022-07-05:
> 

The e-Golf doesnt have the one under the seat like the eup from what I understand. On my 2015 egolf, the OBD port is left side below steering wheel.

pscegolf, 2022-07-05:
I am fighting with my OVMS for my eGolf too! Till now without better results. The connection to the comort bus I made by an connection to the cables near by the OBD port at the cable connection to the left door. But the firmware does not run. Only 12V battery voltage are readed.
[image]

pscegolf, 2022-07-05:
Do have anybody new information where the failure is? 
If I am right, it is able to connect the Komfort bus on all cables at the bus? I measure the voltage between the too cables and ground and it seams it is an CAN bus. And I think the connection should work. 
Where is the failure?

pscegolf, 2022-07-05:
Maybe the compiled -.bin had a failure? I tryed to create self a -.bin file, but it failed? What is the correct way to create the firmware? Where I found the best instruction to do this?

qtwre, 2022-07-05:
I need to dig up my research but there's a communication module (as used for the VW app) behind the gauge cluster. It has the connection to the Comfort CAN bus I was considering. 

I just recently wiped my computer and I fear I lost my firmware but I did have it compile and it still only showed the 12v battery. This was only connected through the OBD port though.

devmarxx, 2022-07-06:
When I was experimenting with the e-Golf I tapped into the door cables for the Comfort CAN and for OBD CAN in the OBD port. Here the cable colors:

Comfort CAN (Left Door)
Orange/Brown  Comfort CAN Hi
Orange/Green Comfort CAN Lo

OBD port
Orange/Red  OBD CAN Hi
Orange/Brown OBD CAN Lo (this is how I documented it, but I'm not sure now ???)
Red +12V
Black GND

If you are are using the standard firmware with VW e-UP as "car" you should see some OBD information (only if you are connected to the OBD port of course) like SOC. I never could see any Comfort CAN information. The first step is to set the rate of CAN3 to 500 kb/s in the e-UP code (e-UP uses 100 kb/s, e-Golf 500 kb/s on the Comfort CAN). But even then I could not see any communication on the Comfort CAN and that was the point where I gave up.

Success to all who continue the quest :-)

Regards

Chris
(devmarxx)

dexterbg, 2022-07-07:
> What is the correct way to create the firmware? Where I found the best instruction to do this?

- Developer Guide: https://docs.google.com/document/d/1q5M9Lb5jzQhJzPMnkMKwy4Es5YK12ACQejX_NWEixr0
- Mailing list: http://lists.openvehicles.com/mailman/listinfo/ovmsdev

Check out the list archive for some general introductions on CAN & OBD processing.

dexterbg, 2022-07-07:
- https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/255#issuecomment-835379478

qtwre, 2023-01-21:
This person managed to sniff the comfort bus recently and shared some messages they logged. He tapped in behind the HVAC control panel. 
https://www.vwvortex.com/threads/canbus-telematics-infotainment-interactions.9519863/post-115761620

I've also read it can be tapped from behind the infotainment brain in the glovebox to bypass the gateway and see a clear flow of messages.

pscegolf, 2023-01-22:
I logged the CAN bus too, but this is all where I had sucess. I coss check quickliy the log in the link from qtwre and it seam I logged the same ECUs at my eGolf.

Here you found my log:
[eGolf_CAN3_log3_20230102.txt](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/10474060/eGolf_CAN3_log3_20230102.txt)

I start to search DBC information to the eGolf in iNet, but it seams I fond not the right and correct one. I found this one but this does noch mach with my loged data.
[VW_MQB_2010.dbc.txt](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/10474067/VW_MQB_2010.dbc.txt)

I have no experience in CAN software programming, decoding,... , and I think this could be be biggest problem to come forward.
And it seams that not all, and especially not all eGolf ECUs are in the VW_MQB.dbc file?

I am still searching and working a litle bit on the project but, meanwhile I think, I am not able without further detailed informations and help on the eGolf CAN bus to adapt  the system to the OVMS.

JaredG1, 2023-05-09:
I have recently purchased an eGolf and have software development experience. My hope is that I can continue where this community left off. I would like to get the OBD version of OVMS working for the eGolf, then move on to the comfort CAN. I'm sure I will have a lot of questions :)

dexterbg, 2023-05-10:
Welcome Jared, don't hesitate to ask.

pscegolf, 2023-05-11:
Welcome Jared from my side too!
And hopefull you have more sucess with you software experience. Please keep us in the loop on your work and I hope that all your questions can solve,..

jrr, 2023-05-11:
I'm interested in following along (and maybe helping a little bit 🤏!). Where's the best place to collaborate? Forum / Mailing list / maybe a discord?

JaredG1, 2023-05-16:
Thanks for the support! I am currently waiting on the hardware to arrive, and starting to get the development environment set up. @jrr I will probably ping the OVMS devs [mailing list](http://lists.openvehicles.com/mailman/listinfo/ovmsdev) just to leverage all of the resources on there, but I would also be open to a Discord.

markwj, 2023-05-17:
> but I would also be open to a Discord.

There is a discord server available here:

https://discord.gg/eXp2FzYfJc

Largely unused at the moment, but available if you guys want to give it a spin.

zerblatt007, 2023-11-25:
I would very much want this to happen myself, and wondering what needs to be done. 

Our car still has support from the built-in system, but recon when 2-year plan we paid for just now runs out we are on our own. Response from the car has always been slow..
 
My main concern is the remote heating and reported charge that used to work in Home-Assistant, but we lost again with the new update in October.

dexterbg, 2023-11-26:
You probably should contact @JaredG1 on his current state, see his PR draft here:
https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/932

JDGreco, 2023-11-27:
Hello, I got a little distracted by other life events, but plan to continue working on firmware. Thanks for all of the work already done above!

thomasakarlsen, 2024-06-07:
With the 2G network shutdowns coming here in Europe the next couple years (here in Norway it starts next year), I've started my work on completely replacing the OCU with something of my own. I have decided to just completely ignore the OBD port, as quite frankly, on MQB platform vehicles the OBD port is completely useless unless the car is "ON" so getting remote control features via the OBD port does not work. Since the original OCU is only connected to the Comfort can bus, thats the one I've connected to.

The last couple months I've been banging my head looking at dumps of my canbus and reverse engineering source code from the MIB 2.5 radio in my car. There is a whole lot to how the different modules in the car communicate, but I have finally started to crack it.

So far I am able to:
- Wake the car up from sleep
- Modify the "minimum charge limit", climatisation temperature, "allow climatisation on battery" setting and change the length of climatisation (this is 10 minutes by default, I have increased this to 15 minutes with success)
- Start and stop climatisation 
- Make the car blink its lights and/or beep the horn

While I've not tested it, but should be able to:
- Insert/modify/remove any of the "Charge profiles" (the ones where you configure the upper charge limit, if it should pre-heat, etc. for your departure timers)

Some interesting facts about the charge profiles: While you can add your own profiles the car already has one that is hidden by default. This hidden profile (profile 0) is actually what is used as the global profile where you control the minimum charge limit, climatisation temperature, etc. So updating the "global" settings (minimum charge limit, temperature, etc.) is the exact same as updating any of your other profiles. 

Next on my list is charging and door lock controls before gathering up a list of all interesting data points.
While I do not own any OVMS hardware, and I am currently looking at making my own solution, I do want to share my findings here with you to help participate support for the e-Golf in OVMS.

I need to write up documentation for my findings, and I intend to explain as much as possible, which I will post very soon. Just wanted to share that things are happening! 😄 

Video of running the climatisation (file size too big to embed on github):
[https://www.jottacloud.com/share/21cpfuqr2221](https://www.jottacloud.com/share/21cpfuqr2221)

zerblatt007, 2024-06-07:
> <openvehicles/Open-Vehicle-Monitoring-System-3/issues/690/2154563032@
> github.com>

This is great news! Looking forward to it.. :)

jaymemaurice, 2024-07-19:
Looking forward to your work @thomasakarlsen - will it be posted here or somewhere else?

thomasakarlsen, 2024-07-19:
> Looking forward to your work @thomasakarlsen - will it be posted here or somewhere else?
> 
> 
> Ooo found the activity in the Discord. Thanks!
> 
> 
> 

I have started to document it here: [https://github.com/thomasakarlsen/e-golf-comfort-can](https://github.com/thomasakarlsen/e-golf-comfort-can)

But as life often does, it has come in the way so progress has been a bit slow writing it all up.

dexterbg, 2025-08-05:
How to setup for building is documented in the developer guide, see links:
https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3?tab=readme-ov-file#links

Then maybe have a look at Chris' unfinished work from the OP here, and maybe at the Hyundai Ioniq vFL, as that's reasonably concise yet showcases most commonly used elements.

A new option to enable implementing vehicles purely in Javascript is in development by Michael (@frogonwheels). He certainly won't object if you volunteer as a guinea pig ;-)

frogonwheels, 2025-08-06:
> <openvehicles/Open-Vehicle-Monitoring-System-3/issues/690/3155899211@

Yes indeed!  All the pieces I've done so far are in the latest development
version,  however I will be responsive to suggestions.

One piece that needs doing is how the battery information and stats  is
collected via the js interface, but if you want to go that  route I'm keen
to implement it.

//.ichael

Nueueuet, 2026-08-15:
@kristoficko — back in June you offered to provide official e-Golf pictures for the Android app. Is that still on the table? Nobody took you up on it and it would be genuinely useful now.

The vehicle still falls back to e-Up images, which is what my car shows today. What the app needs, per colour, in `app/src/main/res/drawable-nodpi/`:

| file | size | view |
|---|---|---|
| `car_egolf_<colour>.png` | 615 × 377 | three-quarter, home screen and the picker |
| `ol_car_egolf_<colour>.png` | 250 × 485 | **top-down**, used for the climate and controls tabs |
| `map_car_egolf_<colour>.png` | 49 × 96 | top-down, map marker |

Transparent background, matching the existing `car_vwup_*` set. The top-down is the awkward one — a three-quarter shot is easy to come by, a clean overhead view is not.

One thing worth settling before anyone spends effort: whether images from the VW portal may actually be redistributed under this repo's licence. If not, own photographs are the cleaner route and I'm taking some of my own car in parallel. Either way I'm happy to do the cropping, scaling and cut-out — a usable source is the part I can't produce.

Context, in case it's useful: I've been adding e-Golf climate controls to the app — a target temperature slider and a toggle for pre-conditioning without the charging cable, both driving the car's stored BatteryControl profile through the module. Working on the car; the images are the last thing that still looks wrong.

kristoficko, 2026-08-17:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!
> <openvehicles/Open-Vehicle-Monitoring-System-3/issues/690/5301685561@

Hi,

the official pictures (used by VW web and mobile apps) are stored in
https://media.volkswagen.com/Vilma/ (looks like CloudFront + S3 bucket).
Unfortunately, the list is truncated to 1000 results and I didn't find a
working way to paginate through to the next page(s).

From the web I was able to obtain pictures for white e-golf on Astana
wheels and Madrid wheels, just because I have valid VIN number(s) and I was
able to register these 2 cars in the original app. There's definitely
different colors available in the bucket...

White 2018 with Astana:
https://media.volkswagen.com/Vilma/V/BE2/2018/Back_Right/2eafe418ec97d13c12510509e90b9182d7181931709ce447ffdca3ccff42c4c5.png
https://media.volkswagen.com/Vilma/V/BE2/2018/Side_Left/ec8305f9a631c1e78eb7c132658aaacc12af24197533cd32dd33068082a475a0.png
https://media.volkswagen.com/Vilma/V/BE2/2018/Front_Right/0c65f253c23e7d1c71ec029108c4e5bea46835c7c231e2cd4ffc354b9f471efd.png
https://media.volkswagen.com/Vilma/V/BE2/2018/Front_Left/f1b6b4587154b08ef17d0bd1e18c241b75c61cf050593009eec873fa8069a92c.png
https://media.volkswagen.com/Vilma/V/BE2/2018/Front_Center/be8d2123b73b51cd337e75a02474805ef9636220514c2887d14c4190a6580cd5.png
https://media.volkswagen.com/Vilma/V/BE2/2018/Back_Left/c8cc56f9d798d8bf968eee1e64b97659b86b226a5a81c9450b15df1a5aad92e8.png
https://media.volkswagen.com/Vilma/V/BE2/2018/Side_Right/d15aacebbbdd89bca948990248013682df3869fd7ea4c565775f99078e3ad805.png

White 2018 with Madrid:
https://media.volkswagen.com/Vilma/V/BE2/2018/Side_Left/41545d306edcea50de0332b551cc9d86e784af02406898f66587d76df628bfa2.png
https://media.volkswagen.com/Vilma/V/BE2/2018/Front_Right/f69b1d54a72742019afa5e0f969c6c27408f51c9af459c423354d938ff46b39c.png
https://media.volkswagen.com/Vilma/V/BE2/2018/Front_Left/7751bcb7115d1f3531df07d77845a8c646055ee831daa828a62bd985d1791d53.png
https://media.volkswagen.com/Vilma/V/BE2/2018/Front_Center/7dde2499682a4d11f8323a1f2b9f26718cc209d3814b1646b92dc751419088da.png
https://media.volkswagen.com/Vilma/V/BE2/2018/Back_Right/1c548b722cd8003119f4cce5cde259fe833744e6ef6b2047c7ae7007d4a2462f.png
https://media.volkswagen.com/Vilma/V/BE2/2018/Back_Left/dfefd28214ca29e8fedd449ec39125bca46212bf5d1e3b755f67df117dca7c69.png
https://media.volkswagen.com/Vilma/V/BE2/2018/Side_Right/a4b0e37c192db862ccbcd8ae31b21a3dfbe50f42b9b2ce7ae0a5dae14341aee4.png

Black 2020 with Astana (URL provided by Gemini):
https://media.volkswagen.com/Vilma/V/BE2/2020/Front_Right/4535cf3902b72e2370618e1ffdf01aabcd7cc4e79c9af261def6a7c5c8012d68.png

jarnodrenth, 2026-08-17:
Hello Everyone,

I bought an OVMS module and installed it in my 2020 Golf E. I can see all the information in the app and on the portal. I installed the EDGE firmware. Unfortunately, I can’t seem to get the climate control to start or stop. According to the website, this feature should be supported by now. Can anyone point me in the right direction? Thanks so much in advance.

Nueueuet, 2026-08-19:
@jarnodrenth Short version: your firmware can most likely already do it — it is the app that will not ask for it.

The e-Golf climate control has been in edge for a while (SCjona's work in `vehicle_vwegolf`), so a current EDGE build has the car side. What is missing is the announcement the app waits for. The app only enables its climate button when the car reports V2 command 26, and no V3 module reports anything at all, because `OvmsServerV2::TransmitMsgCapabilities()` is an empty stub upstream. The app works around that with a hard-coded list of car types — Leaf, Smart, e-Up, ZOE Ph2, Volt/Ampera and a few others — and `VWEG` is not on it. So the button stays greyed out for us no matter how good the firmware is.

Two-minute check, nothing to flash. Module web UI, Tools → Shell:

```
climatecontrol on
climatecontrol off
```

If the car starts and stops conditioning from there, the firmware side is fine and everything below is only about getting the button to work.

What I would run: SCjona's branch from #1496. It is the current, self-contained implementation of the e-Golf's BAP control and it also gets you the charge current limit:

https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/1496

For the app button, the module has to announce the capability. That is one small commit, not merged anywhere, so you would carry it on top yourself:

https://github.com/Nueueuet/Open-Vehicle-Monitoring-System-3/commit/749b74fa

Both mean building from source. If you have not done that before: the esp-idf setup, the fork, the branch and the flashing are exactly the kind of thing you can hand to whatever AI assistant you use — that is how I got from never having touched an ESP32 to a module running my own build, and it is far quicker than reading yourself into the toolchain first.

Mine is a 2016, SCjona's is newer, and the climate control works on both, so your 2020 should be in good company.

jarnodrenth, 2026-08-25:
@Nueueuet Thank you so much for your response! 
It’s great that you’re all so actively working on this.

I tested the following:
climate control on
climate control off

It works! I’m really happy about that.

I’m very interested in the solutions you suggested. Unfortunately, I don’t have enough computer knowledge to do this myself. Is this something you’d be willing to share with me? Because if I understand you correctly, this is a special version of the firmware with enhancements from you and SCjona. I’d be happy to help with testing if possible. Will these changes be implemented in upcoming updates? In other words, do I need to be a little patient?

Is there any other way I can contribute?

danielklejnstrup, 2026-08-28:
I'm curious as to whether or not I should be able to use the VW e-Golf profile, for getting SoC and other metrics from the ODBII-port.
Just got my OVMS today, and set it up as best I could.
I've reached a point, where I can now read the 12V battery voltage, and the GPS-coordinates from the modules webinterface, as well as from https://dexters-web.de
I guess it's the module itself monitoring the supply voltage, and getting the GPS coordinates using the attached antenna.

As far as getting other metrics, I'm getting some reasonable ones:

v.e.charging12v = true
v.e.temp = 16
v.type = VWEG
v.vin = MY_EGOLFS_VIN

And some not so reasonable:

v.b.soc = 0
v.b.temp = 215 (maybe 21,5?)

Module:
Last boot was 781 second(s) ago
Time at boot: 2026-08-28 20:01:38 CEST
  This is reset #3 since last power cycle
  Detected boot reason: SoftReset (12/12)
  Reset reason: esp_restart (3)
  Crash counters: 0 total, 0 early
  Auto init level: Full
Hardware:          OVMS WIFI BLE BT cores=2 rev=ESP32/3
Firmware:          3.3.006-172-g362f87425/ota_1/main (build idf v3.3.4-854-g9063c8662c Aug  1 2026 09:47:50, product v3.3-5)
Partition type:    v3-35 (ota1, ota2, no factory, maximized store)
Partition table:   0x8000
Running partition: ota_1
Boot partition:    ota_1
OTA_O image:       3.3.006-172-g362f87425
OTA_1 image:       3.3.006-172-g362f87425

I'd prefer getting up and running on the ODBII-port, before touching the wiring in the car.
Any good advice?

GerjanT, 2026-08-29:
> the latest edge firmware build (3.3.006-259-g1aa1daff1 build at Sat Aug 22 17:04:50 CEST 2026) has my climate and charge control implementation (read the docs if you plan to use charge control).
> 
> regarding the app side: you can see [my old comment](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/690#issuecomment-4895518133) about how to get it working now, but i'd just wait a bit more till the app officially supports it. easiest path if you don't want to wait would be to use the push notifications screen command input, its practically like the shell in the web UI. `climatecontrol on` / `climatecontrol off`. if it fails you get a notification about it.

I'm still waiting for my connectors to build the cable so I can connect OVMS into my e-Golf (2020). I still have a Carnet subscription until November this year. Are there specific things you want to see that I can reverse engineer?

danielklejnstrup, 2026-08-29:
I've noticed that the e-Golf implementation doesn't seem to use the OBD-BUS at all.
For "onboarding" purposes, for new users like myself, I'd prefer to be able to have some metrics working, before messing with the internal wiring of my car.
Is there an easy way to run a minimal version of the implementation, where climate and charging is not supported?
I wonder if using a rewired version the supplied OBD-cable, to connect the OBD-BUS to can2 or can3 would make the existing implementation work for the metrics that can be accessed across the J533 gareway?
My primary goal is to replace the connectedcars.io module (minVolkswagen) since I plan on having my car serviced at a workshop not renewing that service, from now on.
I've been used to ignition-on-only metrics for 2 years now, and any additional features are welcome, but not strictly needed for me.
Would the admins accept a VWEGOBD vehicle, essentially being a stripped down version of VWEG - or would it make more sense to have a setting in the VWEG implementation, that could toggle the used cans, supported metrics and functions?

danielklejnstrup, 2026-08-30:
> I'm still waiting for my connectors to build the cable so I can connect OVMS into my e-Golf (2020). I still have a Carnet subscription until November this year. Are there specific things you want to [...]

Ended up ordering this:
https://www.aliexpress.com/item/1005012982921300.html
I'm planning to connect the busses from that, and power from the OBD-port, or somewhere else - not having to put together that harness myself, is hopefully better for my peace of mind :)

SCjona, 2026-08-30:
> Ended up ordering this: https://www.aliexpress.com/item/1005012982921300.html I'm planning to connect the busses from that, and power from the OBD-port, or somewhere else - not having to put together that harness myself, is hopefully better for my peace of mind :)

I bought one of these extension harnesses too. Problem was that it had the wrong CAN busses extended (the connector carries them all as it is connected to the CAN gateway) and I had to re-pin most of the thing. Took me forever. The connector also carries 12V and Ground, so you can pull these from there too.

GerjanT, 2026-08-30:
> 
> Ended up ordering this:
> https://www.aliexpress.com/item/1005012982921300.html
> I'm planning to connect the busses from that, and power from the OBD-port, or somewhere else - not having to put together that harness myself, is hopefully better for my peace of mind :)

This prefabricated cable doesn't have 12 volt. You want to pick the 12 volt from the ODB-port? You can consider to splice the 12 volt

zerblatt007, 2026-08-31:
> This prefabricated cable doesn't have 12 volt. You want to pick the 12 volt from the ODB-port? You can consider to splice the 12 volt

It does not, but it is easy to modify to get it. There also was some pins I had to move around, but I do not remember or document it, sadly. The documentation on the website should followed..

[image]

danielklejnstrup, 2026-08-31:
> It does not, but it is easy to modify to get it. There also was some pins I had to move around, but I do not remember or document it, sadly. The documentation on the website should followed..

Nice 👌
I wonder if under the steering wheel is the overall optimal position, or simply the easiest to get the job done? Have you guys been fiddling with other placements? The audio amplifier in the trunk is probably only connected to the infotainment CAN?
The connectedcars.io module was connected to an OBD extension cable, and placed to the left of the steering wheel, behind a flat panel hidden by the closed door.
I guess I should just forget about the GPS antenna, and rely on the GPS coordinates from the KCAN.

I struggle to find a good placement for the GSM/WiFi antenna, to get the OVMS to connect reliably to my home wifi, when the car is parked at home. Any suggestions?

mikus-ancevs, 2026-09-08:
Does enyone else has problem with OVMS hanging and becoming unresponsive?
I restarted by unplugging yesterday, did not drive. Started to drive and after 10 mins values stopped updating.
WiFi AP is visible, can connect, but config page never loads.
I am on edge builds, but maybe not on very latest, cannot confirm right now.

kristoficko, 2026-09-08:
> <openvehicles/Open-Vehicle-Monitoring-System-3/issues/690/5579572295@

Hi,

I'm on edge version 3.3.006-275-g85074a0ae and experienced a stuck OVMS
during the weekend. UI was also timing out, wifi client connected to home
wifi.

Ended up crawling under the dashboard and unplugging/re-plugging the unit.
Since then it's online again.

Mike

GerjanT, 2026-09-08:
> Nice 👌 I wonder if under the steering wheel is the overall optimal position, or simply the easiest to get the job done? Have you guys been fiddling with other placements? The audio amplifier in the [...]

I installed the module this weekend. Fortunately, I have a relatively long serial cable. I secured the module with cable ties just behind the OBD-II connector, which makes it relatively easy to access the SD card. I routed the antenna cables from the side—near the removable panel—up to the top of the dashboard.

Spoiledstuff, 2026-09-08:
I have been using my unit through the Home Assistant integration, and it's been working great; it avoids all the issues with the OVMS app. I was just wondering whether lock/unlock has been worked on yet? And is there any way to donate to the people who have helped get this working? Thank you, guys, for all your hard work!!

danielklejnstrup, 2026-09-09:
Just received this: ( https://www.aliexpress.com/item/1005012982921300.html )
[image]
[image]
I'm quite happy with the quality, but as you guys mentioned, it doesn't splice the powertrain or diagnostics CAN.
The 2-pin connectors are connected to pair (5,15) (7,17) and (8,18) which seems to be CAN_CONVENIENCE, CAN_EXTENDED and CAN_RUNNINGGEAR.

Pin 4 is NOT connected in either end of the extension cable, but it seems safe to omit, since it can't find a pinout diagram where it's used.

Do you have any advise, as to which tools to use, to rearrange the connectors? (terminal removal tool, or whatever it's called)

oliskippy, 2026-09-09:
I connected successfully the module and read the first metrics.
GPS data from the car (supposed to come from the KCAN, not via modem antenna) is missing. 
v.p.gpslock = false
v.p.latitude = 0  --> no position

Any one managed to get data for an e-Golf ?

kristoficko, 2026-09-09:
> <openvehicles/Open-Vehicle-Monitoring-System-3/issues/690/5608178413@

Works reliably for me...

danielklejnstrup, 2026-09-10:
> > > Ended up ordering this:
> > > https://www.aliexpress.com/item/1005012982921300.html
> > > I'm planning to connect the busses from that, and power from the OBD-port, or somewhere else - not having to put together that harness myself, is hopefully better for my peace of mind :)
> > 
> > 
> > This prefabricated cable doesn't have 12 volt. You want to pick the 12 volt from the ODB-port? You can consider to splice the 12 volt
> 
> It does not, but it is easy to modify to get it. There also was some pins I had to move around, but I do not remember or document it, sadly. The documentation on the website should followed..
> <img alt="Image" width="480" height="640" src="https://private-user-images.githubusercontent.com/6605239/643390648-e0c32050-aeb8-4c4f-9ff4-4677d049bef0.jpeg?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3ODkwMjM0MTIsIm5iZiI6MTc4OTAyMzExMiwicGF0aCI6Ii82NjA1MjM5LzY0MzM5MDY0OC1lMGMzMjA1MC1hZWI4LTRjNGYtOWZmNC00Njc3ZDA0OWJlZjAuanBlZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjA5MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwOTEwVDA2NTE1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTViOTc4NDhkY2E4MTFhYWU0OTk5N2VkODViZTNmNzk3NWEzNmU4YzNiZDMwMGU4MmRhNTNiM2UyZTI5MTNiMTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JnJlc3BvbnNlLWNvbnRlbnQtdHlwZT1pbWFnZSUyRmpwZWcifQ.Bb49nilS3a84fENg1CDMaWSlXgljJA7aE5gQkj0fiJ4">

> > > Ended up ordering this:
> > > https://www.aliexpress.com/item/1005012982921300.html
> > > I'm planning to connect the busses from that, and power from the OBD-port, or somewhere else - not having to put together that harness myself, is hopefully better for my peace of mind :)
> > 
> > 
> > This prefabricated cable doesn't have 12 volt. You want to pick the 12 volt from the ODB-port? You can consider to splice the 12 volt
> 
> It does not, but it is easy to modify to get it. There also was some pins I had to move around, but I do not remember or document it, sadly. The documentation on the website should followed..
> [<img alt="Image" width="480" height="640" src="https://private-user-images.githubusercontent.com/6605239/643390648-e0c32050-aeb8-4c4f-9ff4-4677d049bef0.jpeg?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3ODkwNDI4OTcsIm5iZiI6MTc4OTA0MjU5NywicGF0aCI6Ii82NjA1MjM5LzY0MzM5MDY0OC1lMGMzMjA1MC1hZWI4LTRjNGYtOWZmNC00Njc3ZDA0OWJlZjAuanBlZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjA5MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwOTEwVDEyMTYzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNlMmMzYzI2NmRiYWY0MDg4Yzg2MWQwMTMyNGIzNGQ0ZmMxZTk3NzljN2FiNzA3Nzk4ZWY2NTI0ZDMxZWRmZmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JnJlc3BvbnNlLWNvbnRlbnQtdHlwZT1pbWFnZSUyRmpwZWcifQ.g9WgV0znLQ9ObHZdpOGSctIZ65PiKL03cBvfv7X1lAA">](https://private-user-images.githubusercontent.com/6605239/643390648-e0c32050-aeb8-4c4f-9ff4-4677d049bef0.jpeg?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3ODkwMjM0MTIsIm5iZiI6MTc4OTAyMzExMiwicGF0aCI6Ii82NjA1MjM5LzY0MzM5MDY0OC1lMGMzMjA1MC1hZWI4LTRjNGYtOWZmNC00Njc3ZDA0OWJlZjAuanBlZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjA5MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwOTEwVDA2NTE1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTViOTc4NDhkY2E4MTFhYWU0OTk5N2VkODViZTNmNzk3NWEzNmU4YzNiZDMwMGU4MmRhNTNiM2UyZTI5MTNiMTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JnJlc3BvbnNlLWNvbnRlbnQtdHlwZT1pbWFnZSUyRmpwZWcifQ.Bb49nilS3a84fENg1CDMaWSlXgljJA7aE5gQkj0fiJ4)

I'm putting together a guide to modify these aliexpress harnesses, to fit the needs for connecting the OVMS.
Looking at the egolf implementation, it seems the OBD CAN is not utilized at all (can1):

vehicle_vwegolf.cpp:
 line 54:  RegisterCanBus(2, CAN_MODE_LISTEN, CAN_SPEED_500KBPS);  // FCAN — powertrain (read-only)
 line 55:  RegisterCanBus(3, CAN_MODE_ACTIVE, CAN_SPEED_500KBPS);  // KCAN — comfort / clima

So I'm considering only connecting the other 2 busses, and repurpose the third splice for power.
Can you confirm that the OBD CAN can be disconnected, without loosing metrics/functionality?

SCjona, 2026-09-10:
At the moment OBD CAN is not used. However it could be used for enhanced metrics (the ones that aren't broadcasted on the bus) in the future. I re-pinned all three CAN pairs and manually spliced 12V and GND for my setup.
It would be the most user friendly if we could just buy ready made cables for the e-Golf OVMS integration. but probably not enough demand to justify the manufacturing costs.

danielklejnstrup, 2026-09-11:
> At the moment OBD CAN is not used. However it could be used for enhanced metrics (the ones that aren't broadcasted on the bus) in the future. I re-pinned all three CAN pairs and manually spliced 12V [...]

@SCjona - I guess I'll connect all 3 then ;)
Any precausions (other than removing the 12V battery negative terminal, I suppose) while installing the adaptor?

danielklejnstrup, 2026-09-12:
Okay...
@zerblatt007
@SCjona
@GerjanT
Last hurdle. I'm still a little suspicious about the missing pin 4 connection in the extension cable.
Did any of you install the adaptor with pin 4 not connected without experiencing issues?

GerjanT, 2026-09-12:
I installed all pins

danielklejnstrup, 2026-09-12:
> Okay... [@zerblatt007](https://github.com/zerblatt007) [@SCjona](https://github.com/SCjona) [@GerjanT](https://github.com/GerjanT) Last hurdle. I'm still a little suspicious about the missing pin 4 [...]

Right. I checked a bunch of harnesses on aliexpress listing compatibility with Golf MK7, and they all seem to be missing pin 4.
It seems to be a redundant(?) ground line, so I went ahead and tested it out.
After reconnecting the 12V battery negative terminal, the car threw about every possible error code in the instrument cluster.
I couldn't turn on the ignition, to the "READY" state, before retrying ignition a couple of times. Errors remained, but I could now drive. Took the car for a 1km ride, and returned to our driveway. After turning the ignition to the off position, and back on, everything seems to be back at normal :)
And now I have the OVMS sending payloads. AWESOME!
Thanks for the help along the way.

[48 more replies]