openvehicles/Open-Vehicle-Monitoring-System-3 #691: Nissan Leaf: charget

drc38, 2022-01-30:
[image]

mjkapkan, 2022-01-30:
Hi, what do you have in mind by `SuspendedEV`?

drc38, 2022-01-30:
> Hi, what do you have in mind by `SuspendedEV`?

`SuspendedEV` is when the vehicle has stopped the charge for some reason. In ovms the logic will set it to `timerwait`. 

Initially I thought the issue may have been the new SoC logic but it could be the start charge message does not work for AZE0 Leafs. Does the start charge button work for you after stopping the charge with the MITM code?

mjkapkan, 2022-01-30:
Understood. The stop - start works perfectly on my 2013 MY. I heard it was working fine for @dalathegreat as well. Daniel can you double check when you have a chance?

drc38, 2022-01-31:
@mjkapkan I've investigated a bit more and it seems it may be related to the new SoC code. To replicate the issue set a SoC limit and tolerance, wait for the charge to stop, raise SoC limit to above tolerance to initiate charge again (does not charge on my Leaf nor does using Start Charge).

If the SoC limit is not reached Stop Charge and then Start Charge work as expected.

mjkapkan, 2022-01-31:
Hm, I tested this many times. Was working as expected. If you set the tolerance to 0 does it still do that? Also do you use charge to SOC, charge to Range or both? If so are you sure those other parameters did not conflict? You also should get a notification of why the charge is being started / stopped. What do you get in this case?

drc38, 2022-01-31:
I am only using the charge to SoC, I will have to wait as the car is currently charged to test what happens is a tolerance is not set.

drc38, 2022-02-02:
Looks like it may have been a glitch, so will close. Was working as expected today, see below:
[image]

closed 2022-02-02 by drc38

drc38, 2022-02-02:
This was from the previous attempt where it started to charge then stopped again after c. 1% charge:
[image]

mjkapkan, 2022-02-03:
I think what happened was that allowed soc drop was set and thus prevented charge from starting. I will make a change to allow charge to start ignoring the allowed drop of the target SOC/Range is updated.

mjkapkan, 2022-02-03:
Created #696