openvehicles/Open-Vehicle-Monitoring-System-3 #693: Move to platformio

kallisti5, 2022-02-02:
Just throwing this one out there. It's a potentially big project, but is a lot more common pattern used by ESP32 projects.

[PlatformIO](https://platformio.org) fully supports:
* Getting, installing toolchains and requirements cross-platform (Windows, Mac, Linux)
* [Supplying libraries from their wide selection](https://registry.platformio.org)
* Supports multiple IDE's
* Supports integration testing + unit tests

nOe of the biggest hurdles around getting started hacking on OVMS was figuring out

kallisti5, 2022-02-02:
To build a project in platformio you:

* Install platformio
* Enter the project's directory
* run ```pio run``` to pull down toolchains, compile.
* run ```pio run -t upload``` to upload to your device

leres, 2022-02-02:
I've used platformio for [Tasmota](https://tasmota.github.io/docs/) and I have to say I'm not a fan. It's heavyweight and I would lose the ability to develop OVMS under FreeBSD if it was adopted for this project so I'm not in favor of switching.

markwj, 2022-02-04:
Probably. better to raise this issue on the developers mailing list, rather than a GitHub issue.

In general, we need to do something regarding the base platform, as what we have is getting dated and support for newer devices will require the latest ESP IDF platform. But converting either to some abstraction platform like platformio, or the raw latest ESP IDF, will not be at all simple.

kallisti5, 2022-07-25:
> I've used platformio for [Tasmota](https://tasmota.github.io/docs/) and I have to say I'm not a fan. It's heavyweight and I would lose the ability to develop OVMS under FreeBSD if it was adopted for this project so I'm not in favor of switching.

Last I checked FreeBSD worked just fine.  Here's something to reference:
  https://community.platformio.org/t/platformio-on-freebsd/16472/2

It's probably just "less used" on FreeBSD so it needs a little help sometimes.  As for being heavy-weight, that's not really an argument.  It's a pretty simple python system to setup dev environments.

Spending an hour or two every few months trying to re-setup the ESP-IDF is pretty heavyweight in comparison.