openvehicles/Open-Vehicle-Monitoring-System-3 #696: NL: Enhance Charge to Limit Feature

mjkapkan, 2022-02-03:
Currently when charge limit is updated the system will not start the charge if the new limit is less than current charge + allowed drop.

So if the limit is updated a small amount and the allowed drop is less than that, the charge will not start. 

To fix this, allowed drop value can be ignored once when the charge limit is updated.

Lenbok, 2022-02-19:
I found the current behaviour of the charge to limit feature confusing (see https://www.openvehicles.com/comment/8109#comment-8109).

My background is from using the built-in NL charge to 80% feature (but admittedly I've never noticed if the built-in version does periodic "top-ups"). I thought I could simply use OVMS as an equivalent with a user specified limit, so it threw me that I needed to **enable** a different checkbox to allow me to effectively **disable** automatic charge starting (particularly as the Web UI for the sufficient SOC slider says nothing about **starting** charging). My particular use case is also to have charging actually initiated via home automation to be coordinated with solar PV output, so that may have biased my expectation that the SOC limit feature would not **initiate** charging.

It would be much clearer to the user if the "Limit to SOC" (or range) only controlled when charging is stopped (i.e. it should not enable automatic starting of charging at all), and the enabling of automatic starting is separately controlled by the "Allowed SOC (or range) drop" methods. I.e. if you have limit to SOC set but have NOT enabled allowed SOC (or range) drop, the leaf would never start charging automatically but will stop charging once the target is reached. 

If the user wants OVMS to automatically initiate charging as well, then they can enable (checkbox) either the SOC or range drop features and set the limit appropriately.

Lenbok, 2022-02-19:
Additionally, there seems a bug in the implementation. I currently have the SOC limit set to 80% and the allowed SOC drop set to 30%, however yesterday it automatically started a charge when the drop was only 18%:

[image]

mjkapkan, 2022-02-19:
@Lenbok Thanks for the feedback. It would be of course nice to have more flexibility. Originally it was made to start charging to allow user to set higher SOC limit than he had previously. As this was just the first implementation, it is not perfect. Regarding the bug. Please submit all the charge limit metrics before and after the issue happened, so I can try to replicate this.

As for further enhancements, I am currently working on a plugin that will allow to do this for all cars, not just the Leaf. So more features and flexibility will be available with it. All in all, all feedback is welcome!

Lenbok, 2022-02-19:
@mjkapkan I'm kind of new to this, so not sure how to get all the metrics from before / after, but I have enabled some SD card logging which hopefully is sufficient. Starting just before the leaf decided to start charging:
```
2022-02-19 19:39:08.301 NZDT I (11417881) ovms-server-v2: Incoming Msg: MP-0 AFA
2022-02-19 19:39:08.311 NZDT I (11417891) ovms-server-v2: Send MP-0 a
2022-02-19 19:43:38.571 NZDT I (11688151) ovms-server-v2: Send MP-0 S62.0,K,0,0.00,stopped,standard,60,44,0,0,0,0,0,21,0,0,0,0,49.36,0,0,0,80,-1,0,0,0,0,0,0,0,0.00,383.50,75,0.00,100.00,0.00,0.0
2022-02-19 19:43:38.571 NZDT I (11688151) ovms-server-v2: Send MP-0 D0,0,5,15,16.1111,21,0,931373,0,85933,22,0,0,0,11.75,0,12.5,0,0,0,19.4444
2022-02-19 19:43:53.581 NZDT I (11703161) housekeeping: 2022-02-19 19:43:52 NZDT (RAM: 8b=75772-76484 32b=12344 SPI=3384656-3419456)
2022-02-19 19:48:53.581 NZDT I (12003161) housekeeping: 2022-02-19 19:48:52 NZDT (RAM: 8b=75772-76484 32b=12344 SPI=3384656-3419424)
2022-02-19 19:50:13.571 NZDT I (12083151) v-nissanleaf: RemoteCommandHandler
2022-02-19 19:50:13.571 NZDT I (12083151) v-nissanleaf: Sending Wakeup Frame
2022-02-19 19:50:13.571 NZDT I (12083151) ovms-server-v2: Send MP-0 PICharge level below target (62%)
2022-02-19 19:50:13.671 NZDT I (12083251) v-nissanleaf: RemoteCommandTimer 24
2022-02-19 19:50:13.671 NZDT I (12083251) v-nissanleaf: OLD TCU on EV Bus
[...]
2022-02-19 19:50:15.971 NZDT I (12085551) v-nissanleaf: OLD TCU on EV Bus
2022-02-19 19:50:15.971 NZDT I (12085551) v-nissanleaf: Sending Wakeup Frame
2022-02-19 19:50:15.971 NZDT I (12085551) v-nissanleaf: Start Charging
2022-02-19 19:50:16.071 NZDT I (12085651) v-nissanleaf: RemoteCommandTimer 0
2022-02-19 19:50:24.571 NZDT I (12094151) powermgmt: Charging 12V battery..
2022-02-19 19:50:24.581 NZDT I (12094161) ovms-server-v2: Send MP-0 PIStandard - Charging|220.0V/12.7A 10.2km/h|Full: 12:23h|80%: 5:52h|Charged: 0.0kWh|SOC: 62.0%|Ideal range: 61km|Est. range: 45km|ODO: 93137.3km|CAC: 49.4Ah|SOH: 75%
2022-02-19 19:53:39.571 NZDT I (12289151) ovms-server-v2: Send MP-0 S63.0,K,220,12.73,charging,standard,61,45,12,203,0,1,2,1,0,0,0,0,49.36,115,0,0,80,-1,0,0,0,0,52,72,0,2.70,385.50,75,2.80,96.25,-7.00,10.2
2022-02-19 19:53:39.581 NZDT I (12289161) ovms-server-v2: Send MP-0 D28,0,5,15.5556,16.6667,21,0,931373,0,86534,22,0,1,1,14.13,0,12.5,16,0,0,20.5556
2022-02-19 19:53:53.581 NZDT I (12303161) housekeeping: 2022-02-19 19:53:52 NZDT (RAM: 8b=75772-76332 32b=12344 SPI=3384656-3417632)
2022-02-19 19:54:10.261 NZDT I (12319841) ovms-server-v2: Incoming Msg: MP-0 AFA
2022-02-19 19:54:10.271 NZDT I (12319851) ovms-server-v2: Send MP-0 a
2022-02-19 19:58:53.581 NZDT I (12603161) housekeeping: 2022-02-19 19:58:52 NZDT (RAM: 8b=75772-76332 32b=12344 SPI=3384656-3417120)
2022-02-19 20:03:40.571 NZDT I (12890151) ovms-server-v2: Send MP-0 S67.0,K,220,12.73,charging,standard,65,48,12,804,0,5,2,1,0,0,0,0,49.36,97,0,0,80,-1,0,0,0,0,38,72,0,2.89,386.00,75,2.80,96.50,-7.50,11.0
2022-02-19 20:03:40.581 NZDT I (12890161) ovms-server-v2: Send MP-0 D28,0,5,16.6667,17.2222,21,0,931373,0,87135,22,0,1,1,14.07,0,12.5,16,0,0,20.5556
2022-02-19 20:03:53.581 NZDT I (12903161) housekeeping: 2022-02-19 20:03:52 NZDT (RAM: 8b=75772-76332 32b=12344 SPI=3384656-3417016)
2022-02-19 20:05:34.571 NZDT I (13004151) powermgmt: No longer charging 12V battery..
2022-02-19 20:08:53.581 NZDT I (13203161) housekeeping: 2022-02-19 20:08:52 NZDT (RAM: 8b=75772-76332 32b=12344 SPI=3384656-3417120)
2022-02-19 20:09:11.301 NZDT I (13220881) ovms-server-v2: Incoming Msg: MP-0 AFA
2022-02-19 20:09:11.311 NZDT I (13220891) ovms-server-v2: Send MP-0 a
2022-02-19 20:13:41.571 NZDT I (13491151) ovms-server-v2: Send MP-0 S72.0,K,220,12.73,charging,standard,70,51,12,1405,0,10,2,1,0,0,0,0,49.36,82,0,0,80,-1,0,0,0,0,23,71,0,2.71,387.00,75,2.80,96.75,-7.00,10.1
2022-02-19 20:13:41.581 NZDT I (13491161) ovms-server-v2: Send MP-0 D28,0,5,17.2222,18.3333,22,0,931373,0,87736,22.5,0,1,1,12.62,0,12.5,0,0,0,20.5556
2022-02-19 20:13:41.581 NZDT I (13491161) ovms-server-v2: Send MP-0 F3.3.001-282-gbd4919ca/ota_0/edge (build idf v3.3.4-848-g1ff5e24b1 Feb 12 2022 00:04:22),,15,0,NL,Rancho Relaxo,-1,-1,OVMS WIFI BLE BT cores=2 rev=ESP32/3; MODEM SIM7600
2022-02-19 20:13:53.581 NZDT I (13503161) housekeeping: 2022-02-19 20:13:52 NZDT (RAM: 8b=75772-76332 32b=12344 SPI=3384656-3417156)
2022-02-19 20:16:58.551 NZDT I (13688141) ovms-server-v2: Incoming Msg: MP-0 Z1
2022-02-19 20:16:58.551 NZDT I (13688141) ovms-server-v2: One or more peers have connected
2022-02-19 20:16:58.561 NZDT I (13688151) ovms-server-v2: Send MP-0 S73.0,K,220,12.73,charging,standard,71,52,12,1602,0,12,2,1,0,0,0,0,49.36,79,0,0,80,-1,0,0,0,0,20,71,0,2.52,387.00,75,2.80,89.84,-6.50,9.4
2022-02-19 20:16:58.571 NZDT I (13688161) ovms-server-v2: Send MP-0 D28,0,5,17.7778,18.3333,22,0,931373,0,87933,22.5,0,1,1,12.58,0,12.5,0,0,0,20.5556
2022-02-19 20:16:58.571 NZDT I (13688161) ovms-server-v2: Send MP-0 L0,0,0,0,0,0,0,0,0,-2.516,0.084,0.005,0,100,,0,0,0
2022-02-19 20:16:58.571 NZDT I (13688161) ovms-server-v2: Send MP-0 Y4,FL,FR,RL,RR,0,-1,0,-1,0,-1,0,-1
2022-02-19 20:16:58.571 NZDT I (13688161) ovms-server-v2: Send MP-0 W0,0,0,0,0,0,0,0,-1
2022-02-19 20:16:58.571 NZDT I (13688161) ovms-server-v2: Send MP-0 F3.3.001-282-gbd4919ca/ota_0/edge (build idf v3.3.4-848-g1ff5e24b1 Feb 12 2022 00:04:22),,15,0,NL,Rancho Relaxo,-1,-1,OVMS WIFI BLE BT cores=2 rev=ESP32/3; MODEM SIM7600
[... cut ...]
xnl.v.c.chargebars                       0
xnl.v.c.chargeminutes3kW                 0
xnl.v.c.count.l0l1l2                     0
xnl.v.c.count.qc                         0
xnl.v.c.duration
xnl.v.c.event.notification               reset
xnl.v.c.event.reason
xnl.v.c.limit.reason
xnl.v.c.quick                            0
xnl.v.c.state.previous                   stopped
```
I have the "Allowed SOC drop" option in the web UI checked and with the slider set to 30%, but I don't see that in the outputs of "metrics list" at all.

mjkapkan, 2022-02-20:
@Lenbok , what are your settings for range limit and allowed range drop?

Lenbok, 2022-02-20:
Both of those are unset:
[image]

Lenbok, 2022-02-20:
And even through I have Allowed SOC drop set to 30, a few minuters ago, the leaf just started charging even though it's already at 80%.

I don't have the metrics from before, but here's what they are now:
```
OVMS# metrics list
m.egpio.input                            0,1,2,3,4,5,6,7,8,9
m.egpio.monitor
m.egpio.output                           3,4,5,6,7,8,9
m.freeram                                3479860
m.hardware                               OVMS WIFI BLE BT cores=2 rev=ESP32/3; MODEM SIM7600
m.monotonic                              36535Sec
m.net.mdm.iccid                          8944501207213727924
m.net.mdm.mode                           LTE,Online
m.net.mdm.model                          LE20B03SIM7600M21-A
m.net.mdm.netreg                         RegisteredRoaming
m.net.mdm.network                        Spark NZ Hologram
m.net.mdm.sq                             -77dBm
m.net.provider                           Rancho Relaxo
m.net.sq                                 -58dBm
m.net.type                               wifi
m.net.wifi.network                       Rancho Relaxo
m.net.wifi.sq                            -58.3dBm
m.serial
m.tasks                                  18
m.time.utc                               1645346816Sec
m.version                                3.3.001-282-gbd4919ca/ota_0/edge (build idf v3.3.4-848-g1ff5e24b1 Feb 12 2022 00:04:22)
s.v2.connected                           yes
s.v2.peers                               1
s.v3.connected
s.v3.peers
v.b.12v.current
v.b.12v.voltage                          12.04V
v.b.12v.voltage.alert
v.b.12v.voltage.ref                      12.15V
v.b.c.temp                               22.032,22.236,22.032°C
v.b.c.temp.alert                         0,0,0
v.b.c.temp.dev.max                       -0.07,0.14,0.07°C
v.b.c.temp.max                           22.032,22.236,22.134°C
v.b.c.temp.min                           22.032,22.032,22.032°C
v.b.c.voltage                            4.03,4.026,4.026,4.035,4.026,4.026,4.03,4.031,4.026,4.026,4.026,4.031,4.03,4.026,4.026,4.031,4.023,4.026,4.03,4.032,4.026,4.03,4.026,4.031,4.03,4.026,4.026,4.031,4.026,4.026,4.03,4.031,4.026,4.026,4.026,4.031,4.026,4.03,4.026,4.026,4.028,4.023,4.026,4.032,4.026,4.03,4.026,4.028,4.03,4.026,4.021,4.026,4.026,4.026,4.025,4.026,4.026,4.026,4.026,4.028,4.03,4.022,4.026,4.031,4.026,4.026,4.03,4.026,4.026,4.03,4.026,4.028,4.026,4.022,4.022,4.028,4.026,4.022,4.026,4.028,4.03,4.03,4.026,4.031,4.026,4.022,4.021,4.026,4.022,4.026,4.021,4.022,4.026,4.026,4.021,4.03V
v.b.c.voltage.alert
v.b.c.voltage.dev.max
v.b.c.voltage.max                        4.03,4.026,4.026,4.035,4.026,4.026,4.03,4.031,4.026,4.026,4.026,4.031,4.03,4.026,4.026,4.031,4.023,4.026,4.03,4.032,4.026,4.03,4.026,4.031,4.03,4.026,4.026,4.031,4.026,4.026,4.03,4.031,4.026,4.026,4.026,4.031,4.026,4.03,4.026,4.026,4.028,4.023,4.026,4.032,4.026,4.03,4.026,4.028,4.03,4.026,4.021,4.026,4.026,4.026,4.025,4.026,4.026,4.026,4.026,4.028,4.03,4.022,4.026,4.031,4.026,4.026,4.03,4.026,4.026,4.03,4.026,4.028,4.026,4.022,4.022,4.028,4.026,4.022,4.026,4.028,4.03,4.03,4.026,4.031,4.026,4.022,4.021,4.026,4.022,4.026,4.021,4.022,4.026,4.026,4.021,4.03V
v.b.c.voltage.min                        4.025,4.017,4.02,4.025,4.016,4.02,4.02,4.021,4.021,4.021,4.016,4.025,4.02,4.016,4.02,4.025,4.018,4.021,4.024,4.022,4.021,4.02,4.02,4.025,4.02,4.016,4.02,4.025,4.021,4.016,4.024,4.025,4.016,4.02,4.016,4.021,4.02,4.02,4.016,4.021,4.019,4.018,4.021,4.023,4.021,4.02,4.02,4.019,4.02,4.02,4.016,4.021,4.02,4.02,4.02,4.021,4.016,4.02,4.02,4.022,4.02,4.016,4.016,4.021,4.02,4.02,4.025,4.021,4.021,4.02,4.02,4.022,4.021,4.016,4.016,4.022,4.016,4.016,4.016,4.019,4.02,4.024,4.02,4.021,4.02,4.016,4.016,4.021,4.016,4.016,4.011,4.012,4.02,4.02,4.016,4.025V
v.b.cac                                  49.3573Ah
v.b.consumption                          0Wh/km
v.b.coulomb.recd
v.b.coulomb.recd.total
v.b.coulomb.used
v.b.coulomb.used.total
v.b.current                              -0.5A
v.b.energy.recd                          0.00476582kWh
v.b.energy.recd.total
v.b.energy.used                          0.0840625kWh
v.b.energy.used.total
v.b.health
v.b.p.level.avg
v.b.p.level.max
v.b.p.level.min
v.b.p.level.stddev
v.b.p.temp.avg                           22.1°C
v.b.p.temp.max                           22.236°C
v.b.p.temp.min                           22.032°C
v.b.p.temp.stddev                        0.1°C
v.b.p.temp.stddev.max                    0.1°C
v.b.p.voltage.avg                        4.02691V
v.b.p.voltage.grad                       -0.00361V
v.b.p.voltage.max                        4.035V
v.b.p.voltage.min                        4.021V
v.b.p.voltage.stddev                     0.00285V
[... 135 lines cut ...]
xnl.v.b.heatrequested                    no
xnl.v.b.hx                               60.7
xnl.v.b.max.gids                         0
xnl.v.b.output.limit                     110kW
xnl.v.b.range.instrument                 0km
xnl.v.b.regen.limit                      7.25kW
xnl.v.b.soc.instrument                   80%
xnl.v.b.soc.newcar                       61.21%
xnl.v.b.soc.nominal                      79.8%
xnl.v.b.soh.instrument                   0%
xnl.v.b.soh.newcar                       74.7838%
xnl.v.b.type                             1
xnl.v.c.chargebars                       0
xnl.v.c.chargeminutes3kW                 0
xnl.v.c.count.l0l1l2                     0
xnl.v.c.count.qc                         0
xnl.v.c.duration                         60,60,120,5,5,5Min
xnl.v.c.event.notification               reset
xnl.v.c.event.reason
xnl.v.c.limit.reason                     normal
xnl.v.c.quick                            0
xnl.v.c.state.previous                   timerwait
xnl.v.e.hvac.auto                        no

```

Lenbok, 2022-02-20:
Almost suspicious timing in terms of the faux start/stops 24 hours after the last charge finished, it wakes up and starts/stops almost immediately, and then a duplicate "target level reached" / "charge stopped" pair of messages sent an hour later:
[image]

Log from around the time:
[somelog.txt](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/8103677/somelog.txt)

Hope this is helpful in working out why it seems to have a mind of its own!

mjkapkan, 2022-02-20:
@Lenbok  Could it be that you have set the charging timer in the car to start at specific time each day? The logic to start the charging in OVMS is not based on any timers.

Lenbok, 2022-02-20:
I have a timed charge set for once a week on a Friday, so it shouldn't be that.  (Having a timed charge set was the only way I could think of that I can connect the charger to the leaf without it starting to charge immediately, since I want to delay until the solar is ready)

mjkapkan, 2022-02-20:
What model year your car is? Do you have the scheduled preheat/precool enabled?

I have tested this scenario, but can't reproduce it on my 2013 car.

Lenbok, 2022-02-20:
My leaf model year is 2014. I do not have scheduled preheat/precool enabled. The factory TCU is unplugged.

Lenbok, 2022-02-20:
I do get fairly regular crash aborts, e.g.:
```
Last boot was 2673 second(s) ago
Time at boot: 2022-02-21 07:54:12 NZDT
  This is reset #315 since last power cycle
  Detected boot reason: Crash (12/12)
  Reset reason: Task watchdog (6)
  Crash counters: 198 total, 0 early

Last crash: abort() was called on core 0
  Backtrace:
  0x4008ddca 0x4008e065 0x400e9608 0x40084176
  WDT tasks: IDLE0
  Version: 3.3.001-282-gbd4919ca/ota_0/edge (build idf v3.3.4-848-g1ff5e24b1 Feb 12 2022 00:04:22)
```

About 70 crashes on Feb 18th (when I took the leaf out), 6 on Feb 19th (it was home all day), 3 on Feb 20th. I wonder if some state is being reset across reboots?