openvehicles/Open-Vehicle-Monitoring-System-3 #697: NL: Significant bug with SoC limit feature

glynhudson, 2022-02-08:
I've noticed a significant bug with the SoC limit feature. 

Without OVMS connected, my 2014 Leaf charges to reported 98%, this is totally normal due to battery degradation. The car stays at 98% for about a an hour so to cell balance before the car signals the charging station to switch off. 

I've noticed when OVMS is connected running the latest version even when SoC limit feature is switched off (SoC limit = 0) OVMS will force the charging session to restart which results in the charging station switching back on, this **cycle repeats infinity**. Presumably OVMS is trying to get SoC=100% which is impossible on my car. The only workaround I've found is to either unplug OVMS or set the required charge to 98%, but this has the side effect of stopping the charge immediately at 98% and not allowing the battery pack to cell balance. 

In my opinion, this is a significant bug which if left unattended for a long period could damage a car. I propose OVMS to assume anything above 95% = 100%. Above 95% SoC OVMS should not try to stop or start a charging session. 

[image]

mjkapkan, 2022-02-08:
Hi @glynhudson,

I have already added handling exactly for this type of issue, as my leaf also does not charge to 100. You need to set allowed SOC drop value to 2% or more.
[image]

However, this should really not be happening if you set the SOC limit to 0. I am using this feature for over a month and this never happened to me. Are you sure you have set the range limit to 0 as well?

mjkapkan, 2022-02-08:
The only way, I think, this could be triggered if you set the range limit to more than your max range and set the SOC limit to 0. Then the logic is to stop the charging at 100%. And as your car can not reach this, it will start the charging again.

So the easy fix here is to use both range limit and SOC limit. And set the allowed drop to 2% or more. 

As for the permanent fix, please provide the xnl metrics  so I can try to reproduce this.

glynhudson, 2022-02-11:
You're right, I had accidentally set the range to be more than my max range. Setting everything to '0' resulted in the car stopping charging normally and not restarting when full charge was reached. Next I'll try to set the Allowed SoC drop. 

It looks like this was user error rather than a bug! Nice work, thanks for your reply

[image]

closed 2022-02-11 by glynhudson

mjkapkan, 2022-02-11:
Good to hear. Even if it's user error, it shows that the documentation is not clear enough. I will try to update it to be more clear. And also will update so that it still respects soc drop value when soc limit is 0 but range limit is in use.