openvehicles/Open-Vehicle-Monitoring-System-3 #698: NL: Update documentation for Remote HVAC support on MY 2016-2017

mjkapkan, 2022-02-09:
2016-2017 LEAF models now support remote HVAC activation. Documentation needs to be updated accordingly. 

(It seems like lates changes to wakeup commands accidently made this work 😄 )

glynhudson, 2022-02-09:
Awesome! Do HVAC request work when the vehicle is locked and asleep? I.e not charging 

Could you confirm if door lock/unlock works on this model year?

mjkapkan, 2022-02-09:
Have not tried. @dalathegreat should be able to confirm.

yusijs, 2022-06-07:
Can confirm this on my 2016 leaf as well. Works w/ TCU still connected, but like @biddster I also have to tap the a/c button 3 times.

kuro68k, 2024-08-18:
Hi from 2024. How is this working today? Still reliable with 3 taps? Thanks.

yusijs, 2024-08-20:
I wouldnt say reliably 😅 sometimes it works on the first go, other times I need to tap i 2-3, and when the car is in a shit mood it may take like 7 taps. I created an automation in homeassistant that runs it 5 times with a 10 second delay between if it fails to turn on, but that's just me being lazy.