openvehicles/Open-Vehicle-Monitoring-System-3 #704: Bolt EV - swcan needed

kallisti5, 2022-03-04:
I'm not sure of the details to set this up.. but the Bolt EV is need of the swcan?   The ODB2 port goes to "sleep" after a while and doesn't respond to commands.

kallisti5, 2022-03-04:
https://github.com/mjuhanne/OVMS-SWCAN seems to be what folks are talking about.  Information is limited, but it sounds like swcan needs to be wired to the private GMLAN bus?

I plan on trying to fully remove my Onstar module at some point (and wiring the microphone to restore bluetooth microphone support)  Guessing the OnStar module would have unrestricted GMLAN access?

kallisti5, 2022-03-04:
I found some documentation of the OnStar connectors online from the Bolt EV...   Looks like GMLAN low and high speed is readily available at the Onstar box...

[image]
[image]

kssmll, 2022-03-05:
SWCAN(Low speed GMLAN) is on pin 1 of the OBD2 connector.

kallisti5, 2022-03-05:
> SWCAN(Low speed GMLAN) is on pin 1 of the OBD2 connector.

I think in the Bolt EV you can't access the whole set of GMLAN commands however. There is a gateway between the OBD port and the ECU to prevent access to door lock control , etc.

kssmll, 2022-03-05:
First of all, you need swcan for car's wake up. And GM cars does not use gateways. What  set of GMLAN commands you talking about?

kallisti5, 2022-03-06:
> First of all, you need swcan for car's wake up. And GM cars do not use gateways. What set of GMLAN commands you talking about?

https://www.chevybolt.org/threads/can-bus-commands-to-remote-start-the-bolt.33637/#post-544071

Msbog, 2022-03-23:
Hello, I want to confirm that sending commands via SWcan located behind the mirror of the camera connector
( black wire : GND and green wire : SWCAN)  works and can start air conditioning, heating, unlocking, locking, etc. Tried!

kallisti5, 2022-03-24:
> Hello, I want to confirm that sending commands via SWcan located behind the mirror of the camera connector ( black wire : GND and green wire : SWCAN) works and can start air conditioning, heating, unlocking, locking, etc. Tried!

Yup.  The mirror and the OnStar module are two places you can get a hold of GMLAN not behind the OBD firewall on the Bolt EV.

Now... we just need to get a reliable source for the SWCAN module for the OVMS

markwj, 2022-03-25:
I am willing to build SWCAN modules in the same way we do the KLINE modules today.

Just need a circuit design. Presumably the single SWCAN wire can be brought out onto the DB9 in the same way.

Regards, Mark

Msbog, 2022-03-25:
I made my own module with esp32 and sim800l, which communicates via mqtt with my Home assistant server. The space under the camera cover is large and everything fits normally. I'm wondering if there is information on the swcan bus about SOC, temperature, remaining distance, etc. . My idea is if only the SWcan bus can be used. I also see that there is information from the GPS data, but I don't know how to decode it?

kallisti5, 2022-05-17:
@markwj I'd love a SWCAN module for OVMS.  The only reference I see is this one:
https://github.com/mjuhanne/OVMS-SWCAN

However, it is oddly a 4 layer board, and only includes gerbers :-(

kallisti5, 2022-05-17:
@Msbog have you published any of the source or hardware for your creation anywhere?  I'd imagine a lot of people would be interested.

markwj, 2022-05-18:
> @markwj I'd love a SWCAN module for OVMS. The only reference I see is this one: https://github.com/mjuhanne/OVMS-SWCAN
> 
> However, it is oddly a 4 layer board, and only includes gerbers :-(

There is a PDF schematic there. Doesn't sound overly complicated to simplify this and do it the same way as the K-Line board. But are we sure there is sufficient quantity for this to make it worthwhile? Has anybody looked at SWCAN on the Bolt-EV?

kallisti5, 2022-05-18:
I wanted to look at SWCAN on the Bolt EV which is why I wanted one :-)

markwj, 2022-05-25:
Regarding the SWCAN board, given component vs board manufacturing cost, it seems simplest to combine this with the existing k-line option board, to produce a new ‘combo’ option board containing both K-line and SWCAN (DIP switch selectable one or the other, not both).

We are currently working on the design of this board, and discussions are ongoing on the developers mailing list.

melaz, 2022-10-19:
> Regarding the SWCAN board, given component vs board manufacturing cost, it seems simplest to combine this with the existing k-line option board, to produce a new ‘combo’ option board containing both [...]

Is there any updates on this board? I really like my OVMS, but on volt it is not much fun :)

kallisti5, 2022-10-19:
Yeah, i'd love a combined swcan + k-line board.  Hell, just open source the design and i'll manufacture it myself.  I ended up pulling OVMS out of my car because it was lacking a lot of functionality without the add-on board

markwj, 2022-10-20:
Status update: Some prototype boards are being produced now so we can start to work on firmware support and confirm the design. The board provides K-line, SWCAN, and high speed CAN, software selectable.

cavram, 2022-10-20:
> ID: <openvehicles/Open-Vehicle-Monitoring-System-3/issues/704/1285109051@
> github.com>

Do you want a Holden Volt tester?

kallisti5, 2022-11-26:
@markwj i'd be available to test (and would even toss some money in for it)

As of my latest changes, here's the BoltEV during Level 2 charging:
```
I (3550249) ovms-server-v3: Transmit all metrics
I (3550249) ovms-server-v3: Tx metric ovms/bolt/metric/m/egpio/input=0,1,2,3,4,5,6,7,8,9
I (3550249) ovms-server-v3: Tx metric ovms/bolt/metric/m/egpio/output=3,4,5,6,7,8,9
I (3550249) ovms-server-v3: Tx metric ovms/bolt/metric/m/freeram=3480512
I (3550249) ovms-server-v3: Tx metric ovms/bolt/metric/m/hardware=OVMS WIFI BLE BT cores=2 rev=ESP32/3; MODEM SIM7600
I (3550249) ovms-server-v3: Tx metric ovms/bolt/metric/m/monotonic=3547
I (3550249) ovms-server-v3: Tx metric ovms/bolt/metric/m/net/mdm/iccid=XXXXXXREMOVED
I (3550249) ovms-server-v3: Tx metric ovms/bolt/metric/m/net/mdm/mode=LTE,Online
I (3550249) ovms-server-v3: Tx metric ovms/bolt/metric/m/net/mdm/model=LE20B03SIM7600M21-A
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/m/net/mdm/netreg=NotRegistered
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/m/net/mdm/network=AT&T Hologram
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/m/net/mdm/sq=0
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/m/net/provider=XXX
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/m/net/sq=-77
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/m/net/type=wifi
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/m/net/wifi/network=xxx
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/m/net/wifi/sq=-76.7
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/m/tasks=17
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/m/time/utc=1669497042
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/m/version=3.3.003-156-g2b63300f/ota_1/edge (build idf v3.3.4-849-g6e214dc335 Nov 26 2022 10:54:44)
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/s/v2/connected=yes
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/s/v2/peers=0
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/s/v3/connected=yes
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/s/v3/peers=0
I (3550259) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/12v/voltage=13.8
I (3550269) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/12v/voltage/ref=12.6
I (3550269) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/c/temp=15,15,16,18,16,15
I (3550269) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/c/temp/alert=0,0,0,0,0,0
I (3550269) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/c/temp/dev/max=-0.83,-0.83,0.17,2.17,0.17,-0.83
I (3550269) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/c/temp/max=15,15,16,18,16,15
I (3550279) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/c/temp/min=15,15,16,18,16,15
I (3550279) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/consumption=0
I (3550279) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/p/temp/avg=15.83
I (3550279) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/p/temp/max=18
I (3550279) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/p/temp/min=15
I (3550279) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/p/temp/stddev=1.07
I (3550279) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/p/temp/stddev/max=1.07
I (3550279) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/p/voltage/stddev/max=0
I (3550279) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/range/ideal=345
I (3550289) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/soc=83
I (3550289) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/temp=16
I (3550289) ovms-server-v3: Tx metric ovms/bolt/metric/v/b/voltage=382.2
I (3550289) ovms-server-v3: Tx metric ovms/bolt/metric/v/c/charging=yes
I (3550289) ovms-server-v3: Tx metric ovms/bolt/metric/v/c/climit=32
I (3550289) ovms-server-v3: Tx metric ovms/bolt/metric/v/c/current=30
I (3550289) ovms-server-v3: Tx metric ovms/bolt/metric/v/c/kwh=10
I (3550289) ovms-server-v3: Tx metric ovms/bolt/metric/v/c/mode=standard
I (3550289) ovms-server-v3: Tx metric ovms/bolt/metric/v/c/pilot=yes
I (3550289) ovms-server-v3: Tx metric ovms/bolt/metric/v/c/state=charging
I (3550289) ovms-server-v3: Tx metric ovms/bolt/metric/v/c/substate=onrequest
W (3550489) websocket: WebSocketHandler[0x3f82b89c]: job queue overflow detected
W (3550489) websocket: WebSocketHandler[0x3f82b89c]: job queue overflow resolved, 28 drops
```

Overall that seems "sane" for level 2.    I know DCFC isn't detected correctly for some reason, so that's a big improvement that I need to work on more.

melaz, 2023-04-12:
Hi, is there any updates on this? Almost a year passed.

markwj, 2023-04-13:
> Hi, is there any updates on this? Almost a year passed.

We identified two SPI CAN driver ICs that could support our new expansion board's requirements, but both have been subject to extreme chip shortages (as have many higher-end automotive chips). Distributors were quoting lead times of 2024 or later, and what few chips were on the market were 100 times the normal price and likely fake/faulty anyway.

Thankfully, that seems to be resolving now. The MCP2517fd is now available in small quantities, and that is the older obsolete version of the MCP2518fd that we had identified as one of the options. But both are pin compatible, and probably close to software driver compatible as well. The other (better, imho) alternative is TCAN4551 or TCAN4550, but those are still not reliably available (expecting some versions in summer 2023, but normal availability unlikely until summer 2024).

So, we are building prototypes around MCP2517fd, and we will see if that can meet our needs. Most distributors are showing wider availability now, and production quantity late summer/autumn 2023.

kallisti5, 2023-04-16:
@markwj if you share the PCB, I can try building one myself.   I'm afraid i'm going to be getting rid of the Bolt EV at some point in the next few months... so my time to debug / test is short.

markwj, 2023-04-17:
> @markwj if you share the PCB, I can try building one myself. I'm afraid i'm going to be getting rid of the Bolt EV at some point in the next few months... so my time to debug / test is short.

Prototype design schematic attached. The idea is to unify all our expansion options (K-Line, SWCAN, CAN4, etc) into one board. As we have only two full GPIOs (plus a few spare expansion ones), we would use 74HTC4052 chips to switch between expansion options (only one being usable at any one time, but software switchable).

[OVMS_Unified_Expansion_Prototype.pdf](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/11245067/OVMS_Unified_Expansion_Prototype.pdf)

Design is not finalised, but used for prototype boards only at the moment.

knumat, 2023-08-22:
@markwj I am a software developer and Bolt EUV owner, so I'd love to help with testing this and possibly a few minor development tasks.  I just signed up for the developer mailing list.

kssmll, 2023-08-23:
> So this board: https://www.pcbway.com/project/shareproject/SWCAN_adapter_for_OVMS_b002c070.html doesn't work?
Works (as of Nov 24, 2022) with this fix https://github.com/kssmll/Open-Vehicle-Monitoring-System-3/commit/e278043f148742d941c157ed0e0f2c4cf56997a2

IBUSOL, 2023-08-23:
Nice!

Noob question: how can I change OVMS this lines?
Can't find a file edit option in OVMS.
Just got my OVMS for a week and also bought the SWCAN adapter.

melaz, 2023-08-23:
Where have you got SWCAN from?

melaz, 2023-08-23:
> Sorry, bought should be: build
> 
> The SWCAN adapter is built by PCBWAY. 
> 
> I'm unsure if this works. 
> 
> The minimum order quantity is 5, so I have 4 boards left.

Where are you located, can you sell me one?

IBUSOL, 2023-08-23:
Netherlands, selling is no problem.
Still not sure if the board is working.
You can mail me: hesanay694@jontra.com (temporary mail address)
Hopefully @kssmll or someone else can tell me how i can change the code.

kssmll, 2023-08-23:
> Netherlands, selling is no problem. Still not sure if the board is working. Hopefully @kssmll or someone else can tell me how i can change the code.

You need to build your own configuration with the swcan module enabled (check swcan in configuration). See help for developers.  Also, the code for your car must be use the swcan module. See Chevorlet Volt for example.
Now swcan compilation is broken, so you need the fix from my fork.

IBUSOL, 2023-08-24:
> You need to build your own configuration with the swcan module enabled (check swcan in configuration). See help for developers. Also, the code for your car must be use the swcan module. See [...]

Can somebody help me with that?
So a firmware file with SWCAN enabled and script changes?
I will test the changes ;-)

IBUSOL, 2023-08-24:
Just got a file form @kssmll  (many thanks)
Look very good!

Now I can:

- Lock and unlock my car!
- See Estimated Range (before only Ideal range)
- 12V power voltage (was empty)
- Tire pressure/temperature (was empty)

Not working:

- Start/stop charging (app reports: Failed; I only have a home charger)
- Change load power (10A/16A) 
-  Start/stop pre-heating (app reports: Failed)

IBUSOL, 2023-10-02:
If someone interested, contact me: i n f o @ b u i t e r o n l i n e . n l
I have 4 left

melaz, 2023-10-23:
> If someone interested, contact me: i n f o @ b u i t e r o n l i n e . n l I have 4 left

Sent an email to you. Btw, remind, does your car has onstar?

noideas4thisname, 2024-01-24:
Any updates on this? I just went through the experience of GMs app locking important functions behind a paywall and I want to replace everything. If there is a functional board for the Bolt EV, I'd be keenly interested in getting OVMS set up and working.

brocktice, 2024-01-25:
I got the board from ibusol, just ordered the rest of the kit for my 2017 bolt ev. Once I have it and start wrapping my head around this I will post an update.

kallisti5, 2024-01-25:
Sorry, I can't be any more help on this one (Bolt EV has been sold).  I definitely don't need my OVMS for my new vehicle (Rivian make great software ;-) )

Good luck!

IBUSOL, 2024-01-26:
Only 1 SWCAN unit available.
For me: no reason to post something, it just works.

rbray89, 2024-04-17:
> 2017 bolt ev

Which kit did you order? And from what vendor?

rbray89, 2024-04-17:
> Just got a file form @kssmll (many thanks) Look very good!

Did you have to modify the OVMS hardware or cabling?

IBUSOL, 2024-04-18:
Last print is also sold.
No SWCAN units left anymore.

horstt, 2024-07-31:
Just bought an Ampera-e, would love to make it more connected! What is the current status, how useful is OVMS for the Ampera-e (Bolt) at the moment?

SirSheik, 2024-10-12:
> Just got a file form @kssmll (many thanks) Look very good!
>
> Now I can:
>
> * Lock and unlock my car!
> * See Estimated Range (before only Ideal range)
> * 12V power voltage (was empty)
> * Tire [...]

> i n f o @ b u i t e r o n l i n e . n l

Hello.
Sorry could you please share the .bin file of the firmware already modified for the board to work?
Thank you very much

Donovan22, 2025-01-02:
I am selling SWCAN boards and OBD2 -> DB9 cables with pin 1 (OBD2) connected to pin 1 (DB9) through ebay here:
https://www.ebay.com/itm/365182274464
This will work with 1st Gen Volt and might work with 2nd Gen Volts without a gateway.
I have a friend with a Bolt EV and we are going to try using OVMS with OBD2 port with this bypass cable that looks like it will fit:
https://www.btdieselworks.com/products/2017-sdgm-bypass-harness

I'll post results here when the bypass cable arrives.

dangerusty, 2025-05-05:
> I am selling SWCAN boards and OBD2 -> DB9 cables with pin 1 (OBD2) connected to pin 1 (DB9) through ebay here:
>
> https://www.ebay.com/itm/365182274464
>
> This will work with 1st Gen Volt and might work [...]

I've just discovered all of this, and am wondering if there are any test results for gen2 using your cable?

Donovan22, 2025-05-05:
The gen 2 Volt without gateway (pre 2019) should work. The Bolt EV is different. there, the SDGM harness is useful if you don't want to splice into the OEM harness for testing, but beyond that, it doesn't do much. I still have yet to do much testing outside the gen 1 Volt.

jamesb2147, 2025-11-17:
I'm garbage at soldering and wanted PCBWay to handle that, but apparently there's no Centroid/pick-and-place file provided. (Of course, only caught after providing payment.)

@IBUSOL @kssmll - Can anybody help? Seems I may give up (my soldering skills are that bad).

rbray89, 2025-12-04:
if anyone gets the updated FW working, I'd be interested in that too.

jamesb2147, 2026-01-30:
I don't see **any** lights when it's running the command, and the console suggests it's not really getting through, either, even though OVMS is responding:

[image]

FWIW I mentioned HA only so folks would have the context. I've already got it running on v2 and haven't moved to v3 (for which there's a HACS integration) because I'm not sure yet how much data it'll use. v2 is like 2.5MB/mo, which is _insanely_ efficient!

jamesb2147, 2026-02-03:
I have played with it a few times now and can confirm that my own boards ordered through PCBWay using the design from @kssmll even though we didn't have a centroid file. I DID have to point out the orientation of some components was incorrect in PCBWay's first attempt, so keep that in mind! When I get a chance, I'll post the high resolution images of the functional layout for anyone that wants to order these boards on their own later. For now, use @WaferThinLogic eBay listing mentioned earlier in the thread, and if they run out (or you're just local to Central IL and feel like meeting a fellow car hacking geek) you can hit me up. I think I've got 20 of them.

Which brings me to my next point. My boards are fine! Yay!!! It turns out that my initial issues were with using the standard OBD-II cable that came with my OVMS kit (I ordered the Kia 4G kit with modem and cable from Medlock & Sons - they apparently forgot to ship it, too, so I had to poke them and then it went out the next day).

@WaferThinLogic - Anything special about your cable? I'm guessing it's the pin layout is different for that one pin mentioned in the documentation for SWCAN support..? Is there anywhere else I can get one? Yours works, but it'd be better for my vehicle if I could get one with a right-angle connector instead of the straight one (at the OBD end). Wife won't be happy with the way it's poking out right now, and that's before I potentially add another OBD adapter for a Comma 4 to pull power...

jamesb2147, 2026-02-11:
New problem, albeit with a poor workaround: Using v3/MQTT, OVMS is sending updates to ovms/$user/$car/event/egpio/output/7/low multiple times per second. That's eating my very expensive data plan from Hologram! (Could not get Google Fi data-only SIM to work with OVMS, but Hologram works fine.)

Just yesterday I blew through 26MB in a day, which at my plan's rate of $0.21/MB is something like $5/day! The server v2 protocol is more like 3MB/month, for comparison.

Anybody have any idea what's causing egpio output 7 to repeatedly report low then not then low again? Or any way to filter it? I see built in ways to filter metric updates (useful specifically for session length... stupid metric to update every second), but nothing on filtering out these low events. Any ideas?

ETA: Sorry, I'm seeing more detail now and it's flipping between output 7 high and output 7 low, not just low/not low.

jamesb2147, 2026-02-14:
I don't think the standard firmware supports the module I'm using.

> Here are the firmwares built from my repo (with swcan enabled as can4 and some Volt/Bolt QoL changes). The latest is "120125" https://drive.google.com/drive/u/1/folders/1bPZOUyHFMoReMyA5mach4ARlp73JWnfH

I ended up using the latest version of this firmware.

Thinking about it, though, I was using the bad cable before... let me try tomorrow or Monday to use the standard firmware (it's still on my secondary flash slot). I'll report back whether it works or not.

jamesb2147, 2026-02-15:
Confirmed that the standard build does not support the module I'm using. Tangentially, it appears PCBWay took the feedback I gave them and built a Centroid file for the public listing, so anyone should be able to request their own fresh module via PCBWay going forward. Yay!

Here's what happens when I try to control cabin climate using the standard build:

[image]

Anyway, since the standard build doesn't include the bits needed for this SWCAN module, I'll refer back to @WaferThinLogic - Any chance you can share that code with the class so we can hopefully make it part of the standard build, or at least a publicly maintained branch?

dexterbg, 2026-02-21:
Sorry, fixed in `3.3.005-704-g6a1fed98`.

jamesb2147, 2026-02-21:
I'm going to reveal my ignorance here. Please forgive me for asking questions that may be obvious to others!

Is there a build sitting somewhere I can pull down? I just ran a fetch manually via the local GUI on the OVMS and its build date is Sep 7 2025. I found where the ReadTheDocs builds are, and the pipelines showing how I'd do that locally if I wanted to. But I don't see any recent GitHub Actions on this repo, and the latest tag/build in GitHub appears to be the initial 3.3.005 release from July 18 2025.

Am I missing something or do I need to build it myself locally from source? Any answer is fine, I'm just looking to try it out both for testing and to hopefully switch back to MQTT since it seems to work significantly more consistently for me. That, and it's not obvious to me what I need to do to built it locally... ;_;

dexterbg, 2026-02-22:
To install this build, open Config→Firmware, set your version tag to "edge". Then either wait for the next auto update run if enabled, or switch to the "Flash from web" tab and do a manual update with empty URL.

jamesb2147, 2026-02-23:
IDK what you're asking about with the HomeLink 1/2, but I can confirm the Bolt climatecontrol on/off commands are working. I used it to pre-heat the car this morning over MQTT with the Home Assistant HACS add-on for OVMS!

Now, comparing data usage, you should know this graph shows a period of time where I'm driving from home (OVMS uses home WiFi) to work (cellular). So it's generally seeing 8-14 hours a day on a cellular connection, and it took me some time to figure what was going on. 

The first part of the chart is using MQTT, the middle part is using the very efficient REST endpoints, and today's data is roughly half a day using MQTT.

[image]

It's obviously not as efficient as the REST endpoint, but I'm personally willing to make that tradeoff. It might be like 1MB/day (2 if somebody could not connect to WiFi overnight), which is like $10-20/mo. Not a great rate, but sooooo much better than it was! Thanks for the work on this, guys!!

IBUSOL, 2026-02-23:
> > Nice work that swcan is now supported by default! Is it possible that the A/C control button on the Car page also be available? (climatecontrol ON/OFF)
> 
> Is climate control On/Off working on the Chevrolet Bolt via HomeLink 1 and 2 now? I will do that for the Volt and can also add it to the Bolt if climate control is working for now.

I never tested this but yes.
Homelink 1: Climatecontrol is turn on
Homelink 2: Climatecontrol is turn off

The APP always give a error: "Homelink command issued"
But after turning it on and off it does not go on again (??)

Climatecontrol on/off is working better.
You can turn it on/off multiple times.