openvehicles/Open-Vehicle-Monitoring-System-3 #705: Bolt EV - DCFC is not detected

kallisti5, 2022-03-04:
I took a trip to a DCFC station, and the OVMS doesn't realize when the vehicle is currently DCFC.   (It shows not charging still via the web ui)

Need to investigate which PID -> OVMS linkages are missing to properly detect CCS DCFC.

daradib, 2022-07-17:
[AllEV.info](https://allev.info/boltpids/) lists PID 0x4531 as the Charging Level on their [Google sheet](http://bit.ly/ChevyBoltPIDs)

    0=None 1=AC L1 2=AC L2 3=DCFast

`OvmsVehicleBoltEV::Ticker1` in `vehicle_boltev.cpp` assumes charging state based on 0x4369 and 0x4368 which are onboard AC charger current and voltage, respectively, and presumably 0 when DCFC.

kallisti5, 2022-07-25:
Ah ha. ok.   I was assuming the logic on line 480 would account for DCFC, however i'm not realizing that requires SWCAN (which I don't have a board for)

markwj, 2022-08-30:
See #704 for swcan.

commit 6a87dcb0 (2022-11-13): Merge pull request #722 from kallisti5/boltev-dcfc

Bolt EV: Better status reporting of DCFC sessions; fixes #705
commit 7eb1d103 (2022-11-13): Bolt EV: Better status reporting of DCFC sessions; fixes #705
closed 2022-11-13 by dexterbg