openvehicles/Open-Vehicle-Monitoring-System-3 #719: Peugeot e-208 interest? (Similar: Opel Corsa-e)

hutje, 2022-07-07:
Are there already devs working on the Peugeot e-208 or Opel Corsa-e?
I am willing to take a look into developing for this model. Mainly because the app service of Stellantis is crap. Not being able to even connect the car to your account for example.

I have experience with programming ESP32 but not with CAN-bus and OVMS yet. Willing to take a look, but if there are other devs already researching, I could help.

dexterbg, 2022-07-07:
Welcome to the team :)

No developers working on these models (to my knowledge).
Check out the OVMS developer guide & use the Ioniq vFL module as a template to begin.
Check out the developer mailing list archive for some general CAN/OBD intro.

See my intro post:
https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/255#issuecomment-835379478

Regards,
Michael

hutje, 2022-07-15:
Thanks!
Will check it out. Package is released by customs so expect to receive it soon.
Just found a website with the current PIDs for Peugeot e208. (https://www.goingelectric.de/forum/viewtopic.php?f=325&t=64034&start=70)
```
#######
####### Setup commands
#######
ATWS ---> Warm reset set back to defaults
ATI ---> Show interface ID
ATE0 ---> Turn off terminal echo
AT@1 --->
ATSP6 ---> Set to protocol 6: ISO 15765-4 CAN (11 bit ID 500 kbaud)
ATAT1 --->
ATL0 ---> Disable line feed
ATS0 ---> Disable sending spaces between hex characters
ATH1 ---> Turn on odb2 header
ATCAF1 ---> Turn on CAN auto formatting
ATSH79B ---> Set request message header 79B
ATFCSH79B ---> Set flow control header (CAN ID)
ATFCSD300000 ---> Set flow control data
ATFCSM1 ---> Set flow control mode; 1 = User defined CAN ID and data

#######
####### Data commands
#######
ATSH6A2 ---> Set the header of transmitted OBD messages to header 6A2
ATFCSH6A2 ---> Set flow control header (CAN ID)
ATCRA682 ---> Set the CAN hardware filter to 682
22D4341 ---> Ambient temperature
22D8EF1 ---> Battery temperature
22D4021 ---> Speed
ATSH6B4 ---> Set the header of transmitted OBD messages to header 6B4
ATFCSH6B4 ---> Set flow control header (CAN ID)
ATCRA694 ---> Set the CAN hardware filter to 694
22D8161 ---> Battery current
22D8151 ---> Battery voltage
22D86F1 ---> Battery minimum voltage
22D8701 ---> Battery maximum voltage
22D4101 ---> SOC calibrated
22D8651 ---> kWh available
22D8601 ---> SOH
22D8101 ---> SOC

hutje, 2022-07-26:
**[Car Scanner](https://play.google.com/store/apps/details?id=com.ovz.carscanner) is compatible with PSA EVs and have listed the same PIDs from above.**
If I understand correctly, to be able to receive the right messages, this string needs to be send in the beginning to configure the connection.
<table>
<tr><th>**Car Scanner - Configuration for PSA**</th> <th>**Car Scanner - PID for PSA**</th></tr>
<tr><th>[image]</th>
<th>[image]</th></tr>
</table>

There is no need to decrypting RAW data because this is already done by others.
I think it would be fairly easy for an experienced OVMS-programmer to "convert" this into code for OVMS3.
I really hope some experienced OVMS-programmer is willing to make this possible.

deitrix, 2023-10-15:
I'm interested in an integration for my Corsa-e. My understanding is that PSA are using a non-standard protocol for the OBD2 port, which ABRP and Car Scanner have manually integrated. What's holding back an OVMS integration, now that we know the PIDs? I'd be happy to assist in any way I can.

dexterbg, 2023-10-15:
The above info is not a complete PID documentation. It specifically lacks info about the response payload structure & conversion (offsets, factors, bit lengths, signed/unsigned), and it doesn't give any info on what to expect if the car is off / how to detect the car being switched on.

As you own a Corsa-e, you can try to fill these gaps using the OVMS obd request command.

For example, ELM327…
```
ATSH6A2 ---> Set the header of transmitted OBD messages to header 6A2
ATFCSH6A2 ---> Set flow control header (CAN ID)
ATCRA682 ---> Set the CAN hardware filter to 682
22D4341 ---> Ambient temperature
```
…should translate to OVMS…
```
obd can1 req dev 6a2 682 22d434
```

The command will show you the response payload as a hex dump. You then need to analyse the data to see how the raw data needs to be translated to get the ambient temperature.

Once you've gathered all info on the PID responses, test which PID request can be used to detect the car being switched on, i.e. which can be sent without causing the car to be kept awake.

Based on all of this, you then either implement a car adapter yourself based on the examples available, or donate some funding to an OVMS developer to get an initial implementation.

Regards,
Michael

th84sl, 2024-12-17:
Hello, here is my description how to connect Peugeot 208e  via WICAN OBD2-Adapter with Smarthome (IO Broker) and transmit the SOC to the Wallbox  OpenWB.
[get SOC with WICAN from Peugeot 208e v2.pdf](https://github.com/user-attachments/files/18172358/get.SOC.with.WICAN.from.Peugeot.208e.v2.pdf)

th84sl, 2024-12-18:
Hello,
I was able to get my Peugeot 208e connected to my smarthome via WICAN OBD2 adapter.
Please see the attached description.
Regards
Heiko
[get SOC with WICAN from Peugeot 208e v2.pdf](https://github.com/user-attachments/files/18177157/get.SOC.with.WICAN.from.Peugeot.208e.v2.pdf)

ramack, 2025-01-06:
I am working on the FIAT eDoblo, which is a relabled PSA Rifter, so I guess there are many services and CAN frames in common... See #976 for more details.

RoaldNakstad, 2025-04-11:
I own a peugeot e2008 from 2020. The myPeugeot app is crap, and since it only is equiped with support for 2G and 3G it will stop working at all since 3G already is history and 2G is removed during 2025.
So I gladly donate some money if i know there is ongoing work for support of this PSA-cars in OVMS. I do have a ODBII bluethooth dongle for https://powercruisecontrol.com/ and with that I can use Car Scanner app and read out a lot of information. Example:

[image]
[image]
[image]
[image]
[image]

I also already own a OVMS using it in my second car a Leaf from 2013, so maybee it is possible to do a data dump with that and the ODBII bt dongle while using the car scanner app when hooking both devices to the OBDII conector with a splitter like this? https://www.aliexpress.com/item/1005006093167446.html

I'm not a skilled developer but can support someone with the skills an maybee grab some live data to share with you.

ramack, 2025-04-11:
Mh, I have my doblo in operation and I guess it should work for the e2008, also. Would really be time to publish it 😬 - but VIN read and vehicle state (charging, parked, etc) is not working at all...

ramack, 2025-04-11:
just pushed the state without double checking to https://github.com/ramack/Open-Vehicle-Monitoring-System-3/tree/eDoblo so in case you wanna try it, feel free. And maybe it would push me out of this "there is so much cleanup and feature-need left, so I cannot show it" mode and into getting things done ;-(

RoaldNakstad, 2025-04-12:
Thanks :-) Just purshased a 4G modem to upgrade an OVMS I have used in a Smart EV earlier, so I don't need to remove the one daily used in my Leaf. So i'l bee back when that is in plase.

jtq4u, 2025-06-23:
Another +1 for e-208 and e-2008

I've never contributed to omvs before, but I work for IBM on a mainframe development team, so I am quite used to wading in bits and bytes and assembler.  Happy to learn if guided.  I have access to a e-2008

My motivation is the fact you can't ask it to stop charging at 80%
I am also wanting the PID for the brake servo vacuum pressure.

ramack, 2026-04-05:
did you forget to push your branch?

you mean the ` if(d[1] != 0x06)`, right? - This is not there to irgnore other frames, but to keep the soc value clean. - Sometimes there is a 3a8 frame transmitted in my edoblo, which has byte1 == 0x06 and in those cases the soc is simply wrong (0). It doesn't happen often, but if, the consequences are ugly: wrong soc shown. Maybe you don't need it, maybe this status signal is not even there, or carries other values in these cases.

on my doblo I have seen:
```
1775419101.593206 1R11 5C3 d5 9f e7 31 00 07 d0 00
1775419224.551416 1R11 3A8 d6 10 fe 51 a3 c0
```

we it looks like we are not compatible here, as 5c3 doesn't carry the 51 (with the 81% soc)...

Synovialbasher, 2026-11-14:
For anyone that's interested, I've been working hard on an integration for the Opel Corsa-e under the Fiat e-doblo profile. I've added the following features to the somewhat-workign code for the fiat e-doblo. If anyone would like to try it on an eCMP vehicle, my fork is here: https://github.com/Synovialbasher/Open-Vehicle-Monitoring-System-3/tree/corsa-e-ecmp

Charging

- AC and DC detection from frame 0x462, with the charge type (type 2 or CCS) and the pilot signal.
- Charge power, current and voltage. This fixed the 0 V shown while charging under the e-doblo profile.
- Time to full from 0x522, which matches the dash.
- Data while charging with the car off: the battery ECU doesn’t answer then, so current, power and SOC come from the frames the car broadcasts.
- Charge state ends with “done”, with the charge values reset to zero.

Battery

- All 108 cell voltages and 54 temperature probes, shown on the Battery Monitor page and in bms volt / bms temp.
- Battery temperature is shown using the average of all probes.
- Current sign fixed: positive when driving, negative when charging or regenerating.
- Trip energy used and recovered, from the car’s own energy counter.

Driving

- Trip distance in 0.1 km steps, from a distance counter found in 0x38D. This fills “Last trip” and “Consumption” in the app.
- Range:
  - an estimated range from the learned average consumption
  - an ideal range from the WLTP figure
  - the range at 100 %
- Gear, B mode, handbrake, and motor and inverter temperatures.

Body

- Lock status from 0x512. The original read a bit that turned out to be the low beam on the Corsa-e.

I also implemented a server fix where the charging power was showing the wrong value. For example, it would show the car charging at 13 kw on AC when the car is only capable of 7. I suppose this would apply to all vehicles, not just the Corsa-e. In the future, I plan on getting another OBD 2 cable to look more into the CAN2 bus. All the values here are from CAN1. 

[image]
[image]
[image]

ramack, 2026-11-14:
That looks impressive. Thanks for sharing. I will try it in my eDoblo  asap, but have a reall though week ahead, so it might take a while.