openvehicles/Open-Vehicle-Monitoring-System-3 #720: 7600g Not establishing PPP with google Fi

rbray89, 2022-07-15:
I just upgraded the WWAN module to the LTE Global variant.

I'm seeing the following when trying to establish a connection:
```

D (845110) cellular: State timeout NetLoss => NetWait
I (845110) cellular: State: Enter NetWait state
D (845110) events: Signal(system.modem.netwait)
I (845120) ovms-server-v3: Tx event system.modem.netwait
V (846140) gsm-mux: ProcessFrame(CHAN=1, ADDR=05, CTRL=ff, FCS=84, LEN=31)
D (846140) gsm-nmea: Incoming RMC: $GPRMC,,V,,,,,,,,,,N*53
V (846140) gsm-mux: ProcessFrame(CHAN=2, ADDR=09, CTRL=ff, FCS=03, LEN=31)
V (846140) gsm-mux: ProcessFrame(CHAN=3, ADDR=0d, CTRL=ff, FCS=c1, LEN=31)
V (846150) gsm-mux: ProcessFrame(CHAN=4, ADDR=11, CTRL=ff, FCS=cc, LEN=31)
V (846150) gsm-mux: ProcessFrame(CHAN=1, ADDR=05, CTRL=ff, FCS=60, LEN=32)
D (846150) gsm-nmea: Incoming GNS: $GNGNS,,,,,,NNN,,,,,,*1D
V (846150) gsm-mux: ProcessFrame(CHAN=2, ADDR=09, CTRL=ff, FCS=e7, LEN=32)
V (846150) gsm-mux: ProcessFrame(CHAN=3, ADDR=0d, CTRL=ff, FCS=25, LEN=32)
V (846150) gsm-mux: ProcessFrame(CHAN=4, ADDR=11, CTRL=ff, FCS=28, LEN=32)
I (847110) gsm-ppp: StatusCallBack: User Interrupt
I (848110) ovms-server-v3: Tx event clock.2333
D (849110) cellular: State transition NetWait => NetStart
I (849110) cellular: State: Enter NetStart state
D (849110) events: Signal(system.modem.netstart)
I (849110) ovms-server-v3: Tx event system.modem.netstart
D (850110) cellular: Netstart AT+CGDCONT=1,"IP","h2g2";+CGDATA="PPP",1
V (850180) gsm-mux: ProcessFrame(CHAN=2, ADDR=09, CTRL=ff, FCS=fb, LEN=24)
D (850180) cellular: mux-rx-line #2: CONNECT 115200
I (850180) cellular: PPP Connection is ready to start
D (851110) cellular: State transition NetStart => NetMode
I (851110) cellular: State: Enter NetMode state
V (851110) cellular: Starting PPP
I (851110) gsm-ppp: Initialising...
V (851110) gsm-ppp: tx: 7e ff 7d 23 c0 21 7d 21 8e 7d 20 7d 34 7d 22 7d | ~.}#.!}!.} }4}"}
V (851110) gsm-ppp: tx: 26 7d 20 7d 20 7d 20 7d 20 7d 25 7d 26 96 7d 2b | &} } } } }%}&.}+
V (851110) gsm-ppp: tx: e7 22 7d 27 7d 22 7d 28 7d 22 e6 a0 7e          | ."}'}"}(}"..~   
D (851110) events: Signal(system.modem.netmode)
I (851110) ovms-server-v3: Tx event system.modem.netmode
V (851140) gsm-mux: ProcessFrame(CHAN=1, ADDR=05, CTRL=ff, FCS=84, LEN=31)
D (851140) gsm-nmea: Incoming RMC: $GPRMC,,V,,,,,,,,,,N*53
V (851140) gsm-mux: ProcessFrame(CHAN=3, ADDR=0d, CTRL=ff, FCS=c1, LEN=31)
V (851140) gsm-mux: ProcessFrame(CHAN=4, ADDR=11, CTRL=ff, FCS=cc, LEN=31)
V (851150) gsm-mux: ProcessFrame(CHAN=1, ADDR=05, CTRL=ff, FCS=60, LEN=32)
D (851150) gsm-nmea: Incoming GNS: $GNGNS,,,,,,NNN,,,,,,*1D
V (851150) gsm-mux: ProcessFrame(CHAN=3, ADDR=0d, CTRL=ff, FCS=25, LEN=32)
V (851150) gsm-mux: ProcessFrame(CHAN=4, ADDR=11, CTRL=ff, FCS=28, LEN=32)
V (851150) gsm-mux: ProcessFrame(CHAN=2, ADDR=09, CTRL=ff, FCS=82, LEN=58)
V (851150) gsm-ppp: rx: 7e ff 7d 23 c0 21 7d 21 8d 7d 20 7d 39 7d 22 7d | ~.}#.!}!.} }9}"}
V (851150) gsm-ppp: rx: 26 7d 20 7d 20 7d 20 7d 20 7d 23 7d 25 c2 23 7d | &} } } } }#}%.#}
V (851150) gsm-ppp: rx: 25 7d 25 7d 26 40 49 51 97 7d 27 7d 22 7d 28 7d | %}%}&@IQ.}'}"}(}
V (851150) gsm-ppp: rx: 22 2c 93 7e                                     | ",.~            
V (851150) gsm-mux: ProcessFrame(CHAN=2, ADDR=09, CTRL=ff, FCS=a1, LEN=52)
V (851150) gsm-ppp: rx: 7e ff 7d 23 c0 21 7d 22 8e 7d 20 7d 34 7d 22 7d | ~.}#.!}".} }4}"}
V (851150) gsm-ppp: rx: 26 7d 20 7d 20 7d 20 7d 20 7d 25 7d 26 96 7d 2b | &} } } } }%}&.}+
V (851150) gsm-ppp: tx: 7e ff 7d 23 c0 21 7d 23 8d 7d 20 7d 28 7d 23 7d | ~.}#.!}#.} }(}#}
V (851160) gsm-ppp: rx: e7 22 7d 27 7d 22 7d 28 7d 22 7d 2d c9 7e       | ."}'}"}(}"}-.~  
V (851160) gsm-ppp: tx: 24 c0 23 a5 7f 7e                               | $.#..~          
V (851180) gsm-mux: ProcessFrame(CHAN=2, ADDR=09, CTRL=ff, FCS=8b, LEN=56)
V (851180) gsm-ppp: rx: 7e ff 7d 23 c0 21 7d 21 8e 7d 20 7d 38 7d 22 7d | ~.}#.!}!.} }8}"}
V (851180) gsm-ppp: rx: 26 7d 20 7d 20 7d 20 7d 20 7d 23 7d 24 c0 23 7d | &} } } } }#}$.#}
V (851180) gsm-ppp: rx: 25 7d 26 40 49 51 97 7d 27 7d 22 7d 28 7d 22 73 | %}&@IQ.}'}"}(}"s
V (851180) gsm-ppp: rx: 89 7e                                           | .~              
V (851180) gsm-ppp: tx: 7e ff 7d 23 c0 21 7d 22 8e 7d 20 7d 38 7d 22 7d | ~.}#.!}".} }8}"}
V (851180) gsm-ppp: tx: 26 7d 20 7d 20 7d 20 7d 20 7d 23 7d 24 c0 23 7d | &} } } } }#}$.#}
V (851180) gsm-ppp: tx: 25 7d 26 40 49 51 97 7d 27 7d 22 7d 28 7d 22 bf | %}&@IQ.}'}"}(}".
V (851180) gsm-ppp: tx: 64 7e                                           | d~              
V (851180) gsm-ppp: tx: 7e ff 03 c0 23 01 30 00 2e 23 27 7d 5d 22 7d 5d | ~...#.0..#'}]"}]
V (851180) gsm-ppp: tx: 28 7d 5d 22 73 89 7d 5e 20 7d 5d 20 7d 5d 20 7d | (}]"s.}^ }] }] }
V (851180) gsm-ppp: tx: 5d 20 7d 5d 23 7d 5d 24 c0 23 7d 5d 34 3f 08 40 | ] }]#}]$.#}]4?.@
V (851180) gsm-ppp: tx: 18 26 08 40 30 0d 06 05 2c de fb 3f 12 29 52 7e | .&.@0...,..?.)R~
V (851210) gsm-mux: ProcessFrame(CHAN=2, ADDR=09, CTRL=ff, FCS=f2, LEN=26)
V (851210) gsm-ppp: rx: 7e ff 7d 23 c0 21 7d 2b 8f 7d 20 7d 28 40 49 51 | ~.}#.!}+.} }(@IQ
V (851210) gsm-ppp: rx: 97 fb 5c 7e                                     | ..\~            
V (851210) gsm-mux: ProcessFrame(CHAN=2, ADDR=09, CTRL=ff, FCS=3c, LEN=17)
V (851210) gsm-ppp: rx: 7e c0 23 02 30 00 05 00 b4 60 7e                | ~.#.0....`~     
V (851210) gsm-ppp: tx: 7e ff 03 80 21 01 30 00 1c 02 06 00 2d 0f 01 03 | ~...!.0.....-...
W (851300) websocket: WebSocketHandler[0x3f8b9814]: job queue overflow detected
W (851300) websocket: WebSocketHandler[0x3f8b9814]: job queue overflow resolved, 22 drops
W (852110) cellular: Lost network connection (+PPP disconnect in NetMode)
D (852110) cellular: State transition NetMode => NetLoss
I (852110) cellular: State: Enter NetLoss state
D (852110) cellular: mux-tx #3: AT+CGATT=0
V (852110) v-nissanleaf: Range: ideal=0 km, est=0 km, full=0 km
V (852110) cellular: Stopping PPP
D (852110) v-nissanleaf: Poll state: 0
I (852110) gsm-ppp: Shutting down (hard)...
V (852110) gsm-ppp: tx: 7e ff 7d 23 c0 21 7d 25 8f 7d 20 7d 30 55 73 65 | ~.}#.!}%.} }0Use
V (852110) gsm-ppp: tx: 72 20 72 65 71 75 65 73 74 c7 3d 7e             | r request.=~    
I (852110) gsm-ppp: PPP is shutdown
D (852120) events: Signal(system.modem.netloss)
I (852120) ovms-server-v3: Tx event system.modem.netloss
D (852120) events: Signal(system.modem.down)
D (852120) netmanager: send cmd 3 from 0x0
D (852120) netmanager: done cmd 3 from 0x0
I (852120) ovms-server-v3: Tx event system.modem.down
D (852140) netmanager: MongooseTask: got cmd 3 from 0x0
D (852140) netmanager: CleanupConnections: conn 3f8c5efc -> iface st1
D (852150) netmanager: CleanupConnections: conn 3f8b9814 -> iface st1
D (852150) netmanager: CleanupConnections: conn 3f8c7e38 -> iface st1
D (852150) netmanager: CleanupConnections: conn 3f8b98b0 -> iface st1
D (852150) netmanager: MongooseTask: done cmd 3 from 0x0
V (858110) gsm-ppp: tx: 7e ff 7d 23 c0 21 7d 25 90 7d 20 7d 30 55 73 65 | ~.}#.!}%.} }0Use
V (858110) gsm-ppp: tx: 72 20 72 65 71 75 65 73 74 96 3e 7e             | r request.>~    
D (862110) cellular: State timeout NetLoss => NetWait
I (862110) cellular: State: Enter NetWait state
```

leres, 2022-07-15:
I seem to remember having trouble with one of my hologram.io sims not immediately  connecting when I switched to the new 7600g modem.

Have you tried rebooting or powering the module off for a few minutes and retrying?

rbray89, 2022-07-15:
@leres Yeah, I tried disconnecting 12V for 30s, as well as powering down the module for a while. Haven't gone as long as a few minutes though.

rbray89, 2022-07-25:
Update: Hologram.IO sim works.

May try another Google Fi sim.

markwj, 2022-08-30:
Please could you try this with the latest 3.3-47 edge firmware. We have added support for AT+CGREG and AT+CEREG, as well as the existing AT+CREG, network statuses and that should cope better with networks such as Google Fi and in general with cellular services that deny registration on some protocols while accepting on others.

rbray89, 2022-09-04:
@markwj still a no-go...
`
I (2893876) cellular: State: Enter Identify state
D (2894186) cellular: tx-cmd: AT+CGMM
I (2894196) cellular: Identified cellular modem: SIM7600/Experimental support for SIMCOM SIM7600
D (2894196) cellular: Remove old 'auto' modem driver
I (2894196) cellular: Set modem driver to 'SIM7600'
I (2894196) cellular: State: Enter PoweredOn state
D (2894196) events: Signal(system.modem.installed)
D (2894206) events: Signal(system.modem.poweredon)
V (2903186) v-nissanleaf: Range: ideal=0 km, est=0 km, full=0 km
D (2903186) v-nissanleaf: Poll state: 0
D (2904186) cellular: tx-cmd: AT+CPIN?;+CREG=1;+CGREG=1;+CEREG=1;+CTZU=1;+CTZR=1;+CLIP=1;+CMGF=1;+CNMI=1,2,0,0,0;+CSDH=1;+CMEE=2;+CSQ;+AUTOCSQ=1,1;E0;S0=0
D (2904216) cellular: mux-rx-line #0: AT+CPIN?;+CREG=1;+CGREG=1;+CEREG=1;+CTZU=1;+CTZR=1;+CLIP=1;+CMGF=1;+CNMI=1,2,0,0,0;+CSDH=1;+CMEE=2;+CSQ;+AUTOCSQ=1,1;E0;S0=0
D (2904226) cellular: mux-rx-line #0: +CPIN: READY
D (2904226) cellular: mux-rx-line #0: +CSQ: 13,99
I (2904226) cellular: Signal Quality is: 13 (-87 dBm)
D (2904656) cellular: mux-rx-line #0: +CSQ: 20,99
I (2904656) cellular: Signal Quality is: 20 (-73 dBm)
D (2906186) cellular: tx-cmd: AT+CGMR;+ICCID
D (2906206) cellular: mux-rx-line #0: +CGMR: LE20B03SIM7600M21-A
D (2906206) cellular: mux-rx-line #0: +ICCID: 8901260971129626899
V (2913186) v-nissanleaf: Range: ideal=0 km, est=0 km, full=0 km
D (2913186) v-nissanleaf: Poll state: 0
D (2914186) cellular: tx-cmd: AT+CMUX=0
I (2914206) cellular: State: Enter MuxStart state
V (2914206) cellular: Starting MUX
I (2914206) gsm-mux: Start MUX
D (2914206) events: Signal(system.modem.muxstart)
V (2914226) gsm-mux: ProcessFrame(CHAN=0, ADDR=03, CTRL=73, FCS=d7, LEN=6)
I (2914226) gsm-mux: Channel #0 is open
V (2914236) gsm-mux: ProcessFrame(CHAN=1, ADDR=07, CTRL=73, FCS=15, LEN=6)
I (2914236) gsm-mux: Channel #1 is open
V (2914256) gsm-mux: ProcessFrame(CHAN=2, ADDR=0b, CTRL=73, FCS=92, LEN=6)
I (2914256) gsm-mux: Channel #2 is open
V (2914266) gsm-mux: ProcessFrame(CHAN=3, ADDR=0f, CTRL=73, FCS=50, LEN=6)
I (2914266) gsm-mux: Channel #3 is open
V (2914286) gsm-mux: ProcessFrame(CHAN=4, ADDR=13, CTRL=73, FCS=5d, LEN=6)
I (2914286) gsm-mux: Channel #4 is open
D (2915186) cellular: State transition MuxStart => NetWait
I (2915186) cellular: State: Enter NetWait state
V (2915186) cellular: Starting NMEA
I (2915186) gsm-nmea: Startup
D (2915186) cellular: mux-tx #4: AT+CGPS=0
D (2915186) events: Signal(system.modem.netwait)
D (2917186) cellular: mux-tx #4: AT+CGPSNMEA=258
D (2917186) cellular: mux-tx #4: AT+CGPSINFOCFG=5,258
D (2917186) cellular: mux-tx #4: AT+CGPS=1,1
V (2917186) gsm-mux: ProcessFrame(CHAN=4, ADDR=11, CTRL=ff, FCS=02, LEN=12)
V (2917206) gsm-mux: ProcessFrame(CHAN=4, ADDR=11, CTRL=ff, FCS=02, LEN=12)
V (2917216) gsm-mux: ProcessFrame(CHAN=4, ADDR=11, CTRL=ff, FCS=02, LEN=12)
V (2917216) gsm-mux: ProcessFrame(CHAN=4, ADDR=11, CTRL=ff, FCS=02, LEN=12)
V (2917216) gsm-mux: ProcessFrame(CHAN=1, ADDR=05, CTRL=ff, FCS=84, LEN=31)
D (2917216) gsm-nmea: Incoming RMC: $GPRMC,,V,,,,,,,,,,N*53
V (2917216) gsm-mux: ProcessFrame(CHAN=2, ADDR=09, CTRL=ff, FCS=03, LEN=31)
V (2917226) gsm-mux: ProcessFrame(CHAN=3, ADDR=0d, CTRL=ff, FCS=c1, LEN=31)
V (2917226) gsm-mux: ProcessFrame(CHAN=4, ADDR=11, CTRL=ff, FCS=cc, LEN=31)
V (2917226) gsm-mux: ProcessFrame(CHAN=1, ADDR=05, CTRL=ff, FCS=60, LEN=32)
D (2917226) gsm-nmea: Incoming GNS: $GNGNS,,,,,,NNN,,,,,,*1D
V (2917226) gsm-mux: ProcessFrame(CHAN=2, ADDR=09, CTRL=ff, FCS=e7, LEN=32)
V (2917226) gsm-mux: ProcessFrame(CHAN=3, ADDR=0d, CTRL=ff, FCS=25, LEN=32)
V (2917226) gsm-mux: ProcessFrame(CHAN=4, ADDR=11, CTRL=ff, FCS=28, LEN=32)
V (2922216) gsm-mux: ProcessFrame(CHAN=1, ADDR=05, CTRL=ff, FCS=2f, LEN=46)
D (2922216) gsm-nmea: Incoming RMC: $GPRMC,061240.74,V,,,,,,,040922,,,N*72
D (2922216) time: gsm-nmea (stratum 2 trusted 1) provides time Sun Sep  4 06:12:40 2022 (000000us) UTC
V (2922216) gsm-mux: ProcessFrame(CHAN=2, ADDR=09, CTRL=ff, FCS=a8, LEN=46)
V (2922226) gsm-mux: ProcessFrame(CHAN=3, ADDR=0d, CTRL=ff, FCS=6a, LEN=46)
V (2922226) gsm-mux: ProcessFrame(CHAN=4, ADDR=11, CTRL=ff, FCS=67, LEN=46)
V (2922226) gsm-mux: ProcessFrame(CHAN=1, ADDR=05, CTRL=ff, FCS=d7, LEN=41)
D (2922226) gsm-nmea: Incoming GNS: $GNGNS,061240.74,,,,,NNN,,,,,,*31
[... 389 lines cut ...]
W (2980186) cellular: Lost network connection (+PPP disconnect in NetMode)
D (2980186) cellular: State transition NetMode => NetLoss
I (2980186) cellular: State: Enter NetLoss state
D (2980186) cellular: mux-tx #3: AT+CGATT=0
V (2980186) cellular: Stopping PPP
I (2980186) gsm-ppp: Shutting down (hard)...
V (2980186) gsm-ppp: tx: 7e ff 7d 23 c0 21 7d 25 2c 7d 20 7d 30 55 73 65 | ~.}#.!}%,} }0Use
V (2980186) gsm-ppp: tx: 72 20 72 65 71 75 65 73 74 7d 35 97 7e          | r request}5.~   
I (2980186) gsm-ppp: PPP is shutdown
D (2980186) events: Signal(system.modem.netloss)
D (2980196) events: Signal(system.modem.down)
D (2980196) netmanager: send cmd 3 from 0x0
D (2980196) netmanager: done cmd 3 from 0x0
D (2980236) netmanager: MongooseTask: got cmd 3 from 0x0
D (2980236) netmanager: CleanupConnections: conn 3f8c0f98 -> iface st1
D (2980236) netmanager: CleanupConnections: conn 3f8d78d8 -> iface st1
D (2980236) netmanager: CleanupConnections: conn 3f8c1084 -> iface st1
D (2980236) netmanager: MongooseTask: done cmd 3 from 0x0
V (2983186) v-nissanleaf: Range: ideal=0 km, est=0 km, full=0 km
D (2983186) v-nissanleaf: Poll state: 0
V (2986186) gsm-ppp: tx: 7e ff 7d 23 c0 21 7d 25 2d 7d 20 7d 30 55 73 65 | ~.}#.!}%-} }0Use
V (2986186) gsm-ppp: tx: 72 20 72 65 71 75 65 73 74 3f df 7e             | r request?.~    
D (2990186) cellular: State timeout NetLoss => NetWait
I (2990186) cellular: State: Enter NetWait state
D (2990186) events: Signal(system.modem.netwait)`

dexterbg, 2022-09-04:
> D (2945316) cellular: mux-rx-line #3: NO CARRIER

Maybe you've got a signal quality issue? Hologram could work because they let you connect to a cell tower closer than the one used by Google Fi.

- Check the internal pigtail cable & connection, verify it's connected to the right socket and works (these break easily)
- Try placing the antenna as high as possible
- Try replacing the antenna

rbray89, 2022-09-04:
@markwj 

Possibly? Though it was working with 3G without issues with the same antennas and placement.

I assume band support is pretty decent? FI is an MVNO leveraging T-Mobile.

Also, this is the build you're referring to?

`Firmware: 3.3.003-2-gd00de6c0/ota_0/main (build idf v3.3.4-848-g1ff5e24b1 Sep 2 2022 17:00:44)`

rbray89, 2022-09-04:
> Maybe you've got a signal quality issue? Hologram could work because they let you connect to a cell tower closer than the one used by Google Fi.

Unlikely. SSRI is the same. Additionally, Hologram is using the same T-Mobile network it seems.

Could this be an IPv6 issue?